SpikeInterface/spikeinterface #1332: Sphinx (6.1.3) Fails to build documentation

pauladkisson, 2023-02-16:
Sphinx throws an exception when trying to build the docs.  Looks like the issue is in widgets.  See traceback:
```
...
updating environment: [new config] 81 added, 0 changed, 0 removed
/Users/pauladkisson/Documents/CatalystNeuro/spikeinterface_select_peaks_bug/spikeinterface/spikeinterface/widgets/base.py:docstring of spikeinterface.widgets.base.define_widget_function_from_class.<locals>.widget_func:94: WARNING: Inline strong start-string without end-string.

Exception occurred:
  File "/opt/anaconda3/envs/sienv_selectpeaks/lib/python3.9/site-packages/sphinx/ext/extlinks.py", line 98, in role
    full_url = base_url % part
TypeError: not all arguments converted during string formatting
```

Note: Previous version of sphinx (5.0.2) raises a warning about this exact error:
```
...
loading intersphinx inventory from https://probeinterface.readthedocs.io/en/stable/objects.inv...
WARNING: extlinks: Sphinx-6.0 will require base URL to contain exactly one '%s' and all other '%' need to be escaped as '%%'.
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 81 source files that are out of date
updating environment: [new config] 81 added, 0 changed, 0 removed
reading sources... [100%] whatisnew                                                                                                            
/Users/pauladkisson/Documents/CatalystNeuro/spikeinterface_select_peaks_bug/spikeinterface/spikeinterface/widgets/base.py:docstring of spikeinterface.widgets.base.define_widget_function_from_class.<locals>.widget_func:99: WARNING: Inline strong start-string without end-string.
...
```

samuelgarcia, 2023-02-17:
Hi Paul.
Yes I saw that. Do you have time to have look what is the this `%s` not allowed ?

pauladkisson, 2023-02-17:
Sure, I can take a look!

PR #1340 (2023-02-17): Fixes Bug that was preventing sphinx>=6.0.0 to render documentation properly
Fixes #1332 

See [here](https://github.com/sphinx-doc/sphinx/issues/11094) for a discussion of the issue and solution.

I also cleaned up one of the docstrings that I noticed wasn't rendering properly.
closed 2023-02-17 by pauladkisson