SpikeInterface/spikeinterface #1333: Issue with order_channels_by_depth

bradlytstone, 2023-02-16:
I have attempted several ways to use the 'detect_bad_channels' function from spikeinterface.preprocessing, however....I get an error "IndexError: arrays used as indices must be of integer (or boolean) type" which is evoked within the subfunction 'order_channels_by_depth'. 

The error occurs during this step: **locations = locations[channel_inds, :]**

Investigating this further, I realized that the 'channel_ids' variable is intended to be a vector of integer values corresponding to the channels, but (at least in my case) the 'channel_ids' variable is a vector full of strings. As a first pass, I changed the error-causing line to:

#converting the string variables to integers and making the indices pythonic
**locations[[int(x)-1 for x in channel_inds], :]**  

Following this change, I ran detect_bad_channels and was pleased that it worked, but a "warning" emerged: 

#the code for detection
bad_channel_ids, channel_labels = spikeinterface.preprocessing.detect_bad_channels(reader_filtered)

#warning label
**UserWarning: nperseg = 244 is greater than input length  = 32, using nperseg = 32
  warnings.warn('nperseg = {0:d} is greater than input length '**

The 'input length = 32' was concerning and leads me to believe (because I am working with a 32-channel shank) that the data is being read in a transposed fashion, but I cannot be sure?

For clarity on this issue....I am using a 32-channel neuronexus probe and recording using TDT, and importing into spikeinterface using read_tdt.

Thanks

alejoe91, 2023-02-17:
Thanks for the report! I'll take a look. Note that channel_inds is not channel_ids! inds are the indices, so (in principle) they should be ints.

alejoe91, 2023-02-17:
@bradlytstone can you paste the entire code?

bradlytstone, 2023-02-17:
import os
import easygui
import numpy as np
import matplotlib.pyplot as plt

import spikeinterface.full as si  
import spikeinterface.preprocessing as spre
import spikeinterface.extractors as se

# Create recording extractor object
reader = se.read_tdt(folder_path=dir_name, stream_name = 'SU_1')

# Set Parameters
params = {'chunk_duration': '1s',
 'common_ref_removal': False,
 'detect_sign': -1,
 'detect_threshold': 5,
 'freq_max': 3000.0,
 'freq_min': 300.0,
 'n_jobs': 20,
 'nested_params': None,
 'progress_bar': True}

# Design the probe based on experiment parameters
probe = generate_multi_columns_probe(num_columns=2,
                                     xpitch=43, ypitch=50,
                                     num_contact_per_column=[16, 16],
                                     y_shift_per_column=[0, 25],
                                     contact_shapes='circle', contact_shape_params={'radius': 7.5})

# these are the remapped (done with Myles from TDT)
device_channel_indices=[7,5,3,1,30,28,25,27,6,29,4,31,2,24,0,26,9,11,\
                        13,15,16,18,23,21,8,19,10,17,12,22,14,20]
all_channels = device_channel_indices.copy()
    
# the probe has to be linked to the recording
probe.set_device_channel_indices(device_channel_indices)
fig,ax=plt.subplots(figsize=(8,8))
plot_probe(probe,ax=ax,with_device_index=True)

# set probe within reader object
reader = reader.set_probe(probe)

# Get sampling rate
fs = reader.get_sampling_frequency()

# Apply a filter
reader_filtered = si.bandpass_filter(reader, freq_min=params['freq_min'], freq_max=params['freq_max'])

# Detect bad channels and remove them
bad_channel_ids, channel_labels = spre.detect_bad_channels(reader_filtered)

# ERROR
Here is the first error (without me changing the subfunction: order_channels_by_depth)
**Traceback (most recent call last):

  Input In [5] in <cell line: 1>
    bad_channel_ids, channel_labels = spre.detect_bad_channels(reader_filtered)

  File ~\miniconda3\lib\site-packages\spikeinterface\preprocessing\detect_bad_channels.py:167 in detect_bad_channels
    order_f, order_r = order_channels_by_depth(recording=recording, dimensions=('x', 'y'))

  File ~\miniconda3\lib\site-packages\spikeinterface\core\recording_tools.py:227 in order_channels_by_depth
    locations = locations[channel_inds, :]

IndexError: arrays used as indices must be of integer (or boolean) type**

If I modify the subfunction, order_channels_by_depth, I can proceed....

# Detect bad channels and remove them
bad_channel_ids, channel_labels = spre.detect_bad_channels(reader_filtered)

# WARNING
Here is the subsequent _Warning_
**C:\Users\SuperComputer1\miniconda3\lib\site-packages\scipy\signal\_spectral_py.py:2014: UserWarning: nperseg = 244 is greater than input length  = 32, using nperseg = 32
  warnings.warn('nperseg = {0:d} is greater than input length '**

Thank you. Again, I am happy to send any/all recordings for replication of error(s)/warning(s)

alejoe91, 2023-02-17:
Thanks! What is the output of:
```
print(reader)
print(len(reader.channel_ids))
```

?

bradlytstone, 2023-02-17:
print(reader)
TdtRecordingExtractor: 32 channels - 1 segments - 24.4kHz - 602.302s

print(len(reader.channel_ids))
32

alejoe91, 2023-02-17:
and `reader.get_channel_locations().shape`?

bradlytstone, 2023-02-17:
reader.get_channel_locations().shape
(32, 2)

alejoe91, 2023-02-17:
Ok, I'm out of guesses...it would be great if you can share one of the dataset sthat's causing the issue

bradlytstone, 2023-02-17:
@alejoe91 would you prefer this be done via link in Github or email? I had previously sent a link to the data via email.

alejoe91, 2023-02-17:
ah, saw it now. Can you share a small example? I guess the zip file you shared contain several datasets?

alejoe91, 2023-02-17:
There was indeed a bug on our side.

Can you try installing from this PR #1338 ?

BTW, I haven't found any noisy channels on your data, so probably it won't solve the other issue

bradlytstone, 2023-02-17:
I installed the newest version, and now it does not allow me to do anything.

Traceback (most recent call last):

  Input In [9] in <cell line: 1>
    import spikeinterface.full as si

  File ~\spikeinterface\spikeinterface\full.py:16 in <module>
    from .extractors import *

  File ~\spikeinterface\spikeinterface\extractors\__init__.py:1 in <module>
    from .extractorlist import *

  File ~\spikeinterface\spikeinterface\extractors\extractorlist.py:10 in <module>
    from .neoextractors import *

  File ~\spikeinterface\spikeinterface\extractors\neoextractors\__init__.py:15 in <module>
    from .maxwell import (MaxwellRecordingExtractor, MaxwellEventExtractor,

  File ~\spikeinterface\spikeinterface\extractors\neoextractors\maxwell.py:5 in <module>
    from spikeinterface import BaseEvent, BaseEventSegment

alejoe91, 2023-02-17:
What is the error? please paste the full traceback.

If you installed from sources, you might need to run from another folder, otherwise Python thinks that `spikeinterface` refers to the folder, not the package.

bradlytstone, 2023-02-17:
I installed using: git clone https://github.com/SpikeInterface/spikeinterface.git

The error traceback (in full) is what I just submitted. But here it is:

import spikeinterface.full as si
Traceback (most recent call last):

  Input In [10] in <cell line: 1>
    import spikeinterface.full as si

  File ~\spikeinterface\spikeinterface\full.py:16 in <module>
    from .extractors import *

  File ~\spikeinterface\spikeinterface\extractors\__init__.py:1 in <module>
    from .extractorlist import *

  File ~\spikeinterface\spikeinterface\extractors\extractorlist.py:10 in <module>
    from .neoextractors import *

  File ~\spikeinterface\spikeinterface\extractors\neoextractors\__init__.py:15 in <module>
    from .maxwell import (MaxwellRecordingExtractor, MaxwellEventExtractor,

  File ~\spikeinterface\spikeinterface\extractors\neoextractors\maxwell.py:5 in <module>
    from spikeinterface import BaseEvent, BaseEventSegment

ImportError: cannot import name 'BaseEvent' from 'spikeinterface' (unknown location)

alejoe91, 2023-02-17:
Solution here: https://github.com/SpikeInterface/spikeinterface/issues/1316

PR #1338 (2023-02-20): Fix ids_to_indices when channel_ids is None
fixes #1333

there was a bug when using `ids_to_indices` with `ids=None` (which normally is not used)
closed 2023-02-20 by samuelgarcia