SpikeInterface/spikeinterface #1334: can not run ironclust in spiketutorial

ZoeChen96, 2023-02-17:
Hi there,
I tried to run the spiketutorial 0.96 and can not run the ironclust:
`Warning! The recording is already filtered, but ironclust filter is enabled
Output exceeds the [size limit](command:workbench.action.openSettings?[). Open the full output data [in a text editor](command:workbench.action.openLargeOutput?23439e66-ef23-48fc-8434-e4e3314bd22b)
---------------------------------------------------------------------------
BrokenProcessPool                         Traceback (most recent call last)
Cell In[67], line 1
----> 1 sorting_IC = ss.run_sorter('ironclust', recording_sub, 
      2                            output_folder=base_folder/'results_IC3',
      3                         verbose=True)

File /imec/other/macaw/chen14/anaconda43/envs/si96_env/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:143, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    136             container_image = singularity_image
    137     return run_sorter_container(
    138         container_image=container_image,
    139         mode=mode,
    140         **common_kwargs,
    141     )
--> 143 return run_sorter_local(**common_kwargs)

File /imec/other/macaw/chen14/anaconda43/envs/si96_env/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:162, in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
    158 output_folder = SorterClass.initialize_folder(
    159     recording, output_folder, verbose, remove_existing_folder)
    160 SorterClass.set_params_to_folder(
    161     recording, output_folder, sorter_params, verbose)
--> 162 SorterClass.setup_recording(recording, output_folder, verbose=verbose)
    163 SorterClass.run_from_folder(output_folder, raise_error, verbose)
    164 if with_output:
...
--> 707         raise BrokenProcessPool(self._broken)
    708     if self._shutdown_thread:
    709         raise RuntimeError('cannot schedule new futures after shutdown')

BrokenProcessPool: A child process terminated abruptly, the process pool is not usable anymore`

I then tried with spike tutorial 0.95 and run into another error. As suggested, I have a look at the spikeinterface_log.json:
`{
    "sorter_name": "ironclust",
    "sorter_version": "5.9.8",
    "datetime": "2023-02-16T18:15:38.934514",
    "runtime_trace": [
        "",
        "< M A T L A B (R) >",
        "Copyright 1984-2020 The MathWorks, Inc.",
        "R2020a Update 7 (9.8.0.1721703) 64-bit (glnxa64)",
        "July 15, 2021",
        "",
        "",
        "< M A T L A B (R) >",
        "Copyright 1984-2020 The MathWorks, Inc.",
        "R2020a Update 7 (9.8.0.1721703) 64-bit (glnxa64)",
        "July 15, 2021",
        "",
        "",
        "",
        "To get started, type doc.",
        "To get started, type doc.",
        "For product information, visit www.mathworks.com.",
        "For product information, visit www.mathworks.com.",
        "",
        "",
        "===================================================",
        "===================================================",
        "IronClust Version: 2",
        "IronClust Version: 2",
        "===================================================",
        "===================================================",
        "Removed 0 lock(s).",
        "Removed 0 lock(s).",
        "Running irc2.m (5.9.8)",
        "Running irc2.m (5.9.8)",
        "Created /imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.95_Sep2022/results_IC2/tmp/raw_geom.prm",
        "Created /imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.95_Sep2022/results_IC2/tmp/raw_geom.prm",
        "Read /imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.95_Sep2022/results_IC2/ironclust_dataset/raw.mda (1/28), took 2.4s (104.8 MB/s, 250.2 MB)",
        "Read /imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.95_Sep2022/results_IC2/ironclust_dataset/raw.mda (1/28), took 2.4s (104.8 MB/s, 250.2 MB)",
        "Caught \"std::exception\" Exception message is:",
        "Message Catalog MATLAB:license was not loaded from the file. Please check file location, format or contents",
        "Caught \"std::exception\" Exception message is:",
        "Message Catalog MATLAB:license was not loaded from the file. Please check file location, format or contents"
    ],
    "error": true,
    "error_trace": "Traceback (most recent call last):\n  File \"/imec/other/macaw/chen14/anaconda43/envs/si95_env/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py\", line 225, in run_from_folder\n    SorterClass._run_from_folder(output_folder, sorter_params, verbose)\n  File \"/imec/other/macaw/chen14/anaconda43/envs/si95_env/lib/python3.9/site-packages/spikeinterface/sorters/ironclust/ironclust.py\", line 253, in _run_from_folder\n    raise Exception(f'Result file does not exist: {result_fname}')\nException: Result file does not exist: /imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.95_Sep2022/results_IC2/tmp/firings.mda\n",
    "run_time": null
}
`

I think I installed the ironclust properly. I did as what you did in spiketutorial 0.95 (git clone and set path), and I checked the installed sorter which shows 
`Setting IRONCLUST_PATH environment variable for subprocess calls to: /imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.95_Sep2022/../Official_Tutorial_SI_0.96_Oct22/ironclust
RUNNING SHELL SCRIPT: /tmp/tmp_shellscript8wt1d0od/script.sh
RUNNING SHELL SCRIPT: /tmp/tmp_shellscriptrn0gfaqg/script.sh
RUNNING SHELL SCRIPT: /tmp/tmp_shellscriptv_u9mm4_/script.sh
RUNNING SHELL SCRIPT: /tmp/tmp_shellscriptan7ht20u/script.sh
RUNNING SHELL SCRIPT: /tmp/tmp_shellscriptczcl11yg/script.sh
RUNNING SHELL SCRIPT: /tmp/tmp_shellscript9r9faimb/script.sh
RUNNING SHELL SCRIPT: /tmp/tmp_shellscriptok9p6b2e/script.sh
RUNNING SHELL SCRIPT: /tmp/tmp_shellscriptvxm37la0/script.sh

['ironclust', 'spykingcircus2', 'tridesclous', 'tridesclous2']`

BTW, I installed the spikeinterface environment 0.96 and 0.95 seperately, for 0.96 I keep the origonal environment.yml, and for 0.95 I changed one line which is 
`    - spikeinterface[full]==0.95.1` to ensure I got the 0.95 version. But when I run the ironclust in SI 0.95, I keep using the dataset in 0.96 (the nwb one).

ZoeChen96, 2023-02-17:
Another error for your information: when I use SI 0.97 (but 0.96 tutorial), I have another problem with saving binary file:
[image]

I am not familiar with python but it seems a problem with multi-process, so I tried to deleted the `**job_kwargs` ,and the code is like: 
`if (base_folder / "preprocessed6").is_dir():
    recording_saved = si.load_extractor(base_folder / "preprocessed6")
else:
    recording_saved = recording_sub.save(folder=base_folder / "preprocessed6")`
Then the error was gone.

And I found in the video you use SI 0.95 with the SI 0.96 tutorial. So I go back to SI 0.95 and tried, the error did not exist.

ZoeChen96, 2023-02-17:
> Hi!
> 
> The problem seems to be related to some parallel processing settings. Can you try to set `n_jobs=1` to see if at least it runs?

Hi, I hope I set the n_jobs correctly (you can find in the code). I tried with the SI 0.97 environment, it was still there. 
[image]

alejoe91, 2023-02-17:
you should propagate sorter_params to the run function:

```
sorting_IC = ss.run_sorter('ironclust', recording_sub, output_folder=base_folder /  "results_IC_job1", verbose=True, **sorter_params)
```

ZoeChen96, 2023-02-17:
Hi Alessio,

Thanks a lot! I ran it successfully and have the raster plots.
[image]

closed 2023-02-17 by ZoeChen96