SpikeInterface/spikeinterface #1335: Issue when running HDsort

ZoeChen96, 2023-02-17:
Hi there,

Sorry for disturbing again. I wanted to run HDsort, and I keep most of the file the same as spiketutorial-0.96. However it raises the error:
`sorting_HDS = ss.run_sorter('hdsort', recording_sub, output_folder=base_folder /  "results_HDS",verbose = True)`
> Warning! The recording is already filtered, but hdsort filter is enabled
ValueError                                Traceback (most recent call last)
Cell In[49], line 1
----> 1 sorting_HDS = ss.run_sorter('hdsort', recording_sub, output_folder=base_folder /  "results_HDS",verbose = True)
File /imec/other/macaw/chen14/anaconda43/envs/si96_pure/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:143, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    136             container_image = singularity_image
    137     return run_sorter_container(
    138         container_image=container_image,
    139         mode=mode,
    140         **common_kwargs,
    141     )
--> 143 return run_sorter_local(**common_kwargs)
File /imec/other/macaw/chen14/anaconda43/envs/si96_pure/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:162, in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
    158 output_folder = SorterClass.initialize_folder(
    159     recording, output_folder, verbose, remove_existing_folder)
    160 SorterClass.set_params_to_folder(
    161     recording, output_folder, sorter_params, verbose)
--> 162 SorterClass.setup_recording(recording, output_folder, verbose=verbose)
    163 SorterClass.run_from_folder(output_folder, raise_error, verbose)
    164 if with_output:

File /imec/other/macaw/chen14/anaconda43/envs/si96_pure/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py:196, in BaseSorter.setup_recording(cls, recording, output_folder, verbose)
    194     all_params = json.load(f)
    195     sorter_params = all_params['sorter_params']
--> 196 cls._setup_recording(recording, sorter_output_folder, sorter_params, verbose)
File /imec/other/macaw/chen14/anaconda43/envs/si96_pure/lib/python3.9/site-packages/spikeinterface/sorters/hdsort/hdsort.py:212, in HDSortSorter._setup_recording(cls, recording, sorter_output_folder, params, verbose)
    209         print('Using MaxOne format directly')
    210 else:
    211     # Generate three files dataset in Mea1k format
--> 212     trace_file_name = cls.write_hdsort_input_format(recording,
    213                                                     str(sorter_output_folder / 'recording.h5'),
    214                                                     chunk_memory=params["chunk_memory"])
    215     # ~ self.params['file_format'] = 'mea1k'
    216     file_format = 'mea1k'
File /imec/other/macaw/chen14/anaconda43/envs/si96_pure/lib/python3.9/site-packages/spikeinterface/sorters/hdsort/hdsort.py:309, in HDSortSorter.write_hdsort_input_format(cls, recording, save_path, chunk_memory)
    307 # channel should be from 0 to num_channel - 1
    308 for i, ch in enumerate(recording.get_channel_ids()):
--> 309     mapping[i] = (ch, x[i], y[i], i)
    310 ephys.create_dataset('mapping', data=mapping)
    311 # save traces
ValueError: invalid literal for int() with base 10: 'AP0'

ZoeChen96, 2023-02-17:
BTW, I just come from the issue of ironclust: [https://github.com/SpikeInterface/spikeinterface/issues/1334](url)
There you suggest me to add the **sorter_params, but I found it does not work with HDsort and SpykingCircus2, it will give the error of bad params. Could you please tell me the reason?

The code is:
`job_kwargs = dict(n_jobs=1, chunk_duration="1s", progress_bar=True)
sorter_params = job_kwargs
sorting_HDS = ss.run_sorter('hdsort', recording_sub, output_folder=base_folder /  "results_HDS",verbose = True,**sorter_params)`
And the error is:

> job_kwargs = dict(n_jobs=1, chunk_duration="1s", progress_bar=True)
sorter_params = job_kwargs
sorting_HDS = ss.run_sorter('hdsort', recording_sub, output_folder=base_folder /  "results_HDS",verbose = True,**sorter_params)
AttributeError                            Traceback (most recent call last)
Cell In[51], line 3
      1 job_kwargs = dict(n_jobs=1, chunk_duration="1s", progress_bar=True)
      2 sorter_params = job_kwargs
----> 3 sorting_HDS = ss.run_sorter('hdsort', recording_sub, output_folder=base_folder /  "results_HDS",verbose = True,**sorter_params)
File /imec/other/macaw/chen14/anaconda43/envs/si96_pure/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:143, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    136             container_image = singularity_image
    137     return run_sorter_container(
    138         container_image=container_image,
    139         mode=mode,
    140         **common_kwargs,
    141     )
--> 143 return run_sorter_local(**common_kwargs)
File /imec/other/macaw/chen14/anaconda43/envs/si96_pure/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:160, in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
    157 # only classmethod call not instance (stateless at instance level but state is in folder)
    158 output_folder = SorterClass.initialize_folder(
    159     recording, output_folder, verbose, remove_existing_folder)
--> 160 SorterClass.set_params_to_folder(
    161     recording, output_folder, sorter_params, verbose)
    162 SorterClass.setup_recording(recording, output_folder, verbose=verbose)
    163 SorterClass.run_from_folder(output_folder, raise_error, verbose)
File /imec/other/macaw/chen14/anaconda43/envs/si96_pure/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py:166, in BaseSorter.set_params_to_folder(cls, recording, output_folder, new_params, verbose)
    164         bad_params.append(p)
    165 if len(bad_params) > 0:
--> 166     raise AttributeError('Bad parameters: ' + str(bad_params))
    168 params.update(new_params)
    170 # custom check params
AttributeError: Bad parameters: ['n_jobs', 'chunk_duration', 'progress_bar']

alejoe91, 2023-02-17:
Parameters are sorter-specific, so not all sorters will accept the same parameters.

You can check which parameters are available with this function:
```
params = ss.get_default_sorter_params("hdsort") # or other sorter names
print(params)
```

Regarding HDsort, this is a small bug on our side. I'll push a fix. Note, however, that HDsort is not currently maintained and it works very poorly on Neuropixels data. I recommend not using it ;)

ZoeChen96, 2023-02-17:
wow! You reply so fast! I really appreciate your help!!!!!

closed 2023-02-17 by ZoeChen96