SpikeInterface/spikeinterface #1336: error during whitenning

maxjuv, 2023-02-17:
Hello, 
I was trying to add whitenning in my preprocess chain, but I got the following error:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\juventin\AppData\Local\Programs\Python\Python310\lib\multiprocessing\spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "C:\Users\juventin\AppData\Local\Programs\Python\Python310\lib\multiprocessing\spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
  File "D:\local_code_library\spike_sorters\spikeinterface\spikeinterface\core\base.py", line 370, in from_dict
    extractor = _load_extractor_from_dict(d)
  File "D:\local_code_library\spike_sorters\spikeinterface\spikeinterface\core\base.py", line 844, in _load_extractor_from_dict
    kwargs[k] = _load_extractor_from_dict(v)
  File "D:\local_code_library\spike_sorters\spikeinterface\spikeinterface\core\base.py", line 861, in _load_extractor_from_dict
    extractor = cls(**kwargs)
  File "D:\local_code_library\spike_sorters\spikeinterface\spikeinterface\preprocessing\whiten.py", line 58, in __init__
    rec_segment = WhitenRecordingSegment(parent_segment, W, M, dtype_)
UnboundLocalError: local variable 'M' referenced before assignment

The issue is associated to the following code:
the recording is a SpikeGLXRecordingExtractor: 384 channels - 1 segments - 30.0kHz - 608.451s

`

    recording = spikeinterface.preprocessing.bandpass_filter(recording)
    recording = spikeinterface.preprocessing.phase_shift(recording)
    recording = spikeinterface.preprocessing.whiten(recording)
    recording = spikeinterface.preprocessing.common_reference(recording, reference='global', operator='median')

    peaks = detect_peaks(recording, method='locally_exclusive', **job_kwargs)

`
Surprisingly, if the replace the detect_peaks by a recording.get_traces(), I don't get the error.

alejoe91, 2023-02-17:
Hi @maxjuv 

Are you using the latest 0.97 version? Let me check!

Cheers
Alessio

maxjuv, 2023-02-17:
Hi Alessio,Yes, I recently update to  0.98.0.dev0 From: Alessio BuccinoSent: vendredi 17 février 2023 15:33To: SpikeInterface/spikeinterfaceCc: maxjuv; MentionSubject: Re: [SpikeInterface/spikeinterface] error during whitenning (Issue #1336) Hi @maxjuvAre you using the latest 0.97 version? Let me check!CheersAlessio

alejoe91, 2023-02-17:
I thought we fixed it, but apparently we forgot to push the fix...

Here is a PR: #1337

Can you test from there?

PR #1337 (2023-02-17): Fix whiten: add M to parameters and kwargs
@samuelgarcia I think we forgot to push this fix!

fixes #1336
closed 2023-02-17 by alejoe91