SpikeInterface/spikeinterface #1344: Move to src package structure

h-mayorquin, 2023-02-19:
We have some errors like #1316 and #1333 that would have not been an issue had we used a src structure:
https://blog.ionelmc.ro/2014/05/25/python-packaging/#the-structure

Plus this is the recommended way of the python packaking authority:
https://github.com/pypa/sampleproject

I honestly don't know any downside of using that strcuture other than aestethical concerns that it looks bad.

What do you guys think of making this change?

samuelgarcia, 2023-02-20:
We already discuss this when refactoring to pyproject.toml.
I do not like it but I admit that it avoid many beginners installation errors.

alejoe91, 2023-02-20:
@h-mayorquin after discussing with @samuelgarcia we agree that this should be the way to go!

h-mayorquin, 2023-02-27:
Great, I am thinking that we can do this after #1354. As that is a big change of its own, what do you think?

samuelgarcia, 2023-02-28:
Can we wait a bit more ?

h-mayorquin, 2023-02-28:
Yes, when do you think is a good point @samuelgarcia?  Do you have some PRs on the line that would be disturbed by this? 

This needs to be done quick as it will stop all the repo on the rails.  Maybe we can agree to do it at the end of weekly meeting. I prepare everything, we quickly checking everything together and we merge. I reckon it will take around 15 minutes of our time.

h-mayorquin, 2023-03-07:
I think this is another error that was generated by the workflow of importing from the base #1388.

alejoe91, 2023-03-30:
should also fix #1465

h-mayorquin, 2023-04-04:
We are moving forward with this. I will keep a list of the PRs that have to be merged:
* #1477 changes in docs. Nothing to be done.
* #1475 this is merged in #1478
*  #1471 change in docs. Nothing to be done.

h-mayorquin, 2023-04-05:
This is done

closed 2023-04-05 by h-mayorquin