SpikeInterface/spikeinterface #1345: 'Object of type CommonReferenceRecording is not JSON serializable' when saving recording

LaurenzMuessig, 2023-02-20:
Hi,

After the latest update, when I try to do _recording.save(...)_ for a recording that had been filtered, phase corrected, CAR'd and motion corrected, I now get this error when trying to write it to disk. Is this a bug or am I doing  something wrong? Thanks for your help.

Cheers

Laurenz

alejoe91, 2023-02-20:
Hi @LaurenzMuessig 

I think that the motion correction is not (yet) serializable. @samuelgarcia can you confirm?

LaurenzMuessig, 2023-02-20:
Hey @alejoe91 . I think before the update writing a motion-corrected recording to disk worked fine for me. 

Cheers

alejoe91, 2023-02-20:
Can you paste the entire error?

LaurenzMuessig, 2023-02-20:
Here you go - thanks for looking into this:

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Cell In [13], line 1
----> 1 rec_Fin.save(folder=os.path.join(outDir,'raw'),n_jobs=-1,chunk_duration='1s',verbose=True,progress_bar=True)
      2     # sort

File ~/spikeInterface/lib/python3.8/site-packages/spikeinterface-0.98.0.dev0-py3.8.egg/spikeinterface/core/base.py:620, in BaseExtractor.save(self, **kwargs)
    618     loaded_extractor = self.save_to_zarr(**kwargs)
    619 else:
--> 620     loaded_extractor = self.save_to_folder(**kwargs)
    621 return loaded_extractor

File ~/spikeInterface/lib/python3.8/site-packages/spikeinterface-0.98.0.dev0-py3.8.egg/spikeinterface/core/base.py:689, in BaseExtractor.save_to_folder(self, name, folder, verbose, **save_kwargs)
    687 provenance_file = folder / f'provenance.json'
    688 if self.check_if_dumpable():
--> 689     self.dump(provenance_file)
    690 else:
    691     provenance_file.write_text(
    692         json.dumps({'warning': 'the provenace is not dumpable!!!'}),
    693         encoding='utf8'
    694     )

File ~/spikeInterface/lib/python3.8/site-packages/spikeinterface-0.98.0.dev0-py3.8.egg/spikeinterface/core/base.py:458, in BaseExtractor.dump(self, file_path, relative_to, folder_metadata)
    447 """
    448 Dumps extractor to json or pickle
    449 
   (...)
    455     If not None, file_paths are serialized relative to this path
    456 """
    457 if str(file_path).endswith('.json'):
--> 458     self.dump_to_json(file_path, relative_to=relative_to, folder_metadata=folder_metadata)
    459 elif str(file_path).endswith('.pkl') or str(file_path).endswith('.pickle'):
    460     self.dump_to_pickle(file_path, relative_to=relative_to, folder_metadata=folder_metadata)

File ~/spikeInterface/lib/python3.8/site-packages/spikeinterface-0.98.0.dev0-py3.8.egg/spikeinterface/core/base.py:483, in BaseExtractor.dump_to_json(self, file_path, relative_to, folder_metadata)
    477 dump_dict = self.to_dict(include_annotations=True,
    478                          include_properties=False,
    479                          relative_to=relative_to,
    480                          folder_metadata=folder_metadata)
    481 file_path = self._get_file_path(file_path, ['.json'])
    482 file_path.write_text(
--> 483     json.dumps(check_json(dump_dict), indent=4),
    484     encoding='utf8'
    485 )

File /usr/lib/python3.8/json/__init__.py:234, in dumps(obj, skipkeys, ensure_ascii, check_circular, allow_nan, cls, indent, separators, default, sort_keys, **kw)
    232 if cls is None:
    233     cls = JSONEncoder
--> 234 return cls(
    235     skipkeys=skipkeys, ensure_ascii=ensure_ascii,
    236     check_circular=check_circular, allow_nan=allow_nan, indent=indent,
    237     separators=separators, default=default, sort_keys=sort_keys,
    238     **kw).encode(obj)

File /usr/lib/python3.8/json/encoder.py:201, in JSONEncoder.encode(self, o)
    199 chunks = self.iterencode(o, _one_shot=True)
    200 if not isinstance(chunks, (list, tuple)):
--> 201     chunks = list(chunks)
    202 return ''.join(chunks)

File /usr/lib/python3.8/json/encoder.py:431, in _make_iterencode.<locals>._iterencode(o, _current_indent_level)
    429     yield from _iterencode_list(o, _current_indent_level)
    430 elif isinstance(o, dict):
--> 431     yield from _iterencode_dict(o, _current_indent_level)
    432 else:
    433     if markers is not None:

File /usr/lib/python3.8/json/encoder.py:405, in _make_iterencode.<locals>._iterencode_dict(dct, _current_indent_level)
    403         else:
    404             chunks = _iterencode(value, _current_indent_level)
--> 405         yield from chunks
    406 if newline_indent is not None:
    407     _current_indent_level -= 1

File /usr/lib/python3.8/json/encoder.py:405, in _make_iterencode.<locals>._iterencode_dict(dct, _current_indent_level)
    403         else:
    404             chunks = _iterencode(value, _current_indent_level)
--> 405         yield from chunks
    406 if newline_indent is not None:
    407     _current_indent_level -= 1

File /usr/lib/python3.8/json/encoder.py:438, in _make_iterencode.<locals>._iterencode(o, _current_indent_level)
    436         raise ValueError("Circular reference detected")
    437     markers[markerid] = o
--> 438 o = _default(o)
    439 yield from _iterencode(o, _current_indent_level)
    440 if markers is not None:

File /usr/lib/python3.8/json/encoder.py:179, in JSONEncoder.default(self, o)
    160 def default(self, o):
    161     """Implement this method in a subclass such that it returns
    162     a serializable object for ``o``, or calls the base implementation
    163     (to raise a ``TypeError``).
   (...)
    177 
    178     """
--> 179     raise TypeError(f'Object of type {o.__class__.__name__} '
    180                     f'is not JSON serializable')

TypeError: Object of type CommonReferenceRecording is not JSON serializable

alejoe91, 2023-02-22:
@LaurenzMuessig 

Can you also share the code until the line that triggers the error?

LaurenzMuessig, 2023-02-22:
Hey @alejoe91 ;

Here you go - let me know if you need anything else:

import spikeinterface.preprocessing as pp
import spikeinterface.sorters as ss
import spikeinterface.extractors as se
from spikeinterface.sortingcomponents import peak_detection, peak_selection, peak_localization, peak_pipeline, motion_correction, motion_estimation
from spikeinterface.core import write_binary_recording, concatenate_recordings, set_global_job_kwargs
#
import subprocess, shlex, os, shutil, datetime
#
global_job_kwargs = dict(n_jobs=-1, chunk_duration="1s")
set_global_job_kwargs(**global_job_kwargs)
#
recording_list = [list of some trials recorded with spikeGLX]
rec = concatenate_recordings(recording_list)
#gInd = exp['trialName'][0].find('_g')
#fNameOut = os.path.join(outDir,exp['trialName'][0][:gInd] + '_procData_g0_t0.imec' + str(params['probeN']) + '.ap.bin')
#pre-processing
rec_filt = pp.bandpass_filter(rec,freq_min=300,freq_max=7000)
rec_filt = pp.phase_shift(rec_filt)
rec_filt = pp.common_reference(rec_filt)
# motion correction
peaks    = peak_detection.detect_peaks(rec_filt,method='by_channel',verbose=True,progress_bar=True)
nPeaks   = int(params['FractPeaks'] * peaks.shape[0]) # params['FractPeaks']=0.1
selPeaks = peak_selection.select_peaks(peaks, method='uniform', seed=None, n_peaks=nPeaks)
peakLoc  = peak_localization.localize_peaks(rec_filt,selPeaks,method='monopolar_triangulation',verbose=True,progress_bar=True)
rec_Fin  = motion_correction.CorrectMotionRecording(rec_filt,motion,tempBins,spatBins,border_mode='force_extrapolate',)                                            
rec_Fin .save(folder=os.path.join(outDir,'raw'),verbose=True,progress_bar=True)

alejoe91, 2023-02-22:
Thanks! Can I ask you to try with a single recording (without the `concatenate`)? 

Maybe that's what's causing the issue!

LaurenzMuessig, 2023-02-22:
Hi @alejoe91,

The same error occurs if I just use 1 file instead of a concatenated one. 
Also note the error says 'CommonReferenceRecording' even though _print(recording)_ returns 'CorrectMotionRecording'. I can save the 'CommonReferenceRecording' (before motion correction) with no issues. 

Cheers

Laurenz

alejoe91, 2023-02-22:
That's helpful! We'll try to reproduce the issue and understand what's going on!

LaurenzMuessig, 2023-02-22:
Thanks so much for your help with this.

JonathanAMichaels, 2023-02-24:
I'm seeing essentially the same error on the master branch, but instead "Object of type BinaryFolderRecording is not JSON serializable"

preprocess_folder has spikeinterface generated preprocessed data.

reproduced by:
`rec_preprocessed = si.load_extractor(preprocess_folder)

rec_corrected = CorrectMotionRecording(rec_preprocessed, motion, temporal_bins, spatial_bins,
                                            spatial_interpolation_method='idw',
                                            border_mode='remove_channels')

rec_corrected.save(folder=corrected_folder, format='binary', **job_kwargs)
`

samuelgarcia, 2023-02-27:
This have been borken by this #1325.
I will make a fix soon.

h-mayorquin, 2023-03-22:
This was fixed first by Sam fix in #1367 and now we have done the change as well in #1397 we  should be a permanent solution.