SpikeInterface/spikeinterface #1347: How to install Kilosort2

ZoeChen96, 2023-02-20:
Hi,

Sorry for troubling again for the very beginner's question...Last time I installed the IronClust correctly, but I encountered questions when installing the Kilosort. I tried Kilosort2, and I know it requires GPU/CUDA. In the server I used, I checked `nvidia-smi`, and it shows:

> Mon Feb 20 12:35:28 2023       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 455.32.00    Driver Version: 455.32.00    CUDA Version: 11.1     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  Tesla K80           Off  | 00000000:0A:00.0 Off |                    0 |
| N/A   27C    P8    27W / 149W |     18MiB / 11441MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   1  Tesla K80           Off  | 00000000:0B:00.0 Off |                    0 |
| N/A   24C    P8    28W / 149W |      0MiB / 11441MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   2  Tesla K80           Off  | 00000000:86:00.0 Off |                    0 |
| N/A   29C    P8    26W / 149W |      0MiB / 11441MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   3  Tesla K80           Off  | 00000000:87:00.0 Off |                    0 |
| N/A   23C    P8    33W / 149W |      0MiB / 11441MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      3581      G   /usr/bin/Xorg                      18MiB |
|    1   N/A  N/A      3581      G   /usr/bin/Xorg                       0MiB |
|    2   N/A  N/A      3581      G   /usr/bin/Xorg                       0MiB |
|    3   N/A  N/A      3581      G   /usr/bin/Xorg                       0MiB |
+-----------------------------------------------------------------------------+

So I assume the CUDA 11.1 is installed. The Matlab I used is R2020a. I gitclone the Kilosort and set path as is suggested by your document in [https://spikeinterface.readthedocs.io/en/latest/install_sorters.html#kilosort2](url), which previously works for Ironclust, but I can't find the installed Kilosort2:
`ss.installed_sorters()`

> ['herdingspikes', 'ironclust', 'spykingcircus2', 'tridesclous', 'tridesclous2']

I also tried to check the Kilosort webpage, and I found I can't run through the `mexGPUall.m` file. I think that's maybe the reason why I did not installed the Kilosort2 correctly, but I am not sure and don't know the direction for solution. Do you have any suggestions?

Regards,
Zoe

alejoe91, 2023-02-20:
Hi Zoe, 

You need to be able to run `mexGPUall.m` in order to be able to run KS..you can also check the option of running a spike sorter in a container: see docs [here](https://spikeinterface.readthedocs.io/en/latest/modules/sorters.html#running-sorters-in-container-docker-singularity)

ZoeChen96, 2023-02-20:
Hi Alessio, 
Thanks for the reply! Then I had better go to the Kilosort repository first.

closed 2023-02-20 by ZoeChen96