SpikeInterface/spikeinterface #1349: `Black` + `Isort` + `flake8` support

caniko, 2023-02-20:
## black + isort
Code formatters that will run on commit, the commit will be blocked if there are logical errors in any of the files. This means the code would fail to run anyway

## flake8
Checks if the code is formatted in accordance with rules set on `pyproject.toml`. Will block the PR and/or CI/CD if there are rule violations. We can pick and choose what rules to ignore.

## Solution
Create a new branch `black+isort+flake8` for merging the PR (#1342), and keep merging old PRs into the master. Thanks to @CodyCBakerPhD for the input. Merge `black+isort+flake8` into `master` when there are no PRs with the old format left.

### PR challenges
TLDR: New PRs must use `black+isort+flake8` as reference. Old PRs must merge into master ASAP.

Make sure new PRs are developed on `black+isort+flake8` I would mention this in `CONTRIBUTE.md`; failing to do this, you will need to copy over the `pyproject.toml` and `.pre-commit-config.yaml` into your own PR branch, and run the formatter yourself (`pre-commit run --all-files`) just note that there is one file on examples that have weird characters that need to be removed from its comments.

caniko, 2023-02-20:
[List of rules](https://flake8.pycqa.org/en/latest/user/error-codes.html) for flake8, please pick before proceeding.

alejoe91, 2023-02-22:
Moving converatsion from #1342 here:

@CodyCBakerPhD: What about a manual workflow or a cron job that runs `black` and `isort` on master (main)? I think this would make everyone happy! So open PRs don't need to update all the time due to pre-commit CI and the main branch will (almost) always be clean.

caniko, 2023-02-22:
@alejoe91 pre-commit is an optional module. It is only run if you run `pre-commit install` after `git clone`.

CodyCBakerPhD, 2023-02-26:
> What about a manual workflow or a cron job that runs black and isort on master (main)? I think this would make everyone happy! So open PRs don't need to update all the time due to pre-commit CI and the main branch will (almost) always be clean.

I've done that before, will leave code below

One final point I will make before doing that is have formatted code prior to the time of review itself saves a lot of dev time. I can't even count the wasted hours over the years from myself and people reviewing my code across different repos telling me to re-order my imports and reformat various things. `pre-commit` is very fast and easy to setup when needed and is hardly even noticeable thereafter; the bot, even more so.

```YAML
name: Run pre-commit as a workflow

on:
  pull_request:  # change to schedule or push to main if you want

jobs:
  pre-commit:
    runs-on: ubuntu-latest
    steps:
    - uses: actions/checkout@v3
      with:
        ref: ${{ github.head_ref }}
        persist-credentials: false # otherwise, the token used is the GITHUB_TOKEN, instead of your personal access token.
        fetch-depth: 0 # otherwise, there would be errors pushing refs to the destination repository.
    - name: Set cache hash
      run: echo "PY=$(python -VV | sha256sum | cut -d' ' -f1)" >> $GITHUB_ENV
    - name: Cache pre-commit
      id: cache-pre-commit
      uses: actions/cache@v1
      with:
        path: ~/.cache/pre-commit
        key: pre-commit|${{ env.PY }}|${{ hashFiles('.pre-commit-config.yaml') }}
    - if: steps.cache-pre-commit.outputs.cache-hit == false
      name: Install pre-commit
      run: |
        pip install pre-commit
        pre-commit install

    - name: Run pre-commit
      run: pre-commit run --all-files || true # `|| true` to not fail when something requires changing
    - name: Commit files
      run: | # `|| true` below to skip if nothing changed from pre-commit
        git config --local user.email "github-actions[bot]@users.noreply.github.com"
        git config --local user.name "github-actions[bot]"
        git commit -m "pre-commit" -a || true

    - name: Push changes
      uses: ad-m/github-push-action@v0.6.0
      with:
        github_token: ${{ secrets.API_TOKEN_GITHUB }}  # this needs to be set as a secret
        branch: ${{ github.head_ref }}
        force: true
```

h-mayorquin, 2023-03-01:
@samuelgarcia Would the solution provided about satisfy you as a compromise?

bendichter, 2023-03-01:
My two cents: I think these tools are very helpful for large collaborative projects. They have been very helpful for CN, and I know other groups such as DataJoint that have seen a lot of success with them. They improve the aesthetics of the code, optimize changed lines in PRs, and alleviate the need for the programmer to worry about style at all since it will be automatically fixed by GitHub.

The one downside I've seen is that when you first start using these tools they tend to create merge conflicts with existing PRs that can be a pain to attend to. For that reason, I think it could be useful to consider the timing of when you would introduce these tools. Maybe right before or right after a release would be ideal, then there are fewer active PRs.

JoeZiminski, 2023-03-10:
I agree these tools are very useful, we use them at the SWC and they do help with reading, reviewing code. Furthermore I have had bugs caught with even a simple code checker like flake8. It is also very beautiful to see a whole repo consistently formatted to a single style. I agree on pre-commit as @CodyCBakerPhD  mentions it means everything is clean in the PR, it is also nice to have CI to check linting too in the PR.

I think it is worth mentioning, these tools (in particular black) are not without some initial overhead. Personally, I do not always like the way black formats code and at the start in particular it was annoying - although there are also immediate benefits as you never have to think about formatting again. After a couple of months I got used to it and the benefits far outweigh any costs.

I've not used black to format mathematical code before but I see many mathematical libraries use it (sympy, statsmodels, scikit-learn, pytorch). I would suggest  copying the linter settings from a repository like this, as they can be a little fiddly (particularly flake8), and avoids discussion on what to include in them.

I think there is definitely some overhead for beginners to learn, but practically it is not too bad as the CLI commands are very convenient (if they don't want to use pre-commit at first). It is also a nice introduction to pep8 and linting concepts.

alejoe91, 2023-04-06:
An update on this:

Yesterday we moved the to the main branch and the `src/spikeinterface` structure!

We set a date (May 10th) as the "official" reformatting date, to reformat the source code with `black` (line length 120) and set pre-commits (either vie pre-commit CI or via black actions directly).

We'll start with `black` because `isort` might require some additional testing since it could create circular imports in core.

h-mayorquin, 2023-04-06:
Thanks for starting and organizing the discussion @caniko !

JoeZiminski, 2023-04-06:
That's great! Not to throw another factor into the mix, but of all the linting I've used adding type hints, and type checking (with mypy or pyright) as to be the best in terms of catching bugs. It is incredibly useful when refactoring etc. I only started with them fairly recently but use them in all projects now for this reason, as feels like code is much safer and secure. At some point I would suggest (and be happy to help with) adding type hints to SI code. Am mentioning now in case it is useful to consider at the same time as the other reformatting.

alejoe91, 2023-06-12:
@caniko Black is done, also with pre-commit CI.

For isort we still want to make sure it doesn't mess up with some relative imports that we have throughout the code to avoid circular imports.

alejoe91, 2023-11-06:
Let's close this and in case open other issues/discussion specific to other linting tools :)

closed 2023-11-06 by alejoe91