SpikeInterface/spikeinterface #1350: plot_unit_templates() in spike Tutorial 0.96 seems not working

ZoeChen96, 2023-02-20:
Hi,
Sorry for troubling again, but I found the code 
`we = si.extract_waveforms(recording_saved, sorting, folder=base_folder / "waveforms", 
                          load_if_exists=False, overwrite=True, **job_kwargs)
w = sw.plot_unit_templates(we, radius_um=30, backend="ipywidgets")`
gives an error:

> ---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
Cell In[98], line 1
----> 1 w = sw.plot_unit_templates(we, radius_um=30, backend="ipywidgets")
File /imec/other/macaw/chen14/anaconda43/envs/si96_pure/lib/python3.9/site-packages/spikeinterface/widgets/base.py:116, in define_widget_function_from_class.<locals>.widget_func(*args, **kwargs)
    113 @copy_signature(widget_class)
    114 def widget_func(*args, **kwargs):
    115     W = widget_class(*args, **kwargs)
--> 116     W.do_plot(W.backend, **W.backend_kwargs)
    117     return W.plotter
File /imec/other/macaw/chen14/anaconda43/envs/si96_pure/lib/python3.9/site-packages/spikeinterface/widgets/base.py:52, in BaseWidget.do_plot(self, backend, **backend_kwargs)
     50 backend = self.check_backend(backend)
     51 plotter = self.possible_backends[backend]()
---> 52 self.check_backend_kwargs(plotter, backend, **backend_kwargs)
     53 plotter.do_plot(self.plot_data, **backend_kwargs)
     54 self.plotter = plotter
File /imec/other/macaw/chen14/anaconda43/envs/si96_pure/lib/python3.9/site-packages/spikeinterface/widgets/base.py:46, in BaseWidget.check_backend_kwargs(self, plotter, backend, **backend_kwargs)
     44 for k in backend_kwargs:
     45     if k not in plotter_kwargs:
---> 46         raise Exception(f"{k} is not a valid plot argument or backend keyword argument. "
     47                         f"Possible backend keyword arguments for {backend} are: {list(plotter_kwargs.keys())}")
Exception: radius_um is not a valid plot argument or backend keyword argument. Possible backend keyword arguments for ipywidgets are: ['width_cm', 'height_cm', 'display']

Does it mean the radius_um is not a parameter? 
After deleting the parameter, with the code:
`w = sw.plot_unit_templates(we, backend="ipywidgets")`
it returns:

> --------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
Cell In[102], line 1
----> 1 w = sw.plot_unit_templates(we, backend="ipywidgets")
File /imec/other/macaw/chen14/anaconda43/envs/si96_pure/lib/python3.9/site-packages/spikeinterface/widgets/base.py:116, in define_widget_function_from_class.<locals>.widget_func(*args, **kwargs)
    113 @copy_signature(widget_class)
    114 def widget_func(*args, **kwargs):
    115     W = widget_class(*args, **kwargs)
--> 116     W.do_plot(W.backend, **W.backend_kwargs)
    117     return W.plotter
File /imec/other/macaw/chen14/anaconda43/envs/si96_pure/lib/python3.9/site-packages/spikeinterface/widgets/base.py:53, in BaseWidget.do_plot(self, backend, **backend_kwargs)
     51 plotter = self.possible_backends[backend]()
     52 self.check_backend_kwargs(plotter, backend, **backend_kwargs)
---> 53 plotter.do_plot(self.plot_data, **backend_kwargs)
     54 self.plotter = plotter
File /imec/other/macaw/chen14/anaconda43/envs/si96_pure/lib/python3.9/site-packages/spikeinterface/widgets/ipywidgets/unit_templates.py:8, in UnitTemplatesPlotter.do_plot(self, data_plot, **backend_kwargs)
      7 def do_plot(self, data_plot, **backend_kwargs):
----> 8     super().do_plot(data_plot, **backend_kwargs)
      9     self.controller["plot_templates"].layout.visibility = 'hidden'
File /imec/other/macaw/chen14/anaconda43/envs/si96_pure/lib/python3.9/site-packages/spikeinterface/widgets/ipywidgets/unit_waveforms.py:84, in UnitWaveformPlotter.do_plot(self, data_plot, **backend_kwargs)
     75 self.widget = widgets.AppLayout(
     76     center=fig_wf.canvas,
     77     left_sidebar=unit_widget,
   (...)
     80     footer=footer
     81 )
     83 # a first update
---> 84 self.updater(None)
     86 if backend_kwargs["display"]:
     87     display(self.widget)
File /imec/other/macaw/chen14/anaconda43/envs/si96_pure/lib/python3.9/site-packages/spikeinterface/widgets/ipywidgets/unit_waveforms.py:150, in PlotUpdater.__call__(self, change)
    147 self.ax_probe.axis("equal")
    149 for unit in unit_ids:
--> 150     self.ax_probe.plot(channel_locations[self.next_data_plot['channel_inds'][unit], 0],
    151                        channel_locations[self.next_data_plot['channel_inds'][unit], 1],
    152                        ls="", marker="o", markersize=3,
    153                        color=self.next_data_plot['unit_colors'][unit])
    154 self.ax_probe.set_xlim(np.min(channel_locations[:, 0])-10, np.max(channel_locations[:, 0])+10)
    155 fig_probe = self.ax_probe.get_figure()
KeyError: 'channel_inds'

alejoe91, 2023-02-20:
Thanks for the report! Indeed it might be a bug due to a recent refactoring of the way we handle sparsity. We'll check!

alejoe91, 2023-02-21:
@ZoeChen96 fixed here: #1353 

Can you test? It works now on my side!

ZoeChen96, 2023-02-21:
Hi, I am happy to test!
I just pip upgrade it, but the version is still 0.97.0. And the problem still exists:
[image]

alejoe91, 2023-02-21:
You need to install from source. You can check the installation page on the docs (the change will be available once we merge the PR)

ZoeChen96, 2023-02-21:
> You need to install from source. You can check the installation page on the docs (the change will be available once we merge the PR)

Hi! I just tried, and the current version is:
[image]
 but I found I can't import the modules properly. The code below will give the error:
`import spikeinterface as si
import spikeinterface.extractors as se 
import spikeinterface.preprocessing as spre
import spikeinterface.sorters as ss
import spikeinterface.postprocessing as spost
import spikeinterface.qualitymetrics as sqm
import spikeinterface.comparison as sc
import spikeinterface.exporters as sexp
import spikeinterface.widgets as sw`

The error is:

> ---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
Cell In[2], line 2
      1 import spikeinterface as si
----> 2 import spikeinterface.extractors as se 
      3 import spikeinterface.preprocessing as spre
      4 import spikeinterface.sorters as ss
File /imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/spikeinterface/spikeinterface/extractors/__init__.py:1
----> 1 from .extractorlist import *
      3 from .toy_example import toy_example
      4 from .bids import read_bids
File /imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/spikeinterface/spikeinterface/extractors/extractorlist.py:10
      4 from spikeinterface.core import (BaseRecording, BaseSorting,
      5                                  BinaryRecordingExtractor, NumpyRecording,
      6                                  NpzSortingExtractor, NumpySorting,
      7                                  NpySnippetsExtractor)
      9 # sorting/recording/event from neo
---> 10 from .neoextractors import *
     12 # non-NEO objects implemented in neo folder
     13 from .neoextractors import NeuroScopeSortingExtractor, MaxwellEventExtractor
File /imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/spikeinterface/spikeinterface/extractors/neoextractors/__init__.py:15
     11 from .edf import (EDFRecordingExtractor, 
     12                   read_edf)
     13 from .intan import (IntanRecordingExtractor, 
     14                     read_intan)
---> 15 from .maxwell import (MaxwellRecordingExtractor, MaxwellEventExtractor, 
     16                       read_maxwell, read_maxwell_event)
     17 from .mearec import (MEArecRecordingExtractor, MEArecSortingExtractor,
     18                      read_mearec)
     19 from .mcsraw import (MCSRawRecordingExtractor, 
     20                      read_mcsraw)
File /imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/spikeinterface/spikeinterface/extractors/neoextractors/maxwell.py:5
      1 import numpy as np
      3 import probeinterface as pi
----> 5 from spikeinterface import BaseEvent, BaseEventSegment
      6 from spikeinterface.core.core_tools import define_function_from_class
      8 from .neobaseextractor import NeoBaseRecordingExtractor
ImportError: cannot import name 'BaseEvent' from 'spikeinterface' (unknown location)

ZoeChen96, 2023-02-21:
Besides, I found the pip show spikeinterface gives different results. In Linux command, it shows `Version: 0.98.0.dev0`, but after open jupyter, it changes to 0.97.0
[image]

ZoeChen96, 2023-02-21:
> Besides, I found the pip show spikeinterface gives different results. In Linux command, it shows `Version: 0.98.0.dev0`, but after open jupyter, it changes to 0.97.0 [image]

I am not sure whether I have the 0.98.0.dev0 correctly...

closed 2023-02-22 by samuelgarcia