SpikeInterface/spikeinterface #1362: Thread safety in `IblStreamingRecordingExtractor`

cwindolf, 2023-02-24:
Thanks for the awesome `IblStreamingRecordingExtractor`!

I ran into an issue while using it on multiple processes: the underlying `brainbox.io.spikeglx.Streamer` objects on each process will interfere with each others cache files. Even if they read traces from disjoint chunks, they will delete each others' cache files due (I think) to [this call to `rmtree`](https://github.com/int-brain-lab/ibllib/blob/master/brainbox/io/spikeglx.py#L152).

I'm not sure if this should be a spikeinterface or a brainbox issue. I'm working around it by patching the dictionary representation of the extractor on each worker process to specify a different cache dir per worker before loading the process-local extractors with `BaseRecording.from_dict`.

CodyCBakerPhD, 2023-02-24:
> **the extractor on each worker process to specify a different cache dir per worker**

That is indeed my suggested way of implementing this, and it is also how I use it when parallelizing as well. All-in-all it's the easiest solution that you can implement right away

That said, I don't know how I could make this happen automatically on our side since the recording extractors on different processes would not usually be able to directly communicate with each other without a more sophisticated setup involving some kind of shared memory

I think the better long-term solution might be for brainbox to send each chunk to a different subfolder so that cleanup of a single chunk wouldn't affect the others

samuelgarcia, 2023-02-27:
Thank you for this discussion.
@oliche do you have ideas that could be implented on brainbox side to avoid this ?

alejoe91, 2023-11-06:
@cwindolf is this still an issue?

zm711, 2024-12-05:
I'm following up one more time a year later :P does this need to stay open or can we close all?

cwindolf, 2024-12-05:
I haven't used this in the year and a half since, but the @CodyCBakerPhD identified should have fixed the problem! I think it's fine to close this.

closed 2024-12-05 by zm711