SpikeInterface/spikeinterface #1364: Problem

ramon-92, 2023-02-25:
spikeinterface 0.97.0
Ubuntu 18.04.6
python 3.8.5

Hi SpikeInterface Community,

I am new to spikeinterface, so please bear with me. I am trying to analyze data recorded with a Ripple Neuro system. Since it seems that Ripple's file format is not supported yet (are there plans to add?), I am converting the data to a .mat file and then reading it in python using scipy.io and then initializing a recording instance using 

`recording=si.core.NumpyRecording(traces_list=tracesList,sampling_frequency=fs_raw, channel_ids=channelIDs)`

with
`fs_raw = 30000Hz`
`channel_ids=[0,1,2]`

I do not have any probe interface information. 

I then follow one of the official tutorials (NeuroDataReHack (4th Oct 2022)). I have no issues visualizing and preprocessing the data. 

However, trying to safe the preprocessed data using 
`job_kwargs = dict(n_jobs=1, chunk_duration="1s", progress_bar=True)
recording_saved = recording_f.save(folder=base_folder / "preprocessed-2", **job_kwargs)`

Throws the following error:
`[.....]`
`~/anaconda3/lib/python3.8/site-packages/spikeinterface/core/core_tools.py in check_json(d)
    106             dc[k] = v.isoformat()
    107         elif isinstance(v, (np.ndarray, list)):
--> 108             if len(v) > 0:
    109                 if isinstance(v[0], dict):
    110                     # these must be extractors for multi extractors`

`TypeError: len() of unsized object`

I am using n_jobs=1 because if I use a greater number, i get the following error: 
`RuntimeError: Recording is not dumpable and can't be processed in parallel. You can use the `recording.save()` function to make it dumpable or set 'n_jobs' to 1.`

When I directly continue to the analysis part (instead of saving the recording object first), and choose one of the two sorters that are part of spikeinterface (e.g. spykingcircus2)

`sorting_spykingcircus = ss.run_sorter('spykingcircus2', 
                                      recording_f, 
                                      output_folder=base_folder/'results_spykingcircus2', 
                                      verbose=True, 
                                      docker_image=False)`

I get the following error
`~/anaconda3/lib/python3.8/site-packages/spikeinterface/core/base.py in dump_to_json(self, file_path, relative_to, folder_metadata)
    474             If not None, file_paths are serialized relative to this path
    475         """
--> 476         assert self.check_if_dumpable()
    477         dump_dict = self.to_dict(include_annotations=True,
    478                                  include_properties=False,
AssertionError: `

When I am trying to use a docker image instead, e.g. choosing herdingspikes and setting docker_image=True (I have installed docker on my ubuntu machine), I get the following error: 

`~/anaconda3/lib/python3.8/site-packages/spikeinterface/core/core_tools.py in check_json(d)
    106             dc[k] = v.isoformat()
    107         elif isinstance(v, (np.ndarray, list)):
--> 108             if len(v) > 0:
    109                 if isinstance(v[0], dict):
    110                     # these must be extractors for multi extractors
TypeError: len() of unsized object`

Can anyone point me to the right direction? I am suspecting something is wrong with my file format? Or maybe there is some sort of information missing? Any advice would be highly appreciated! 

Many thanks!!

samuelgarcia, 2023-02-27:
For your first step, when you do numpy to binary you have to force `n_jobs=1` for the `save()`

alejoe91, 2023-02-27:
@ramon-92 

How are you setting `fs_raw`? Note that `sampling_frequency` needs to be a float, so if you used `30000Hz` this might cause the failure

ramon-92, 2023-02-27:
Thank you, @samuelgarcia and @alejoe91 !!

I can now save recordings to the binary format and load them without issues. I type casted the sampling frequency from `int` to `float` and indeed the initial errors for the sorting disappeared as well, however I am now getting an error that indicates missing probe information, in particular it is complaining about "channel_distances" and "channel_locations". Is there a way to manually input said information?

Since we are building our own probes, I can't use the probe_interface package the way it's shown in the tutorial. I am assuming I somehow have to set the characteristics of my probe in probe_interface? Or can I get away without setting any of those. In my recording the channels are only separated laterally by 50 micron and we are recording activity along a peripheral nerve with 3 channels.

Attached screenshots of the error (first one using a docker image, second one using the in-build tridesclous2 sorter).

Thank you!
[image]
[image]

ramon-92, 2023-03-06:
Thank you @alejoe91 !!

closed 2023-03-06 by ramon-92