SpikeInterface/spikeinterface #1372: Pykilosort running issue: AttributeError: 'Bunch' object has no attribute 'channel_labels'

ZoeChen96, 2023-02-27:
Hi there,

I tried to use pykilosort, but I encounter the Attribute Error. The code was like:
`import spikeinterface.extractors as se 
import spikeinterface.preprocessing as spre
import spikeinterface.sorters as ss
from pathlib import Path
import warnings
warnings.simplefilter("ignore")
base_folder = Path(".")
nwb_file_path = "sub-npI1_ses-20190413_behavior+ecephys.nwb"
recording_nwb = se.read_nwb_recording(file_path=nwb_file_path)
sorting_nwb = se.read_nwb_sorting(file_path=nwb_file_path)
recording_nwb.annotate(is_filtered=False)
recording_to_process = recording_nwb
recording_f = spre.bandpass_filter(recording_to_process, freq_min=300, freq_max=6000)
recording_cmr = spre.common_reference(recording_f, reference='global', operator='median')
fs = recording_cmr.get_sampling_frequency()
recording_sub = recording_cmr.frame_slice(start_frame=0*fs, end_frame=100*fs)
sorting_nwb_sub = sorting_nwb.frame_slice(start_frame=0*fs, end_frame=100*fs)
ss.IronClustSorter.set_ironclust_path('/imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/ironclust')
ss.Kilosort3Sorter.set_kilosort3_path('/imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/Kilosort')
ss.installed_sorters()
sorter_params = dict(n_jobs=1, chunk_duration="1s", progress_bar=True)
sorting_pyKS = ss.run_sorter('pykilosort', recording_sub, 
                            output_folder=base_folder / 'results_pyKS',
                            verbose=True, 
                            **sorter_params)
`

And the output is:

> Setting IRONCLUST_PATH environment variable for subprocess calls to: /imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/ironclust
Setting KILOSORT3_PATH environment variable for subprocess calls to: /imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/Kilosort
['ironclust', 'kilosort3', 'pykilosort', 'spykingcircus2', 'tridesclous2']

> write_binary_recording: 100%|###############################################################################################################################| 100/100 [02:16<00:00,  1.37s/it]
Auto-detection of noisy channels: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 25/25 [00:14<00:00,  1.76it/s]
Computing the whitening matrix: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2/2 [00:02<00:00,  1.45s/it]
Preprocessing: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 46/46 [00:17<00:00,  2.70it/s]
Extracting templates: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 10/10 [00:07<00:00,  1.29it/s]
Optimizing templates: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 92/92 [00:43<00:00,  2.12it/s]
Finding merges: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 313/313 [00:02<00:00, 109.80it/s]
Setting cutoff: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 359/359 [00:01<00:00, 257.28it/s]
Computing templates: 100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 357/357 [00:00<00:00, 1372.75it/s]
Saving template and PC features: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2/2 [00:00<00:00, 33.36it/s]
Error running pykilosort
Traceback (most recent call last):
  File "/imec/other/macaw/chen14/anaconda43/envs/pyks2/lib/python3.10/site-packages/spikeinterface/sorters/basesorter.py", line 226, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "/imec/other/macaw/chen14/anaconda43/envs/pyks2/lib/python3.10/site-packages/spikeinterface/sorters/pykilosort/pykilosort.py", line 190, in _run_from_folder
    run(
  File "/imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/pykilosort/pykilosort/main.py", line 268, in run
    rezToPhy(ctx, dat_path=dat_path, output_dir=output_dir)
  File "/imec/other/macaw/chen14/spikeinterface/Official_Tutorial_SI_0.96_Oct22/pykilosort/pykilosort/postprocess.py", line 1480, in rezToPhy
    _save('channel_labels', probe.channel_labels, np.int8)
AttributeError: 'Bunch' object has no attribute 'channel_labels'
---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
Cell In[6], line 2
      1 sorter_params = dict(n_jobs=1, chunk_duration="1s", progress_bar=True)
----> 2 sorting_pyKS = ss.run_sorter('pykilosort', recording_sub, 
      3                             output_folder=base_folder / 'results_pyKS',
      4                             verbose=True, 
      5                             **sorter_params)
File /imec/other/macaw/chen14/anaconda43/envs/pyks2/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py:143, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    136             container_image = singularity_image
    137     return run_sorter_container(
    138         container_image=container_image,
    139         mode=mode,
    140         **common_kwargs,
    141     )
--> 143 return run_sorter_local(**common_kwargs)
File /imec/other/macaw/chen14/anaconda43/envs/pyks2/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py:163, in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
    160 SorterClass.set_params_to_folder(
    161     recording, output_folder, sorter_params, verbose)
    162 SorterClass.setup_recording(recording, output_folder, verbose=verbose)
--> 163 SorterClass.run_from_folder(output_folder, raise_error, verbose)
    164 if with_output:
    165     sorting = SorterClass.get_result_from_folder(output_folder)
File /imec/other/macaw/chen14/anaconda43/envs/pyks2/lib/python3.10/site-packages/spikeinterface/sorters/basesorter.py:262, in BaseSorter.run_from_folder(cls, output_folder, raise_error, verbose)
    260 if has_error and raise_error:
    261     print(log['error_trace'])
--> 262     raise SpikeSortingError(
    263         f"Spike sorting failed. You can inspect the runtime trace in {output_folder}/spikeinterface_log.json")
    265 return run_time
SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in results_pyKS/spikeinterface_log.json

alejoe91, 2023-03-01:
Hi, maybe they changed something in pykilosort in the meanwhile. Can you try to install version 1.3.3 (https://github.com/int-brain-lab/pykilosort/tree/1.3.3/pykilosort)? If they made some changes in master, we need to update our wrapper. I recommend fixing the version or using the docker image

ZoeChen96, 2023-03-01:
Hi Alessio, thanks! I will give it a try.

alejoe91, 2023-03-20:
@ZoeChen96 did this work?

ZoeChen96, 2023-03-20:
Hi Alessio, sorry I work with kilosort3 so I forgot to check this... I tried but it shows the repository not found.
```
git clone https://github.com/int-brain-lab/pykilosort/tree/1.3.3/pykilosort
Cloning into 'pykilosort'...
fatal: repository 'https://github.com/int-brain-lab/pykilosort/tree/1.3.3/pykilosort/' not found
```

ZoeChen96, 2023-03-20:
Hi Alessio, sorry for the late response. I just tried, it works. Thank you!

closed 2023-03-20 by ZoeChen96

ZoeChen96, 2023-03-20:
PS: for those who wants to use pykilosort, the command for clone pykilosort 1.3.3 is:
`git clone -b 1.3.3 https://github.com/int-brain-lab/pykilosort`
and then I used these commands to install it:
```
conda env create -f pyks2.yml
conda activate pyks2
python setup.py install
```

alejoe91, 2023-03-20:
Great! This is probably an easy fix on our side for the new master version.

I'll make a PR soon if you have time to test it out ;)