SpikeInterface/spikeinterface #1377: AttributeError: module 'spikeinterface.full' has no attribute 'detect_bad_channels'

aeazraoui, 2023-03-01:
Hi everyone,

I am new to spikeinterface and I was just trying the basic notebook for analysing Neuropixel data and I got the following error if someone can help:

```

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-12-2c586acd3e69> in <module>
      1 rec1 = si.highpass_filter(raw_rec, freq_min=400.)
----> 2 bad_channel_ids, channel_labels = si.detect_bad_channels(rec1)
      3 rec2 = rec1.remove_channels(bad_channel_ids)
      4 print('bad_channel_ids', bad_channel_ids)
      5 

AttributeError: module 'spikeinterface.full' has no attribute 'detect_bad_channels'
```
Thanks in advance,
Best

alejoe91, 2023-03-01:
Hi,

Can you make sure you are using the most recent version?

aeazraoui, 2023-03-01:
Hi Alessio,

I cloned the git repository to make sure it is the last version, but I can
check again.

alejoe91, 2023-03-01:
@samuelgarcia can you take a look at this?

yger, 2023-03-01:
I think the function is now called remove_bad_channels, no?

alejoe91, 2023-03-01:
Nope: https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/preprocessing/detect_bad_channels.py#L10

alejoe91, 2023-03-01:
And it's correctly imported: https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/preprocessing/__init__.py#L4

aeazraoui, 2023-03-01:
I managed to install the latest version

```
Name: spikeinterface
Version: 0.98.0.dev0
Summary: Python toolkit for analysis, visualization, and comparison of spike sorting output
Home-page: 
Author: 
Author-email: Alessio Buccino <alessiop.buccino@gmail.com>, Samuel Garcia <sam.garcia.die@gmail.com>
License: 
Location: c:\users\anass\anaconda3\lib\site-packages\spikeinterface-0.98.0.dev0-py3.7.egg
Requires: joblib, neo, numpy, probeinterface, threadpoolctl, tqdm
Required-by: sortingview
```

But now I get this new error:

```
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-2-ad847fbb9c78> in <module>
----> 1 import spikeinterface.full as si
      2 
      3 import numpy as np
      4 import matplotlib.pyplot as plt
      5 from pathlib import Path

~\anaconda3\lib\site-packages\spikeinterface-0.98.0.dev0-py3.7.egg\spikeinterface\__init__.py in <module>
      3 
      4 """
----> 5 import importlib.metadata
      6 __version__ = importlib.metadata.version("spikeinterface")
      7 

ModuleNotFoundError: No module named 'importlib.metadata'
```

JoeZiminski, 2023-03-11:
Hi @aeazraoui, I think this issue is because you seem to be on Python 3.7 but spikeinterface is supported for >3.8 only. In particular, you seem to be running into [this issue](https://stackoverflow.com/questions/59216175/importerror-cannot-import-name-metadata-from-importlib). Try creating a new conda environment with using python 3.8, e.g.:

`conda create -n "myenv" python=3.3.0 ipython`