SpikeInterface/spikeinterface #1379: spikeinterface widgets

aeazraoui, 2023-03-01:
Hello everyone,

I am trying to run a basic notebook to familiarise with spiekinterface but it seems like the widgets do not work.
I run this code

```
# here we use static plot using matplotlib backend
fig, axs = plt.subplots(ncols=3, figsize=(20, 10))

si.plot_timeseries(rec1, backend='matplotlib',  clim=(-50, 50), ax=axs[0])
si.plot_timeseries(rec4, backend='matplotlib',  clim=(-50, 50), ax=axs[1])
si.plot_timeseries(rec, backend='matplotlib',  clim=(-50, 50), ax=axs[2])
for i, label in enumerate(('filter', 'cmr', 'final')):
    axs[i].set_title(label)
```

And it gives as an output this message and an empty widget

```
C:\Users\Anass\anaconda3\lib\site-packages\ipykernel\ipkernel.py:287: DeprecationWarning: `should_run_async` will not call `transform_cell` automatically in the future. Please pass the result to `transformed_cell` argument and any exception that happen during thetransform in `preprocessing_exc_tuple` in IPython 7.17 and above.
  and should_run_async(code)
C:\Users\Anass\anaconda3\lib\site-packages\ipywidgets\widgets\widget.py:502: DeprecationWarning: Passing unrecognized arguments to super(Toolbar).__init__().
__init__() missing 1 required positional argument: 'canvas'
This is deprecated in traitlets 4.2.This error will be raised in a future release of traitlets.
  super().__init__(**kwargs)
C:\Users\Anass\anaconda3\lib\site-packages\ipywidgets\widgets\widget.py:502: DeprecationWarning: Passing unrecognized arguments to super(Toolbar).__init__().
__init__() missing 1 required positional argument: 'canvas'
This is deprecated in traitlets 4.2.This error will be raised in a future release of traitlets.
  super().__init__(**kwargs)

```
[image]

Thank you in advance for the help.

Best,
Anass

alejoe91, 2023-03-01:
Hi Anass,

Can you try with the `%matplotlib inline` option? Do other plots work?

aeazraoui, 2023-03-01:
Yes Alessio, other plots work like this one:

[image]

alejoe91, 2023-03-27:
This is likely due to some dependency issue in the ipympl/matplotlib widget backend. I don't think this is related to SpikeInterface.

closed 2023-03-27 by alejoe91