SpikeInterface/spikeinterface #1380: Documentation Help

zm711, 2023-03-01:
I noticed a few minor words missing in the documentation. Just wanted to see if you wanted an extra pair of eyes in the readthedocs or whether you already had the documentation help you needed.

alejoe91, 2023-03-01:
Hi Zach,

Of course it would be great if we could borrow your extra pair of eyes :) 
Thanks for the suggestion!!

zm711, 2023-03-01:
One quick question for the documentation. It goes back and forth between spikeinterface and SpikeInterface. Which is the preferred way to refer to the package or does the capitalization change the meaning of what is specifically being referred to?

I'm happy to work on making it consistent in the docs if the group has a preference.

alejoe91, 2023-03-01:
Mmm that's a good question. In general the project tis called SpikeInterface, but maybe we can just leave that in the title/index page and use lower case in the actual modules documentation. Sounds fair?

h-mayorquin, 2023-03-06:
Thanks for your contributions @zm711. I am closing this issue. If anybody reads this in the future let it be know that contributions to documentation are highly welcomed and encouraged.

closed 2023-03-06 by h-mayorquin