SpikeInterface/spikeinterface #1382: Question: motion estimation parameters & results interpretation

bramn22, 2023-03-01:
Hi, 

I am trying to correct a sudden non-rigid motion shift between two consecutive recordings (caused by an IP injection in the head-fixed mouse). I am trying to use the motion estimation and correction workflows in spikeinterface, but I am having difficulties with (1) deciding how to set the estimation parameters, (2) interpreting the motion matrix, and (3) the appearance of new peaks after motion correction.
[image] *Original (red line=separation first and second recording)*
 [image] *Corrected*

I use the following function and parameters to estimate the motion:
``motion, temporal_bins, spatial_bins, extra_check = estimate_motion(rec_combined, peaks, peak_locations, direction='y', bin_duration_s=rec1_length, bin_um=10., rigid=False, win_shape='gaussian', win_step_um=50., win_sigma_um=150, progress_bar=False, output_extra_check=True, verbose=True)``
I set the bin_duration_s to be equal to the length of the first recording in order to estimate the shift between the first and second recordings.

1) Can you tell me what the difference is between the meaning of win_step_um and to bin_um. Would this be the stepsize between bins of size bin_um? Also, if the y-distance between channels is 20um (neuropixels 1), does it make sense for me to use bin_um=10?

2) Can you tell me what the values mean in the motion matrix that is returned from the estimate_motion? I expected to see a large motion value around the separation point between the first and second recording.
[image]

3) In the rasterplot after motion correction, new peaks are detected on the top and bottom of the recording. I am wondering whether this is an expected effect of the motion correction, or whether I missing something/doing something wrong.

Thank you,
Bram

alejoe91, 2023-06-12:
@samuelgarcia

alejoe91, 2023-11-06:
Pinging @cwindolf as the drift expert here!