SpikeInterface/spikeinterface #1388: Assertion error

aeazraoui, 2023-03-02:
I tried to run a sorter (after installing the docker) with the following code

```
# run kilosort2.5 without drift correction
params_kilosort2_5 = {'do_correction': False}

sorting = spikeinterface.full.run_sorter('kilosort2_5', r'C:\Users\PC_analyse_2022\Desktop\preprocess\traces_cached_seg0', output_folder=base_folder / 'kilosort2.5_output',
                        docker_image=True, verbose=True, **params_kilosort2_5)
```
And since then i can't import si, I run this code:
```
import spikeinterface.full as si

​

import numpy as np

import matplotlib.pyplot as plt

from pathlib import Path

```

And get the following error
```
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
Input In [4], in <cell line: 1>()
----> 1 import spikeinterface.full as si
      3 import numpy as np
      4 import matplotlib.pyplot as plt

File C:\ProgramData\Anaconda3\lib\site-packages\spikeinterface-0.98.0.dev0-py3.9.egg\spikeinterface\full.py:19, in <module>
     17 from .sorters import *
     18 from .preprocessing import *
---> 19 from .postprocessing import *
     20 from .qualitymetrics import *
     21 from .curation import *

File C:\ProgramData\Anaconda3\lib\site-packages\spikeinterface-0.98.0.dev0-py3.9.egg\spikeinterface\postprocessing\__init__.py:22, in <module>
     18 from .principal_component import WaveformPrincipalComponent, compute_principal_components
     20 from .spike_amplitudes import compute_spike_amplitudes, SpikeAmplitudesCalculator
---> 22 from .correlograms import (CorrelogramsCalculator,
     23                            compute_autocorrelogram_from_spiketrain,
     24                            compute_crosscorrelogram_from_spiketrain,
     25                            compute_correlograms, correlogram_for_one_segment,
     26                            compute_correlograms_numba, compute_correlograms_numpy)
     28 from .isi import (ISIHistogramsCalculator,
     29                   compute_isi_histograms_from_spiketrain, compute_isi_histograms,
     30                   compute_isi_histograms_numpy, compute_isi_histograms_numba)
     32 from .spike_locations import compute_spike_locations, SpikeLocationsCalculator

File C:\ProgramData\Anaconda3\lib\site-packages\spikeinterface-0.98.0.dev0-py3.9.egg\spikeinterface\postprocessing\correlograms.py:69, in <module>
     64     @staticmethod
     65     def get_extension_function():
     66         return compute_correlograms
---> 69 WaveformExtractor.register_extension(CorrelogramsCalculator)
     73 def _make_bins(sorting, window_ms, bin_ms):
     75     fs = sorting.get_sampling_frequency()

File C:\ProgramData\Anaconda3\lib\site-packages\spikeinterface-0.98.0.dev0-py3.9.egg\spikeinterface\core\waveform_extractor.py:345, in WaveformExtractor.register_extension(cls, extension_class)
    343 assert issubclass(extension_class, BaseWaveformExtractorExtension)
    344 assert extension_class.extension_name is not None, 'extension_name must not be None'
--> 345 assert all(extension_class.extension_name != ext.extension_name for ext in cls.extensions), \
    346     'Extension name already exists'
    347 cls.extensions.append(extension_class)

AssertionError: Extension name already exists

```
I tried with the latest version of si and still the same issue, besides it was working fine

alejoe91, 2023-03-07:
Can you try to run the script from a different folder?

Also uninstalling/re-installing might fix it

aeazraoui, 2023-03-07:
> Ahhh I see what might be causing the problem!
>
> You are doing this: spikeinterface.full.do_something
>
> The full is not a real module, but just an import strategy!
>
> You should do, instead:
>
> import spikeinterface.full as si
>
> si.do_something

I tried it too without a positive result. I recently tried to
uninstall/reinstall anaconda and create a virtual env with spike interface
and it works so far.

Thank you for your help!

alejoe91, 2023-03-07:
No worries, closing the issue then!

closed 2023-03-07 by alejoe91