SpikeInterface/spikeinterface #1389: Can we merge the output of several sorters ?

aeazraoui, 2023-03-02:
Hi,

I was wondering if we can merge the output from several sorters to get the units agreed on (counted not as duplicates) and those not agreed on ? So that we can curate this merged output and take advantage of the different sorters which can miss some units found by others.

Thank you in advance,

alejoe91, 2023-03-07:
Hi,

Yes this is definitely possible! 

See documentation here: https://spikeinterface.readthedocs.io/en/latest/modules/comparison.html#compare-the-output-of-multiple-spike-sorters

aeazraoui, 2023-03-07:
Hi Alessio,

I saw that this was to compare outputs, but can we combine them, remove
duplicates and export in one phy output ?

Thank you in advance

alejoe91, 2023-03-07:
To combine them you have to use the `get_agreement_sorting()` function.
You also need to `save()` the sorting object to allow multiprocessing. Then you can continue the analysis by extracting waveforms, postprocess, and epxort to phy:

```
sorting_agreement = mcmp.get_agreement_sorting(minimum_agreement_count=3)
print('Units in agreement for all three sorters: ', sorting_agreement.get_unit_ids())

sorting_agreement = sorting_agreement.save(folder="agreement")

we_agreement = si.extract_waveforms(recording, sorting_agreement, ...)
# more postprocessing
si.export_to_phy(we_agreement, ...)
```

aeazraoui, 2023-03-07:
Thanks a lot for the help Alessio ! So if I understand correctly this will
take only the units in agreement right ? What if I want to add also those
which are not in agreement?

alejoe91, 2023-03-07:
You can play with the `minimum_agreement_count`. 

The `sorting_agreement` have the following useful properties (accessible with the `sorting_agreement.get_property(property_name)` function:

- 'agreement_number': number of sorters that agreed upon that unit
- 'avg_agreement': average agreement score
- 'unit_ids': a dictionary with the unit ID of the original sorters

closed 2023-03-27 by alejoe91