SpikeInterface/spikeinterface #1399: TypeError: Object of type ndarray is not JSON serializable

eduardosand, 2023-03-07:
Obtained the error above when making recording objects from numpy arrays, and then preprocessing(bandpass and common reference), followed by spike sorting. Please advise. I think it's related to another issue opened but I'm not completely sure.

h-mayorquin, 2023-03-07:
I think  #1397 will fix this and all the problems with serializing numpy.

eduardosand, 2023-03-08:
Yes, I git cloned your branch and then ran `pip install -e .` I attempted to rerun and received the following: `AssertionError                            Traceback (most recent call last)
Cell In[3], line 2
      1 # Here we're gonna run a spike sorter with default parameters as a test for this software
----> 2 sorting_TDC = ss.run_sorter(sorter_name="spykingcircus2", recording=recording_cmr, detect_threshold=4, docker_image=False)
      3 print(sorting_TDC)
      4 print('Units found by tridesclous:', sorting_TDC.get_unit_ids())

File ~/.local/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:143, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    136             container_image = singularity_image
    137     return run_sorter_container(
    138         container_image=container_image,
    139         mode=mode,
    140         **common_kwargs,
    141     )
--> 143 return run_sorter_local(**common_kwargs)

File ~/.local/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:158, in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
    155 SorterClass = sorter_dict[sorter_name]
    157 # only classmethod call not instance (stateless at instance level but state is in folder)
--> 158 output_folder = SorterClass.initialize_folder(
    159     recording, output_folder, verbose, remove_existing_folder)
    160 SorterClass.set_params_to_folder(
    161     recording, output_folder, sorter_params, verbose)
    162 SorterClass.setup_recording(recording, output_folder, verbose=verbose)

File ~/.local/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py:134, in BaseSorter.initialize_folder(cls, recording, output_folder, verbose, remove_existing_folder)
    132 rec_file = output_folder / 'spikeinterface_recording.json'
    133 if recording.is_dumpable:
--> 134     recording.dump_to_json(rec_file)
    135 else:
    136     d = {'warning': 'The recording is not dumpable'}

File ~/.local/lib/python3.9/site-packages/spikeinterface/core/base.py:477, in BaseExtractor.dump_to_json(self, file_path, relative_to, folder_metadata)
    465 def dump_to_json(self, file_path=None, relative_to=None, folder_metadata=None):
    466     """
    467     Dump recording extractor to json file.
    468     The extractor can be re-loaded with load_extractor_from_json(json_file)
   (...)
    475         If not None, file_paths are serialized relative to this path
    476     """
--> 477     assert self.check_if_dumpable()
    478     dump_dict = self.to_dict(include_annotations=True,
    479                              include_properties=False,
    480                              relative_to=relative_to,
    481                              folder_metadata=folder_metadata)
    482     file_path = self._get_file_path(file_path, ['.json'])

AssertionError: `

h-mayorquin, 2023-03-08:
Thanks for trying this! 

Is that the asesrtion error that you got, an empty string or your copy is missing something? 

Can you show me how you build the `recording_cmr` so I can run something similar on my side?

eduardosand, 2023-03-08:
It's possible my copy is missing something. There is no NEO extractor for this type of recording (BCI2000) so I packaged the signal into a numpy array. `recording = se.NumpyRecording(traces_list=signal.T[:,0:1], 
                              sampling_frequency=sampling_frequency,
                              channel_ids=[params["ChannelNames"][0:1]])

# Minimal preprocessing
recording_f = preproc.bandpass_filter(recording, freq_min=300, freq_max=6000)
recording_cmr = preproc.common_reference(recording_f, reference='global', operator='median')`

h-mayorquin, 2023-03-08:
I meant to ask whether there was something after the following lines:

```python
480 relative_to=relative_to,
481 folder_metadata=folder_metadata)
482 file_path = self._get_file_path(file_path, ['.json'])

AssertionError: `
```

Thanks for the code, I think I can test that in my side as it is, as I don't think that the contents of the numpy arrays matters for this error.

h-mayorquin, 2023-03-08:
@eduardosand Can you test the latest additions? Maybe change the `detect_threshold` as it was giving me troubles.

eduardosand, 2023-03-08:
Problem solved! I removed the `detect_threshold`. Thanks!

h-mayorquin, 2023-03-09:
@eduardosand That's great that it solved the problem.

I think we should keep the issue open until we decide whether we merge https://github.com/SpikeInterface/spikeinterface/pull/1397 or not though.

h-mayorquin, 2023-03-22:
#1397 now is merged and in master. Thanks again @eduardosand for brining this up.  I will close this issue. Do not hesitate in letting us know if you have any other problems using our software.

closed 2023-03-22 by h-mayorquin