SpikeInterface/spikeinterface #1401: Issue setting sorter parameters using spykingcircus2

bradlytstone, 2023-03-07:
I have ran into an issue specifying/applying sorter parameters when I attempt to run spykingcircus2 as the sorter. 

I have tried the following:

# Set parameters
sorter_params = dict(n_jobs=20, chunk_duration="1s", progress_bar=True)

# Spike sort
sorting_spyk = ss.run_sorter(sorter_name="spykingcircus2", recording = recording_cmr,**sorter_params,\
                            output_folder='spykcirc2_output', apply_preprocessing=False)

# I get this error
Traceback (most recent call last):

  File "C:\Users\SuperComputer1\AppData\Local\Temp\ipykernel_17216\2247076330.py", line 2, in <module>
    sorting_spyk = ss.run_sorter(sorter_name="spykingcircus2", recording = recording_cmr,**sorter_params,\

  File "C:\Users\SuperComputer1\miniconda3\lib\site-packages\spikeinterface-0.98.0.dev0-py3.9.egg\spikeinterface\sorters\runsorter.py", line 143, in run_sorter
    return run_sorter_local(**common_kwargs)

  File "C:\Users\SuperComputer1\miniconda3\lib\site-packages\spikeinterface-0.98.0.dev0-py3.9.egg\spikeinterface\sorters\runsorter.py", line 160, in run_sorter_local
    SorterClass.set_params_to_folder(

  File "C:\Users\SuperComputer1\miniconda3\lib\site-packages\spikeinterface-0.98.0.dev0-py3.9.egg\spikeinterface\sorters\basesorter.py", line 166, in set_params_to_folder
    raise AttributeError('Bad parameters: ' + str(bad_params))

AttributeError: Bad parameters: ['n_jobs', 'chunk_duration', 'progress_bar']

# Confusion
This is confusing, because other spike sorters allow for parameter setting, which is a necessary utility. I have also tried running the sorter as follows....

# Run sorter a different way
params = {'chunk_duration': '1s',
 'common_ref_removal': False,
 'detect_sign': -1,
 'detect_threshold': 5,
 'freq_max': 3000.0,
 'freq_min': 300.0,
 'n_jobs': 20,
 'nested_params': None,
 'progress_bar': True}

sorting_spyk = ss.run_spykingcircus(recording = recording_cmr, **params,\
                            output_folder='spykcirc2_output')

# Getting this error
Traceback (most recent call last):

  File "C:\Users\SuperComputer1\AppData\Local\Temp\ipykernel_17216\3983645263.py", line 1, in <module>
    sorting_spyk = ss.run_spykingcircus(recording = recording_cmr, **params,\

  File "C:\Users\SuperComputer1\miniconda3\lib\site-packages\spikeinterface-0.98.0.dev0-py3.9.egg\spikeinterface\sorters\runsorter.py", line 698, in run_spykingcircus
    return run_sorter('spykingcircus', *args, **kwargs)

  File "C:\Users\SuperComputer1\miniconda3\lib\site-packages\spikeinterface-0.98.0.dev0-py3.9.egg\spikeinterface\sorters\runsorter.py", line 143, in run_sorter
    return run_sorter_local(**common_kwargs)

  File "C:\Users\SuperComputer1\miniconda3\lib\site-packages\spikeinterface-0.98.0.dev0-py3.9.egg\spikeinterface\sorters\runsorter.py", line 160, in run_sorter_local
    SorterClass.set_params_to_folder(

  File "C:\Users\SuperComputer1\miniconda3\lib\site-packages\spikeinterface-0.98.0.dev0-py3.9.egg\spikeinterface\sorters\basesorter.py", line 166, in set_params_to_folder
    raise AttributeError('Bad parameters: ' + str(bad_params))

AttributeError: Bad parameters: ['chunk_duration', 'common_ref_removal', 'freq_max', 'freq_min', 'n_jobs', 'nested_params', 'progress_bar']

alejoe91, 2023-03-07:
you can check the available params with:

```
ss.get_default_sorter_params("spykingcircus2")
```

You'll see that SC2 needs a `job_kwargs` dictionary, not flattened arguments.

alejoe91, 2023-03-07:
These are the available params:
```
{'general': {'ms_before': 2, 'ms_after': 2, 'local_radius_um': 100},
 'waveforms': {'max_spikes_per_unit': 200, 'overwrite': True},
 'filtering': {'dtype': 'float32'},
 'detection': {'peak_sign': 'neg', 'detect_threshold': 5},
 'selection': {'n_peaks_per_channel': 5000, 'min_n_peaks': 20000},
 'localization': {},
 'clustering': {},
 'matching': {},
 'registration': {},
 'apply_preprocessing': True,
 'shared_memory': False,
 'job_kwargs': {}}
```

bradlytstone, 2023-03-07:
Thank you, @alejoe91!

So, if I am reading this correctly, this specific sorter does not support parallel processing? If not, then I am not sure it is a utility for the size of my files moving forward.... right now, a 20-min recording with 25K sampling frequency takes >30min to spike sort with quite a powerful computer.

alejoe91, 2023-03-07:
It does support parallel processing. 
You have to use `job_kwargs`:

```
job_kwargs = dict(n_jobs=20, chunk_duration="1s", progress_bar=True)
sorting_spyk = ss.run_sorter(sorter_name="spykingcircus2", recording = recording_cmr,
                                                   output_folder='spykcirc2_output', apply_preprocessing=False, job_kwargs=job_kwargs)
```

bradlytstone, 2023-03-07:
Thank you, Joe. This worked.

closed 2023-03-07 by bradlytstone