SpikeInterface/spikeinterface #1406: Postprocessing Tutorial compute spike waveforms not as expected.

mandarmp, 2023-03-08:
I am new to spike sorting and was exploring the post processing tutorial in the read the docs. [https://spikeinterface.readthedocs.io/en/0.92.0/modules/toolkit/plot_2_postprocessing.html](url)

The data is a HD-MEA recording of the network activity of neurons.(1024 electrodes with high firing rate from Maxwel)

When I try to plot the waveforms on the particular electrode. The spikes are not aligned as seen in the picture.
[image]

Should I follow some alignment procedure to get better templates? Thank you in advance for helping out.

mandarmp, 2023-03-08:
Also it there a slack group/ discord group for spike sorting related stuff. Thank you.

mandarmp, 2023-03-10:
Nvm resolved the issue.

closed 2023-03-10 by mandarmp

alejoe91, 2023-03-10:
Great to hear. What was the issue in the end?

mandarmp, 2023-03-10:
It was a rather silly misunderstanding from my side. I didn't know that get_waveforms ..returned the electrical footprint on all the electrodes and I was printing the wrong electrode. I used the get_template_extremum_channels to find out the right electrode. Is there any forums where I can ask such questions rather than creating a GitHub issue. New to spike sorting please help. Thank you so much.