SpikeInterface/spikeinterface #1409: New naming for peak data type

h-mayorquin, 2023-03-09:
The current base_peak_dtype:
```
base_peak_dtype = [('sample_ind', 'int64'), ('channel_ind', 'int64'),
                   ('amplitude', 'float64'), ('segment_ind', 'int64')]
```
Do I remember correctly that we agreed to change this to have more explicit naming in a meeting but we did not follow through?

If that is the case, I can do it. 

What is the new naming?

* `sample_ind` -> `sample_index`
* `channel_ind` -> `channel_index`
* `amplitude` -> `peak_ampitude`
* `segment_ind` -> `segment_index

samuelgarcia, 2023-03-09:
Hi
I am agree for `ind` -> `index`
But I think we can keep `amplitudes` 

I prefer `peaks['ampitude']` over `peaks['peak_ampitude']`

I you go the modification can you also check the changes:
 * inside sorter module
 * in doc

h-mayorquin, 2023-03-09:
Good point about the redundancy when referencing it from a peaks object!

alejoe91, 2023-03-20:
`indices` should be the correct plural

h-mayorquin, 2023-03-20:
@alejoe91 @samuelgarcia 
Note that neo uses `channel_indexes` (e.g. `_get_analogsignal_chunk`), are you guys fine with that discrepancy?

samuelgarcia, 2023-03-21:
Copy paste from the web:

**Both “indexes” and “indices” are correct English plurals of the singular noun “index.” But there is a preference for indices outside North America because of solid adherence to Latin plurals.**

I guess Alessio sometimes feel a latin guy.

h-mayorquin, 2023-03-21:
A Roman heritage. 

Outside of that it is easier to edit in bulk when the plurars are different from the singulars but on the other hand if you are looking for places where a term is used it might be a good idea that a simple search (no regular expression) would return both the terms (and that happens with the regular plural). I think that @pauladkisson pointed out something along those lines but can't recall.

No idea what's better. I think it is just better here to pick a standard and stick with it.

pauladkisson, 2023-03-21:
Personally, I think I have a mild preference for 'indexes'.  'Indices' *feels* like the proper usage since it is much more common in mathematical/scientific language, but 'indexes' makes for easier code reading/writing since it is a regular plural.  I found this source to be helpful: [indexing - Proper terminology (Stack Overflow)](https://stackoverflow.com/questions/1378781/proper-terminology-should-i-say-indexes-or-indices)

h-mayorquin, 2023-04-07:
Thanks for the link @pauladkisson . I actually had your mild preference till today when I went and check numpy and it seems that they do prefer indices:
https://numpy.org/doc/stable/reference/arrays.indexing.html

alejoe91, 2023-04-07:
Thanks for the link @h-mayorquin 

Let's settle for `indices` then ;)

PR #1571 (2023-04-25): Change `cluster_ind` to `cluster_index`
Final PR for #1409. This should close #1409. Changing `cluster_ind` to `cluster_index`.
closed 2023-04-25 by samuelgarcia