SpikeInterface/spikeinterface #1417: ValueError: could not broadcast input array - motion estimation

bramn22, 2023-03-14:
Hi , 

When running the motion_estimation function, I get the following error:

"/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/site-packages/spikeinterface/sortingcomponents/motion_estimation.py:1011: RuntimeWarning: invalid value encountered in true_divide
  corr /= npx.sqrt(var_x * var_template)
Traceback (most recent call last):
  File "/home/farrowlab/conda-farrowlab/preprocess_npx.py", line 216, in <module>
    main(cfg)
  File "/home/farrowlab/conda-farrowlab/preprocess_npx.py", line 208, in main
    rec = motion_correction_full(rec, out_path=out_path, shift_idx=shift_idx, visualize=True)
  File "/home/farrowlab/conda-farrowlab/preprocess_npx.py", line 118, in motion_correction_full
    motion, temporal_bins, spatial_bins, extra_check = spikeinterface.sortingcomponents.motion_estimation.estimate_motion(
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/site-packages/spikeinterface/sortingcomponents/motion_estimation.py", line 127, in estimate_motion
    motion, temporal_bins = method_class.run(recording, peaks, peak_locations, direction, bin_duration_s, bin_um,
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/site-packages/spikeinterface/sortingcomponents/motion_estimation.py", line 263, in run
    motion[:, i] = compute_global_displacement(pairwise_displacement,
ValueError: could not broadcast input array from shape (1124,) into shape (1125,)"

function used: `motion, temporal_bins, spatial_bins, extra_check = spikeinterface.sortingcomponents.motion_estimation.estimate_motion(
        rec,
        peaks,
        peak_locations=peak_locations,
        direction='y',
        bin_duration_s=5,
        bin_um=10.,
        rigid=False,
        win_shape='gaussian', win_step_um=50., win_sigma_um=150,
        progress_bar=True,
        output_extra_check=True,
        verbose=True,
    )`

Thank you,
Bram

samuelgarcia, 2023-03-14:
Hi bram,
Thanks for the report.
Which version do you have ?
Is it installed from the github sources ?

bramn22, 2023-03-14:
Hi Samuel,

I am using 0.97.0. I just noticed that there is 0.97.1 so I am trying that now.

bramn22, 2023-03-14:
With version 0.97.1, an error now occurs during the writing of the binary recording:

"write_binary_recording:  20%|█▉        | 1124/5624 [06:04<24:20,  3.08it/s]
concurrent.futures.process._RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/concurrent/futures/process.py", line 243, in _process_worker
    r = call_item.fn(*call_item.args, **call_item.kwargs)
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/concurrent/futures/process.py", line 202, in _process_chunk
    return [fn(*args) for args in chunk]
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/concurrent/futures/process.py", line 202, in <listcomp>
    return [fn(*args) for args in chunk]
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/site-packages/spikeinterface/core/job_tools.py", line 399, in function_wrapper
    return _func(segment_index, start_frame, end_frame, _worker_ctx)
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/site-packages/spikeinterface/core/core_tools.py", line 209, in _write_binary_chunk
    traces = recording.get_traces(start_frame=start_frame, end_frame=end_frame, segment_index=segment_index,
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/site-packages/spikeinterface/core/baserecording.py", line 160, in get_traces
    traces = rs.get_traces(start_frame=start_frame, end_frame=end_frame, channel_indices=channel_indices)
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/site-packages/spikeinterface/sortingcomponents/motion_correction.py", line 335, in get_traces
    trace2 = correct_motion_on_traces(traces, times, self.channel_locations, self.motion,
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/site-packages/spikeinterface/sortingcomponents/motion_correction.py", line 124, in correct_motion_on_traces
    f = scipy.interpolate.interp1d(spatial_bins, motion[bin_ind, :], kind='linear',
IndexError: index 1125 is out of bounds for axis 0 with size 1125
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/farrowlab/conda-farrowlab/preprocess_npx.py", line 216, in <module>
    main(cfg)
  File "/home/farrowlab/conda-farrowlab/preprocess_npx.py", line 209, in main
    save_file(rec, os.path.join(out_path, 'data'), binary=False)
  File "/home/farrowlab/conda-farrowlab/preprocess_npx.py", line 44, in save_file
    rec.save(folder=out_path, n_jobs=10, chunk_duration='1s', progres_bar=True)
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/site-packages/spikeinterface/core/base.py", line 621, in save
    loaded_extractor = self.save_to_folder(**kwargs)
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/site-packages/spikeinterface/core/base.py", line 700, in save_to_folder
    cached = self._save(folder=folder, verbose=verbose, **save_kwargs)
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/site-packages/spikeinterface/core/baserecording.py", line 297, in _save
    write_binary_recording(self, file_paths=file_paths, dtype=dtype, **job_kwargs)
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/site-packages/spikeinterface/core/core_tools.py", line 280, in write_binary_recording
    executor.run()
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/site-packages/spikeinterface/core/job_tools.py", line 364, in run
    for res in results:
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/site-packages/tqdm/std.py", line 1195, in __iter__
    for obj in iterable:
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/concurrent/futures/process.py", line 559, in _chain_from_iterable_of_lists
    for element in iterable:
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/concurrent/futures/_base.py", line 609, in result_iterator
    yield fs.pop().result()
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/concurrent/futures/_base.py", line 439, in result
    return self.__get_result()
  File "/home/farrowlab/conda-farrowlab/anaconda3/envs/si_bram/lib/python3.9/concurrent/futures/_base.py", line 391, in __get_result
    raise self._exception
IndexError: index 1125 is out of bounds for axis 0 with size 1125"

alejoe91, 2023-11-06:
@bramn22 sorry for the super delay on this. We actually made quite some improvements to the motion/drift in more recent releases (>=0.98). Are you still experiencing issues?

borrepp, 2024-03-20:
Hello @alejoe91 , 

I'm getting the same error, basically the peaks.py and peak_locations.py ended with different shapes. In my case is:

ValueError:  could not broadcast input array from shape (588052,) into shape (588055,)

I'm using windows 11, spikeInterface version = 0.100.0

alejoe91, 2024-03-21:
ok, what is the output of `print(len(sorting.to_spike_vector()))`?

borrepp, 2024-03-21:
I don't have a sorting object. The peaks.py and peak_locations.py were created while following the pipeline for motion correction.

alejoe91, 2024-03-22:
Ah I think we have a bug in the torch detection! Can you try with the `locally_exclusive`?

borrepp, 2024-03-22:
With "locally_exclusive" it runs fine. This time, "peaks" and "peak_location" have the same shape. 
I think I will just set the "radius_um" to be smaller than the spacing of the probe that I have, which should behave like "by_channel"

However, when I try to run the next step: "estimate_motion" with "method_motion" = "decentralized". I have the error that I posted here:

https://github.com/SpikeInterface/spikeinterface/issues/2603

I will wait on that issue to not mix channels. 

On the other hand, if I run "estimate_motion" with "method_motion" = "iterative_template". I have no errors. But when I run "interpolate_motion":

#######################################################################################
rec_corrected = interpolate_motion(recording=rec_filter_file, motion=motion,
                                   temporal_bins=temporal_bins,
                                   spatial_bins=spatial_bins,
                                   border_mode="remove_channels",
                                   spatial_interpolation_method="kriging",
                                   sigma_um=30.)
#######################################################################################

I got zero channels:

>>print(rec_corrected)
InterpolateMotionRecording: 0 channels - 30.0kHz - 1 segments - 44,771,040 samples 
                            1,492.37s (24.87 minutes) - float32 dtype - 0.00 B

#######################################################################################

Did I do something wrong?

Thanks again for your help¡¡¡

borrepp, 2024-03-22:
By the way, with `locally_exclusive_torch`. I also got the same number of peaks and peak_locations.

closed 2025-03-19 by zm711