SpikeInterface/spikeinterface #1421: Mapping of Intan Channels with Spikeinterface

kshtjkumar, 2023-03-16:
Hi, so i collected neural data from intan rhx device. We used two types of electrodes 1 is depth electrode(say channel number 28,29,30,31), other is a surface electrode (ECog, say channel number 1,3,5,7 ). Rest channels were deactivated. So now I just have 8 channel ids and i know based on channel number which channel id should correspond to what type of electrode. However when I upload the data using the extractor the channel ids are by default taken from '0' and I get an array of channel_ids = ['0', '1', '2'....]. This makes it difficult for me to group the channels based on type and process the data. Is it possible to see the original channel id as recorded by the intan device so that I can do the grouping easily?

I am using this code:

import spikeinterface as si
file = '1st_DEC_LKO _MICE_Merged.rhs'
reader = si.extractors.neoextractors.intan.IntanRecordingExtractor(file,stream_name ='RHD2000 amplifier channel')
#reader = si.extractors.read_intan(file, stream_name ='RHD2000 amplifier channel')
recording_rhs = reader

recording_rhs.annotate(is_filtered = False)

channel_ids = recording_rhs.get_channel_ids()
fs = recording_rhs.get_sampling_frequency()
num_chan = recording_rhs.get_num_channels()
num_segments = recording_rhs.get_num_segments()

print("Channel_ids = ", channel_ids)
print("Sampling_frequency = ", fs)
print("Number of Channels = ", num_chan)
print("Number of segments = ", num_segments)

alejoe91, 2023-03-16:
Hi, 

Can you paste the output of:
```
print(reader.neo_reader.header['signal_channels']['name'])
```
?

If this returns the correct channel names, we should be able to easily propagate them to the SpikeInterface side

kshtjkumar, 2023-03-16:
['B-000'  'B-006'  'B-007'  'B-008'  'B-009'  'B-011'  'B-029'  'B-030'  'B-031']

yes these are the correct channel numbers.

alejoe91, 2023-03-16:
Thanks, I made a small PR for you to test: #1422 

Can you install SpikeInterface from that PR and test this:
```
import spikeinterface.extractors as se
file = '1st_DEC_LKO _MICE_Merged.rhs'
reader = se.read_intan(file,stream_name ='RHD2000 amplifier channel', use_names_as_ids=True)

print(reader.channel_ids)
```

Do you know how to install from a pull request?

kshtjkumar, 2023-03-16:
Hi , Thank you so much. No, am not sure how to use pull request with PR. Can you please guide ?

kshtjkumar, 2023-03-16:
Thank you so much.

alejoe91, 2023-03-16:
Let me keep this open until #1422 is merged ;)

kshtjkumar, 2023-03-18:
hi, currently after the update (reinstalling with the PR as above)  
I am facing some issues with the sorters, I tried with different numpy versions (1.21.1 , 1.18.5, 1.20.3).  Please have a look at these errors:
[image]
[image]
[image]

alejoe91, 2023-03-20:
Hi,

So the first error probably comes from herding spikes. Can you try to `pip install herdingspikes==0.3.99`?

For the Kilosort errors, have you checked the `kilosort_output/spikeinterface_log.json` ( as the error suggests)?

kshtjkumar, 2023-03-20:
hi, I tried the herdingspikes command, the output is: 
['herdingspikes',
 'spykingcircus',
 'spykingcircus2',
 'tridesclous',
 'tridesclous2']

and upon checking the Kilosort_output json file i ee the following error :
{
     "sorter_name": "kilosort",
     "sorter_version": "unknown",
     "datetime": "2023-03-18T22:45:34.464597",
     "runtime_trace": [],
     "error": true,
     "error_trace": "Traceback (most recent call last):\n  File
 \"c:\\users\\admin\\spikeinterface\\spikeinterface\\sorters\\basesorter.py\",
 line 226, in run_from_folder\n
 SorterClass._run_from_folder(sorter_output_folder, sorter_params,
 verbose)\n  File
 \"c:\\users\\admin\\spikeinterface\\spikeinterface\\sorters\\external\\kilosortbase.py\",
 line 179, in _run_from_folder\n    raise Exception(f'{cls.sorter_name}
 returned a non-zero exit code')\nException: kilosort returned a
 non-zero exit code\n",
     "run_time": null
 }

alejoe91, 2023-03-20:
so kilosort doesn't seem to be installed, right?

kshtjkumar, 2023-03-20:
but we did install it using the same command as given in the Document (https://spikeinterface.readthedocs.io/en/latest/install_sorters.html). 
We used the: git clone https://github.com/cortex-lab/KiloSort  
Is there any other command which I should run ?

alejoe91, 2023-03-20:
Did you do this? 
```
# provide installation path by setting the KILOSORT_PATH environment variable
# or using KilosortSorter.set_kilosort_path()
```

kshtjkumar, 2023-03-21:
yes, i did run this command :
ss.kilosort.KilosortSorter.set_kilosort_path('C:\\Users\\Admin\\KiloSort-master')

which is the path for the downloaded software.

kshtjkumar, 2023-03-26:
I think it could also be because of the GPU settings, there has been some changes in the PC so I will try to re run the sorter after fixing the GPU issue as even the above solutions also didnt resolve the problem. Will update this once the GPU is fixed.

kshtjkumar, 2023-03-26:
Although while analysing the raw data (without using sorters), I followed the preprocessing doc, now i wanted to reference based on channel groups as the channels are from different type of probes. With the recent fix I was able to get the original channel ids:
channel_ids = recording_rhs.get_channel_ids()
and the output was :
Channel_ids = ['B-000' 'B-006' 'B-007' 'B-008' 'B-009' 'B-011' 'B-029' 'B-030' 'B-031'] 

based on this I created a group, so that I can first analyse recordings from one type of probe and then for other and compare the results later.

grp1 = ['B-006','B-007','B-008','B-009'] 

Now to do channel re referecing based on this group of channels only I run the following:

recording_cmr = spre.common_reference(recording_center, reference='global', operator='median', groups= grp1, ref_channel_ids=None, local_radius=(30, 55), verbose=False)

It throws me this error :  
[image]

Further in the preprocessing module I face another error upon using this , to detect bad channels : 

spre.detect_bad_channels(recording_cmr, method='coherence+psd',std_mad_threshold=5, psd_hf_threshold=0.02, dead_channel_threshold=-0.5, noisy_channel_threshold=1.0, outside_channel_threshold=-0.75, n_neighbors=11, nyquist_threshold=0.8, direction='y', chunk_duration_s=0.3, num_random_chunks=10, welch_window_ms=10.0,
highpass_filter_cutoff=300, seed=None)
[image]

alejoe91, 2023-03-27:
Hi,

For common reference, the `groups` argument of the function expects a list of lists, while you are passing a list. If you want to analyse the to groups completely separately, you should create two objects by slicing the channels:

```
rec1 = recording_rhs.channel_slice(['B-006','B-007','B-008','B-009'])
recording_cmr = spre.common_reference(rec1, reference='global', operator='median')
```
with this approach you don't need the `groups` argument. Please read the documentation for more details: https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.preprocessing.common_reference

The second error is likely due to an outdated version of scipy. You should be able to fix it just by upgrading:
```
pip install -U scipy
```

I'm going to close this issue because it's not related to a single problem. Feel free to open a new issue with a specific problem later.

closed 2023-03-27 by alejoe91