SpikeInterface/spikeinterface #1429: `get_unit_spike_train` broken in `PhyKilosortExtractor`

TomBugnon, 2023-03-17:
Hi all  

I am just noticing that [this](https://github.com/SpikeInterface/spikeinterface/commit/966af09c9d03131de8de188a8e193bf8fe57d36d) (@DradeAW ) commit breaks get_unit_spike_train when start_frame or end_frame is provided...!!

This fails with the following error in `searchsorted`
`sorting.get_unit_spike_train(unit_id=0, segment_index=0, start_frame=1)`

```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[49], line 1
----> 1 sorting.get_unit_spike_train(unit_id=0, segment_index=0, start_frame=1)

File ~/projects/ecephys_dev_si_pipeline/spikeinterface/spikeinterface/core/basesorting.py:61, in BaseSorting.get_unit_spike_train(self, unit_id, segment_index, start_frame, end_frame, return_times)
     59 segment_index = self._check_segment_index(segment_index)
     60 segment = self._sorting_segments[segment_index]
---> 61 spike_frames = segment.get_unit_spike_train(
     62     unit_id=unit_id, start_frame=start_frame, end_frame=end_frame).astype("int64")
     63 if return_times:
     64     if self.has_recording():

File ~/projects/ecephys_dev_si_pipeline/spikeinterface/spikeinterface/extractors/phykilosortextractors.py:152, in PhySortingSegment.get_unit_spike_train(self, unit_id, start_frame, end_frame)
    151 def get_unit_spike_train(self, unit_id, start_frame, end_frame):
--> 152     start = 0 if start_frame is None else np.searchsorted(self._all_spikes, start_frame, side="left")
    153     end = len(self._all_spikes) if end_frame is None else np.searchsorted(self._all_spikes, end_frame, side="right")
    155     spike_times = self._all_spikes[start:end][self._all_clusters[start:end] == unit_id]

File <__array_function__ internals>:180, in searchsorted(*args, **kwargs)

File ~/miniconda3/envs/ecephys_dev_si_pipeline/lib/python3.10/site-packages/numpy/core/fromnumeric.py:1387, in searchsorted(a, v, side, sorter)
   1319 @array_function_dispatch(_searchsorted_dispatcher)
   1320 def searchsorted(a, v, side='left', sorter=None):
   1321     """
   1322     Find indices where elements should be inserted to maintain order.
   1323
   (...)
   1385
   1386     """
-> 1387     return _wrapfunc(a, 'searchsorted', v, side=side, sorter=sorter)

File ~/miniconda3/envs/ecephys_dev_si_pipeline/lib/python3.10/site-packages/numpy/core/fromnumeric.py:57, in _wrapfunc(obj, method, *args, **kwds)
     54     return _wrapit(obj, method, *args, **kwds)
     56 try:
---> 57     return bound(*args, **kwds)
     58 except TypeError:
     59     # A TypeError occurs if the object does have such a method in its
     60     # class, but its signature is not identical to that of NumPy's. This
   (...)
     64     # Call _wrapit from within the except clause to ensure a potential
     65     # exception has a traceback chain.
     66     return _wrapit(obj, method, *args, **kwds)

ValueError: object too deep for desired array
```

This is due to `self._all_spikes` being at this point a 2D `(nspikes, 1)` array instead of `(nspikes,)`
Fixed by adding `spike_times = np.atleast_1d(spike_times.squeeze())` eg [here](https://github.com/SpikeInterface/spikeinterface/blob/97c39d19428ef8d037b7b07ca92e7752595cd25c/spikeinterface/extractors/phykilosortextractors.py#L38)

Best

DradeAW, 2023-03-17:
Thanks!
I am at a conference this week-end but I'll look it up and push a fix early next week :)

Best

DradeAW, 2023-03-20:
I tested it on the phy output from Kilosort 2.0, and it works fine for me:

sorting = se.PhySortingExtractor(".")
> _all_spikes.shape = (7224728,)
sorting.get_unit_spike_train(0)
> array([   43075,    43191,    43346, ..., 53988315, 53991057, 53991645])
sorting.get_unit_spike_train(0, start_frame=45000)
> array([   45266,    45340,    45791, ..., 53988315, 53991057, 53991645])

The question becomes why is `_all_spikes` a 2d-array for you and a 1d-array for me?

TomBugnon, 2023-03-21:
Have you tried KS2.5?

I just tried with the basic KS2.5 GUI, no spikeinterface involved, and `spike_times.npy` is 2D. It seems like it has always been for all the sortings I've run so far (all with variations of KS2.5)

DradeAW, 2023-03-22:
I haven't tried yet,

Does anyone know why this array is 2D? @alejoe91 ?
Maybe we can do something like:

```python
if all_spikes.ndim == 2:
    all_spikes = all_spikes[:, 0]
```

alejoe91, 2023-03-22:
I think that `spike_times = np.atleast_1d(spike_times.squeeze())` should work all the times, no?

PR #1453 (2023-03-27): Fix in PhyKilosortSortingExtractor
Fixes #1429
commit fb4ff4d1 (2023-03-27): Fix in PhyKilosortSortingExtractor

Fixes #1429
closed 2023-03-27 by alejoe91