SpikeInterface/spikeinterface #1431: "NotImplementedError: This meta file version of spikeglxis not implemented" error when running read_spikeglx() and get_neo_streams() on npx data

NDolensek, 2023-03-20:
I'm trying to follow the "Analyse Neuropixels datasets" tutorial and am getting stuck at the first step when trying to analyze my own neuropixels data acquired with SpikeGLX, with default settings. Data folder contains .imec0.ap.meta, .imec0.ap.bin, .imec0.lf.bin and .imec0.lf.meta files. Running either si.get_neo_streams or si.read_spikeglx() results in a "NotImplementedError: This meta file version of spikeglxis not implemented". 
Does data need to be in a specific format for the spikeinterface to work?

`stream_names, stream_ids = si.get_neo_streams('spikeglx', path)
stream_names`

```
NotImplementedError                       Traceback (most recent call last)
Cell In[7], line 1
----> 1 stream_names, stream_ids = si.get_neo_streams('spikeglx', path)
      2 stream_names

File [c:\Users\ndolensek\mambaforge\envs\spikeinterface\lib\site-packages\spikeinterface\extractors\neoextractors\neo_utils.py:26](file:///C:/Users/ndolensek/mambaforge/envs/spikeinterface/lib/site-packages/spikeinterface/extractors/neoextractors/neo_utils.py:26), in get_neo_streams(extractor_name, *args, **kwargs)
      5 """Returns the NEO streams (stream names and stream ids) associated to a dataset.
      6 For multi-stream datasets, the `stream_id` or `stream_name` arguments can be used 
      7 to select which stream to read with the `read_**extractor_name**()` function.
   (...)
     23     List of NEO stream ids
     24 """
     25 neo_extractor = get_neo_extractor(extractor_name)
---> 26 return neo_extractor.get_streams(*args, **kwargs)

File [c:\Users\ndolensek\mambaforge\envs\spikeinterface\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py:142](file:///C:/Users/ndolensek/mambaforge/envs/spikeinterface/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:142), in NeoBaseRecordingExtractor.get_streams(cls, *args, **kwargs)
    139 @classmethod
    140 def get_streams(cls, *args, **kwargs):
    141     neo_kwargs = cls.map_to_neo_kwargs(*args, **kwargs)
--> 142     neo_reader = get_reader(cls.NeoRawIOClass, **neo_kwargs)
    144     stream_channels = neo_reader.header['signal_streams']
    145     stream_names = list(stream_channels['name'])

File [c:\Users\ndolensek\mambaforge\envs\spikeinterface\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py:14](file:///C:/Users/ndolensek/mambaforge/envs/spikeinterface/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:14), in get_reader(raw_class, **neo_kwargs)
...
    407                                   'is not implemented')
    408 else:
    409     stream_kind = ''

NotImplementedError: This meta file version of spikeglxis not implemented
```

alejoe91, 2023-03-20:
Hi @NDolensek 

What version pf SpikeGLX are you using? Would you mind sharing the meta file with us?
Maybe a recent SpikeGLX update will require a patch on our side.

Cheers
Alessio

NDolensek, 2023-03-20:
The probes we are using are definitely a bit non-standard, they are 45mm long 1.0-NHP probes (https://www.biorxiv.org/content/10.1101/2023.02.01.526664v2)

alejoe91, 2023-03-20:
I see. We're in the process of supporting the new probes, which requires changes both on the NEO and the [probeinterface](https://github.com/SpikeInterface/probeinterface/pull/156) side.

Can you confirm that the probe type is the same as 1.0, in terms of channel gains and so on? 

We found a discrepancy between the file you shared and the file that @JonathanAMichaels shared with us. In Jonathan's case, the `imDatPrb_type` is 0, in your case it's `1030`. @JonathanAMichaels do you know if the `imDatPrb_type` changed in recent versions of SpikeGLX?

NDolensek, 2023-03-20:
We checked with Bill Karsh, whole develops SpikeGLX:

>The NP1030 is the same as the NP1000 electrically {refs, gains, filters, bands}.
>The site pitch (spacing) is the same vertically (32 microns along the shank), but the NP1030 is 55 microns wider.

Also, the paper I linked above has this to say:

>The Neuropixels 1.0-NHP probe uses the same signal-conditioning circuits as the Neuropixels 1.0 probe1, integrating 384 low-noise readout channels with programmable gain and 10-bit resolution

So it sounds like treating gain in the same way  as with the standard NP1 probes should work fine.

Regarding the discrepancy in `imDatPrb_type` between us and @JonathanAMichaels, that could be caused by our probes differing, he might've recorded with an older, passive NHP probe or a 25mm version, versus our 45mm one.

alejoe91, 2023-03-21:
Thanks, this is helpful. Let me try to intergate this info with  @JonathanAMichaels one from this issue: https://github.com/SpikeInterface/probeinterface/issues/155

So from this document (https://www.neuropixels.org/_files/ugd/328966_97dfdcf9421a41ae9c57c9047928834c.pdf) there are three types of NP1.0-NHP: 
- short
- medium
- long

I don't see a `1030` code, but the closest would be the `NP1032` in the "order table", which corresponds to the "long" version. @NDolensek can you confirm?

Jonathan's .meta file has `1015` (presumably the short shank, @JonathanAMichaels can you confirm?).

So we could do something like this to differentiate between the probes:
If `imDatPrb_pn` is a field in the .meta file, we'll use that:
- `NP101*`: NHP-short (x-pitch=32, 960 electrodes - like NP2)
- `NP102*`: NHP-medium (x-pitch=103, 2496 electrodes)
- `NP103*`:  NHP-long (x-pitch=103, 4416 electrodes)

Does that sound correct? 

Thank you for helping us out with this! It's not so straightforward to gather all these info for the newly developed models :)

JonathanAMichaels, 2023-03-21:
That sounds correct to me. I am using the NHP short. The only thing to add is that none of these new probes use a ‘staggered’ electrode design, which is currently assumed in the code. Thanks!

samuelgarcia, 2023-03-21:
Nate : thank you for all the details and Alessio for your insvestigation.

NDolensek, 2023-03-22:
Relevant pull requests:
NeuralEnsemble/python-neo#1237
SpikeInterface/probeinterface#160

Below is a more detailed probe type summary:
imDatPrb_type=1030 (NP 1.0-NHP 45mm SOI90 - NHP long 90um wide, staggered contacts)
imDatPrb_type=1031 (NP 1.0-NHP 45mm SOI125 - NHP long 125um wide, staggered contacts)
imDatPrb_type=1032 (NP 1.0-NHP 45mm SOI115 / 125 linear - NHP long 125um wide, linear contacts)
imDatPrb_type=1022 (NP 1.0-NHP 25mm - NHP medium)
imDatPrb_type=1015 (NP 1.0-NHP 10mm - NHP short)
All probes are electrically identical to NP 1.0 probes as per official documentation

alejoe91, 2023-03-27:
@NDolensek can you confirm that the probeinterface and NEO PRs solved this issue? In that case, can I close this?

JonathanAMichaels, 2023-04-07:
@alejoe91 any new changes that might resolve this error?
@NDolensek, is it working for you?

JonathanAMichaels, 2023-04-29:
What data is way too large to share. Is it sufficient to share the meta file again?

alejoe91, 2023-04-29:
The error appears in reading the folder structure, so it would be better to get the full folder. If you send me an email I can provide a drive link!

alejoe91, 2023-06-12:
Hey guys, this works for me using the **main** branches of spikeinterface and probeinterface (at leat on a test folder provided by @JonathanAMichaels ).  
Can you test with the latest changes?

Cheers
Alessio

JonathanAMichaels, 2023-06-12:
This now works for me with the main branches!

alejoe91, 2023-11-06:
All new NP probes are now supported!

closed 2023-11-06 by alejoe91