SpikeInterface/spikeinterface #1432: Force `_main_ids` to be strings

alejoe91, 2023-03-20:
Related to the serialization thread (#1397), after discussing with @h-mayorquin, we thought that casting `Base._main_ids` as `str` **always** may solve several issues with keys not being json serializable.
Allowing ids to be integers is mainly to maintain a similar behavior with `spikeextractors`, but now that ids and indices are clearly separated I think we can just use strings.

@samuelgarcia thoughts?

magland, 2023-03-22:
I understand why forcing IDs to be str is desirable. However, I believe this would break many of the recordings/sortings that I work with, that use integer IDs. So not sure what is the best way forward here.

alejoe91, 2023-03-22:
@magland could you elaborate why they would break? If you use, for instance, channel IDs to index or slice an array, that is not the way it should be because we have a clear api distinction between IDs and indices now. Do you have other breaking examples?

magland, 2023-03-22:
I have a lot of existing code that uses integers for IDs when creating recordings/sortings. Frank lab also has recordings/sortings stored to disk in custom formats that I believe use integers for IDs.

magland, 2023-03-22:
For example, here's a line of code that I just wrote:

```
recording_signal = se.NumpyRecording([signal], sampling_frequency=recording_sub_whitened.sampling_frequency, channel_ids=np.arange(1, KK + 1, dtype=np.int32))
```

It would be cumbersome to convert that code to use strings instead of ints.

alejoe91, 2023-03-22:
No we would do the conversion in the base object and under the hood. So if the downstream code does not depend on the IDs being int there's no problem

magland, 2023-03-22:
So we'd be able to pass in ints for any of the API functions and they get translated to strings under the hood?

alejoe91, 2023-03-22:
Well normally you would then the `channel_ids` from the object and pass them around!

alejoe91, 2023-03-22:
I see! I agree it's better going forward. Probably also better to switch sooner than later!

magland, 2023-03-22:
> Maybe it should wise to deprecate explcitly in next release (actual master) and remove ids integer in the following one. So anyone could figure out which part of code will be broken in advance. This would be more smart no ?

I think that would be problematic, since you'd get a bunch of deprecation warnings for things that aren't really going to be problems. Like if I set the unit IDs as integers, that is still going to be supported according to @alejoe91. What you can't test for is how the user interprets the output of `.unit_ids` or `.channel_ids`.

alejoe91, 2023-03-22:
Yep ;)

samuelgarcia, 2023-03-22:
I not sure to support the magical and hidden convertion : int to str in the base class.
Magic stuff are futurs bugs!

magland, 2023-03-22:
Now that I think more about this, I think the problem is not that int IDs are supported... the problem is that int IDs are used as the keys in dicts. A different solution would be for you to always use the stringified versions of IDs when creating dicts. When a user passes in a dict with ID keys, they need to make sure they are string keys. Similarly when a function outputs a dict, it needs to make sure the keys are strings. Then if a user reads a dict they would need to do `X[str(id)]` ...

magland, 2023-03-22:
> I not sure to support the magical and hidden convertion : int to str in the base class. Magic stuff are futurs bugs!

If you don't support int's being passed in, it's going to break a lot of my existing code as well as in spyglass.

samuelgarcia, 2023-03-23:
No worry, the idea is not to break codes!
The key as int in dict is one aspect one the problem, the other is mainly to have something clear without ambiguity for a new user.

For me ket as int and str is OK. Key as str only would be better.
Key that are transform along the way (in base or when making dict or whenever) is a bad solution.

So the 2 options I like are:
  1. keep the actual behavior (int+str), we lived with this since a while without major drawback
  2. str only everwhere since init of the class (but this break code)

All other solutions that make hidden and magic cast to str are a design error I think.

magland, 2023-03-23:
Regarding json serialization, the following could be a better strategy

Replace things like this:

```
{
  1: [...],
  2: [...],
  ...
}
```

with things like this

```
{
  'locations': [
    {'id': 0, 'location': [...]},
    {'id': 1, 'location': [...]},
    ...
  }
}
```

h-mayorquin, 2024-05-03:
I guess we never did agree on this one, did we? 

Can we do this with a very slow deprecation cycle. Start taking throwing deprecation warnings for ints when setting channel_ids and unit_ids?

samuelgarcia, 2024-05-03:
Honestly, having unit_ids as int for sorting is really really convinient in many sututation.
For instance in Numpy.from_time_label where the labels become ids.

JoeZiminski, 2024-05-03:
Is it possible to introduce a getter `_get_main_ids(dtype="int")` so representation is `str` under the hood but can grab as `int` if required / make an easy grep on existing code?

h-mayorquin, 2024-05-03:
> Honestly, having unit_ids as int for sorting is really really convinient in many sututation. For instance in Numpy.from_time_label where the labels become ids.

Can't we just use `ids_to_indices` there or is it more complicated than that?