SpikeInterface/spikeinterface #1447: Problem opening Phy after export_to_phy

marinemm-fr, 2023-03-22:
I used the function export to Phy with the following parameters : 
`export_to_phy(we_clean, base_folder / 'export_to_phy', compute_pc_features=True, compute_amplitudes=True, sparsity=None, copy_binary=True, remove_if_exists=False, peak_sign='neg', template_mode='median', dtype=None, verbose=True, **job_kwargs)`
I have around 540 units and my recording is about 30 min long,tThe .dat file generated is about 40 GB and I didnt had an error while creating the file.
When running phy template.gui I have the following error : 
`from phy.apps.template import template_gui
template_gui("recording.dat")
Traceback (most recent call last):

  Cell In[7], line 2
    template_gui("recording.dat")

  File ~\anaconda3\envs\si_env\lib\site-packages\phy\apps\template\gui.py:209 in template_gui
    model = load_model(params_path)

  File ~\anaconda3\envs\si_env\lib\site-packages\phylib\io\model.py:1433 in load_model
    return TemplateModel(**get_template_params(params_path))

  File ~\anaconda3\envs\si_env\lib\site-packages\phylib\io\model.py:1416 in get_template_params
    params = read_python(params_path)

  File ~\anaconda3\envs\si_env\lib\site-packages\phylib\utils\_misc.py:171 in read_python
    contents = path.read_text()

  File ~\anaconda3\envs\si_env\lib\pathlib.py:1267 in read_text
    return f.read()

  File ~\anaconda3\envs\si_env\lib\encodings\cp1252.py:23 in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]

MemoryError`

Seems like I ran out of memory (I have 16G RAM).

However when I just run Phy directly from the output from Kilosort 2.5 I am able to open the GUI and work on it.
Does anyone have an idea of how I could work around that problem and be able to run Phy on my curated recording ?
Thank you !

alejoe91, 2023-03-24:
Hi,

The error is weird, because it seems 9t be generated by a `read_text()` call in reading the parameters.
Can you share the `params.py`? (Even a copy paste here would work)

zm711, 2023-08-29:
@marinemm-fr 
@alejoe91 

I just realized what the problem with this is. `template_gui` needs to be called on the `params.py` not on the binary file, `recording.dat`. The gui was trying to read the whole ~40gb binary file.

h-mayorquin, 2023-09-07:
@marinemm-fr 
Did you have time to try the solution proposed by @zm711? Did it work for you?

zm711, 2023-10-06:
@h-mayorquin, I can say that my fix is the correct one for using Phy. There could be a separate bug, but this particular issue is fixed by pointing the template-gui to params.py. 

To prevent this issue in the future this was added to the `export_to_phy` at some point (it wasn't me):

https://github.com/SpikeInterface/spikeinterface/blob/03f0f084aad894a071e96b821f13b9e25e238e75/src/spikeinterface/exporters/to_phy.py#L272-L274

Did you want to leave this open? There's not much else we can do. But @marinemm-fr can definitely send more issues if something else comes up!

marinemm-fr, 2023-10-19:
Hello all !  Sorry I didn't sign up on github for a while as I was working on some grant writing. I think that I ended up figuring out that I indeed didn't call the right file as @zm711 mentioned and it worked ! 
Thanks for updating the code to prevent this issue for others in the future. You can close it :)

zm711, 2023-10-19:
@marinemm-fr Hope the grants went well. Feel free to open more issues if anything comes up.

closed 2023-10-19 by zm711

h-mayorquin, 2023-10-20:
Thanks @zm711