SpikeInterface/spikeinterface #1450: OpenEphysBinaryRecordingExtractor crash when not in the right folder

DradeAW, 2023-03-23:
I'm trying to load an OpenEphys Neuropixels recording, but depending on the folder in which I'm running the Python script, I can get a crash or not:

```
[...]/Frequency_Recording$ python
>>> import spikeinterface.extractors as se
>>> se.OpenEphysBinaryRecordingExtractor("2023-03-22_18-24-37/Record Node 111", experiment_names="experiment2", stream_id='0', load_sync_timestamps=True)
Traceback (most recent call last):
  File "plot_frequencies.py", line 4, in <module>
    ap_recording  = se.OpenEphysBinaryRecordingExtractor("2023-03-22_18-24-37/Record Node 111/", experiment_names="experiment2", stream_id='0', load_sync_timestamps=True)
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/spikeinterface/extractors/neoextractors/openephys.py", line 177, in __init__
    self.set_times(times=sync_times, segment_index=segment_index, with_warning=False)
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/spikeinterface/core/baserecording.py", line 261, in set_times
    assert times.ndim == 1, 'Time must have ndim=1'
AttributeError: 'NoneType' object has no attribute 'ndim'
exit()

[...]/Frequency_Recording$ cd 2023-03-22_18-24-37/Record Node 111
[...]/Frequency_Recording/2023-03-22_18-24-37/Record Node 111$ python
>>> import spikeinterface.extractors as se
>>> se.OpenEphysBinaryRecordingExtractor(".", experiment_names="experiment2", stream_id='0', load_sync_timestamps=True)  # Works as intended
```

I think this has to do with the stream names.
In the first case (when it crashes), the stream names are `['Record Node 111#Neuropix-PXI-106.ProbeA-AP', 'Record Node 111#Neuropix-PXI-106.ProbeA-LFP']`. Since the "Record Node 111" is in their, in duplicates it in `stream_folder` (https://github.com/SpikeInterface/spikeinterface/blob/947c47f4eb43a31a04345369f7cdac519e0286f9/spikeinterface/extractors/neoextractors/openephys.py#L163)

DradeAW, 2023-03-23:
If I point to the folder containing the Record Node folder it works! Thanks

We need to change the documentation then :)

alejoe91, 2023-03-23:
Yes...I'm sorry I should have checked better! Would you mind making another PR to fix that?

PR #1451 (2023-03-23): Fix in OpenEphysBinaryRecordingExtractor doc
Fixes #1450
commit ca0f822c (2023-03-23): Fix in OpenEphysBinaryRecordingExtractor doc

Fixes #1450
closed 2023-03-23 by alejoe91