SpikeInterface/spikeinterface #1464: DeprecationWarning

skyetomez, 2023-03-29:
The following produces this deprecation error:

.../python3.10/site-packages/pkg_resources/__init__.py:121: DeprecationWarning: pkg_resources is deprecated as an API
  warnings.warn("pkg_resources is deprecated as an API", DeprecationWarning)
/home/skylerthomas_umass_edu/.conda/envs/spike_sort/lib/python3.10/site-packages/pkg_resources/__init__.py:2870: DeprecationWarning: Deprecated call to `pkg_resources.declare_namespace('google')`.
Implementing implicit namespace packages (as specified in PEP 420) is preferred to `pkg_resources.declare_namespace`. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages
  declare_namespace(pkg)
.../python3.10/site-packages/pkg_resources/__init__.py:2870: DeprecationWarning: Deprecated call to `pkg_resources.declare_namespace('mpl_toolkits')`.
Implementing implicit namespace packages (as specified in PEP 420) is preferred to `pkg_resources.declare_namespace`. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages
  declare_namespace(pkg)
.../python3.10/site-packages/pkg_resources/__init__.py:2870: DeprecationWarning: Deprecated call to `pkg_resources.declare_namespace('google')`.
Implementing implicit namespace packages (as specified in PEP 420) is preferred to `pkg_resources.declare_namespace`. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages
  declare_namespace(pkg)
.../python3.10/site-packages/pkg_resources/__init__.py:2870: DeprecationWarning: Deprecated call to `pkg_resources.declare_namespace('ruamel')`.
Implementing implicit namespace packages (as specified in PEP 420) is preferred to `pkg_resources.declare_namespace`. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages
  declare_namespace(pkg)
.../python3.10/site-packages/pkg_resources/__init__.py:2870: DeprecationWarning: Deprecated call to `pkg_resources.declare_namespace('ruamel')`.
Implementing implicit namespace packages (as specified in PEP 420) is preferred to `pkg_resources.declare_namespace`. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages
  declare_namespace(pkg)
.../python3.10/site-packages/pkg_resources/__init__.py:2870: DeprecationWarning: Deprecated call to `pkg_resources.declare_namespace('ruamel.yaml')`.
Implementing implicit namespace packages (as specified in PEP 420) is preferred to `pkg_resources.declare_namespace`. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages
  declare_namespace(pkg)
.../python3.10/site-packages/pkg_resources/__init__.py:2349: DeprecationWarning: Deprecated call to `pkg_resources.declare_namespace('ruamel')`.
Implementing implicit namespace packages (as specified in PEP 420) is preferred to `pkg_resources.declare_namespace`. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages
  declare_namespace(parent)

It looks like this is a solution: https://github.com/matplotlib/matplotlib/issues/25244#issuecomment-1437354281

But this might also be how I'm loading it. 

```python
import spikeinterface.extractors as se

```

h-mayorquin, 2023-03-30:
Hi, @skyetomez 

I am having trouble to parse, what is the following?

I don't see any code between the error trace and the beginning. Did you miss to add something or am I reading bad?

skyetomez, 2023-03-30:
I wrote it at the bottom of the post. It's just the import. I will move it to the top sorry.

alejoe91, 2023-03-30:
This is due to `setuptools` deprecating namespaces and 3rd party packages, unfortunately..

A workaround is downgrading `setuptools`:

```
pip install setuptools==67.2.0
```

I expect that soon all packages will adapt, but we have to be patient for now...

skyetomez, 2023-03-30:
Will this affect any performance? Also are the modules being loaded lazily?

h-mayorquin, 2023-03-30:
I see. Thanks.

Modules are not being loaded lazily.

h-mayorquin, 2023-03-30:
Are you using this spikeinterface as a part of a develompent environment or a stand alone application that you install with pip? If the latter I think this could help:

https://pypa.github.io/pipx/

@alejoe91 should we put a ceiling of setuptools on the `pyproject.toml`?

h-mayorquin, 2023-05-04:
Was the issue solved @skyetomez ?

Now we are not loading all the modules at the beginning except if you have them.

skyetomez, 2023-05-04:
Hi @h-mayorquin I think so. At least importing spikeinterface itself seems faster; importing the submodules has a bit of hang e.g. `import spikeinterface.extractoras se`. It still returns the PEP 420 error also

```
DeprecationWarning: Deprecated call to `pkg_resources.declare_namespace('ruamel')`.
Implementing implicit namespace packages (as specified in PEP 420) is preferred to `pkg_resources.declare_namespace`. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages
```

h-mayorquin, 2023-05-04:
That's great, that's an improvement.

Yep, we changed mostly to delayed imports here:

https://github.com/SpikeInterface/spikeinterface/pull/1597

But there are still some pending imports that are done with try-except so the speed-up is not uniform across modules. We are still discussing how (and how far) we move in this route.

h-mayorquin, 2023-07-21:
We are mostly done with this. I think some of the try-excepts are confined to the widgets module out of technical necessity but this is probably better addressed on a different issue. 

Closing this.

closed 2023-07-21 by h-mayorquin