SpikeInterface/spikeinterface #1465: ImportError: cannot import name 'BaseEvent' from 'spikeinterface' (unknown location)

magland, 2023-03-30:
I'm getting this error on the master branch when I try to import spikeinterface.spikesorters

```bash
    import spikeinterface.sorters as ss
  File "/home/magland/src/spikeinterface-tmp/spikeinterface/sorters/__init__.py", line 2, in <module>
    from .sorterlist import *
  File "/home/magland/src/spikeinterface-tmp/spikeinterface/sorters/sorterlist.py", line 1, in <module>
    from .external.combinato import CombinatoSorter
  File "/home/magland/src/spikeinterface-tmp/spikeinterface/sorters/external/combinato.py", line 10, in <module>
    from spikeinterface.extractors import CombinatoSortingExtractor
  File "/home/magland/src/spikeinterface-tmp/spikeinterface/extractors/__init__.py", line 1, in <module>
    from .extractorlist import *
  File "/home/magland/src/spikeinterface-tmp/spikeinterface/extractors/extractorlist.py", line 10, in <module>
    from .neoextractors import *
  File "/home/magland/src/spikeinterface-tmp/spikeinterface/extractors/neoextractors/__init__.py", line 15, in <module>
    from .maxwell import (MaxwellRecordingExtractor, MaxwellEventExtractor, 
  File "/home/magland/src/spikeinterface-tmp/spikeinterface/extractors/neoextractors/maxwell.py", line 5, in <module>
    from spikeinterface import BaseEvent, BaseEventSegment
ImportError: cannot import name 'BaseEvent' from 'spikeinterface' (unknown location)
```

magland, 2023-03-30:
> Hi Jeremy, I think this is because you have the spikeinterface repo cloned in the same folder!

What do you mean? I cloned the repo to spikeinterface-tmp and then I ran

```
cd spikeinterface-tmp
pip install -e .
```

Is that not a supported way to do it?

alejoe91, 2023-03-30:
That is fine, but then you should run your script from another folder, otherwise the local `spikeinterface` folder shadows the package one

alejoe91, 2023-03-30:
Implementing this https://github.com/SpikeInterface/spikeinterface/issues/1344 should fix it

magland, 2023-03-30:
Thanks. I am running from another folder, so I'm not sure what's happening. My workaround for now is to do `pip install .` without the `-e`

alejoe91, 2023-03-30:
Do you have any other folder called `spikeinterface`? Might be causing conflicts

h-mayorquin, 2023-04-06:
@magland When you have some time, would you be so kind to see if you still encounter this error?

closed 2023-04-14 by magland