SpikeInterface/spikeinterface #1466: `OpenEphysBinaryRecordingExtractor` crashes when not recording from all channels

DradeAW, 2023-03-30:
Hi,

I used OpenEphys to record a Neuropixels recording, but in the record node I only recorded the first 200 channels.

When trying to open the recording using the `OpenEphysBinaryRecordingExtractor`, I get the following crash:

```
Traceback (most recent call last):
  File "plot_frequencies.py", line 12, in <module>
    lfp_recording = se.OpenEphysBinaryRecordingExtractor("2023-03-30_17-13-30/", experiment_names="experiment1", stream_id='1', load_sync_timestamps=True)
  File "/export/home1/users/nsr/wyngaard/dev/forked_repo/spikeinterface/spikeinterface/extractors/neoextractors/openephys.py", line 149, in __init__
    self = self.set_probe(probe, in_place=True)
  File "/export/home1/users/nsr/wyngaard/dev/forked_repo/spikeinterface/spikeinterface/core/baserecordingsnippets.py", line 74, in set_probe
    return self.set_probes(probegroup, group_mode=group_mode, in_place=in_place)
  File "/export/home1/users/nsr/wyngaard/dev/forked_repo/spikeinterface/spikeinterface/core/baserecordingsnippets.py", line 141, in set_probes
    raise ValueError(
ValueError: The given Probe have "device_channel_indices" that do not match channel count
```

alejoe91, 2023-03-30:
Hi @DradeAW 

The open epbys uses the xpos and this in the settings file. Are they consistent with the number of channels?

Maybe the bug is there..

DradeAW, 2023-03-31:
All 384 channels are in the `settings.xml` file (channels, electrode_xpos and electrode_ypos).

I think the information about which channels are recorded is here in `recording_state`:
```xml
<PROCESSOR name="Record Node" insertionPoint="1" pluginName="Record Node"
               type="0" index="3" libraryName="" libraryVersion="" processorType="8"
               nodeId="111">
      <GLOBAL_PARAMETERS/>
      <STREAM name="ProbeA-AP" description="description" sample_rate="30000.0"
              channel_count="384" device_name="Neuropixels 1.0">
        <PARAMETERS enable_stream="1"/>
      </STREAM>
      <STREAM name="ProbeA-LFP" description="description" sample_rate="2500.0"
              channel_count="384" device_name="Neuropixels 1.0">
        <PARAMETERS enable_stream="1"/>
      </STREAM>
      <STREAM name="PXIe-6341" description="Analog input channels from a NIDAQ device"
              sample_rate="30000.0" channel_count="8">
        <PARAMETERS/>
      </STREAM>
      <CUSTOM_PARAMETERS path="D:\Frequency_Recording" engine="BINARY" recordEvents="1"
                         recordSpikes="0" fifoMonitorsVisible="1">
        <STREAM isMainStream="1" sync_line="0" name="ProbeA-AP" source_node_id="106"
                sample_rate="30000.0" channel_count="384" device_name="Neuropixels 1.0"
                recording_state="111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"/>
        <STREAM isMainStream="0" sync_line="0" name="ProbeA-LFP" source_node_id="106"
                sample_rate="2500.0" channel_count="384" device_name="Neuropixels 1.0"
                recording_state="111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"/>
        <STREAM isMainStream="0" sync_line="0" name="PXIe-6341" source_node_id="113"
                sample_rate="30000.0" channel_count="8" recording_state="10000000"/>
      </CUSTOM_PARAMETERS>
      <EDITOR isCollapsed="0" isDrawerOpen="1" displayName="Record Node" activeStream="0"/>
    </PROCESSOR>
```

alejoe91, 2023-03-31:
Can you share the `settings.xml`?

closed 2023-04-18 by alejoe91