SpikeInterface/spikeinterface #1467: get_noise_levels returning zeros

JonathanAMichaels, 2023-03-30:
I'm having issues with get_noise_levels. My Neuropixels recordings have a noise level (MAD) around 14uV, but the unscaled noise level in int16 is generally 0-4, with a mean of ~2. This is very low, and the function actually returns zeros in some cases, which breaks detect_peaks by dividing by zero! This also makes the detection threshold very noisy since there are only a few discrete values.

I get around this currently by scaling my recording by a gain (8x) before any preprocessing to get more resolution in the noise levels. So my questions are:

1. Should get_noise_levels allow zero to be returned?
2. Does this likely indicate the the gain in my recordings was too low (Neuropixels default of 500)?

Thanks

samuelgarcia, 2023-03-31:
Hi Jonathan.
Thanks for the feedback.
I have never imagined that noise could rounded to 0 on the integer scale!!

This is good news for the electronics.
Maybe the default gain in neuropixel is not the prefect one to describe the noise in that range also but this is another debate.

For me `get_noise_levels(return_scaled=False)` returnong 0. is zeros is the nearest noise estimation is not a bad behavior. What we should do is to ensure that peak detector is robust with this.

But then we have a more theoritical problem : how to deal peak detection  with noise =0 any sample would be a peak.

The only simple work around is what you propose bu inside the get_noise_levels itself.

```
def get_noise_levels(recording, return_scaled=True, factor=None, **random_chunk_kwargs):

    random_chunks = get_random_data_chunks(recording, return_scaled=return_scaled, **random_chunk_kwargs)
    if not return_scaled and factor is not None:
        small_noise = do something really smaller than factor
        random_chunks = random_chunks.astype('float64') * factor + small_noise
    med = np.median(random_chunks, axis=0, keepdims=True)
    # hard-coded so that core doesn't depend on scipy
    noise_levels = np.median(np.abs(random_chunks - med), axis=0) / 0.6744897501960817

    if not return_scaled=True and factor is not None:
        noise_levels = noise_levls / factor
    return noise_levels
```

with factor = 100

Could you try this and make a patch ?

JonathanAMichaels, 2023-03-31:
Thanks @samuelgarcia. I implemented this:

```
def get_noise_levels(recording, factor=None, return_scaled=True, **random_chunk_kwargs):
    """
    Estimate noise for each channel using MAD methods.

    Internally it sample some chunk across segment.
    And then, it use MAD estimator (more robust than STD)

    """
    random_chunks = get_random_data_chunks(
        recording, return_scaled=return_scaled, **random_chunk_kwargs
    )
    if not return_scaled and factor is not None:
        small_noise = np.random.normal(loc=0., scale=factor/100, size=random_chunks.shape)
        random_chunks = random_chunks.astype('float64') * factor + small_noise
    med = np.median(random_chunks, axis=0, keepdims=True)
    
    # hard-coded so that core doesn't depend on scipy
    noise_levels = (
        np.median(np.abs(random_chunks - med), axis=0) / 0.6744897501960817
    )
    
    if not return_scaled and factor is not None:
        noise_levels = noise_levels / factor
     
    return noise_levels
```

Which gets rid of the issue with zeros being generated. However, it doesn't resolve the issue that the estimates of the noise is very discrete (small values near 1,2,3), so when I detect peaks with MAD*5 I get 50-200 million peaks detected. If I instead scale my entire recording by a factor before preprocessing then I get the more expected ~5 million peaks detected. Thoughts?

samuelgarcia, 2023-03-31:
Thanks for testing.
I undesrtand that the problem is deep inside the data itself more that algos.
When the discretisation and range is not appropriate a very small diif rounded and then multiplied by a factor change everything!!!

In short noiselevel = 1.49 then threhold is 5 and noiselevel = 1.51 then threshol is 10.
Maybe you should pass to detect peak a fake noise_levels which is an setimate for all channel and not by channels so that you can control finely the final threhold.

In short you give noise_level = abs_thresh / std_thresh_factor.
No ?

closed 2023-04-17 by JonathanAMichaels