SpikeInterface/spikeinterface #1469: `SpikeGLXRecordingExtractor` missing `firstSample`

DradeAW, 2023-03-31:
Hi,

While trying SpikeGLX, I found out that the first sample of the AP and LFP channels are not exactly synchronized:

- AP channel: 30,000.00 Hz ; firstSample = 9367709 (312.2570 s)
- LFP channel: 2,500.00 Hz , firstSample = 780642 (312.2568 s)

780,642 * 12 = 9,367,704 (compared to 9,367,709)

I wanted to re-align them using `recording.get_times()`, but found out that this starts at 0, not 312s.

I looked in the annotations and properties, but couldn't find the `firstSample` anywhere.
It seems that this piece of information is missing from the recording?

I would be really helpful to then synchronize AP and LFP channels.

alejoe91, 2023-04-04:
Thanks @DradeAW 

This is related to NEO, since the timestamp info is retrieved from there (normally).
We could also have a patch in SI, but it would be better to patch NEO directly.

Can you open an issue there?

alejoe91, 2023-11-06:
Moved to NEO

closed 2023-11-06 by alejoe91

zm711, 2024-02-18:
I answered on Neo, but just so there is an SI answer too:

```python
recording.neo_reader.signals_info_list[0]['meta']['firstSample'] # for first file
```

should provide the desired info. If this is super important maybe an edit to the spikeglx extractor to move this to an easily accessible property. Otherwise just use the what I gave above to find it.

DradeAW, 2024-02-19:
That's good enough for me, thanks!