SpikeInterface/spikeinterface #1476: PathNotFoundError with Zarr recording extractor when extracting waveforms

vncntprvst, 2023-04-03:
I'm trying to load waveforms, from a recording that was saved in zarr format. 
When calling 
```
we = si.extract_waveforms(recording_saved_zarr, sorting, folder=base_folder / "waveforms", 
                          load_if_exists=True, overwrite=False, **job_kwargs)
```
I get this error:
```---------------------------------------------------------------------------
PathNotFoundError                         Traceback (most recent call last)
File ~/mambaforge/envs/si_env/lib/python3.9/site-packages/spikeinterface/core/waveform_extractor.py:162, in WaveformExtractor.load_from_folder(cls, folder, with_recording, sorting)
    161 try:
--> 162     recording = load_extractor(folder / 'recording.json',
    163                             base_folder=folder)
    164     rec_attributes = None

File ~/mambaforge/envs/si_env/lib/python3.9/site-packages/spikeinterface/core/base.py:911, in load_extractor(file_or_folder_or_dict, base_folder)
    910 else:
--> 911     return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)

File ~/mambaforge/envs/si_env/lib/python3.9/site-packages/spikeinterface/core/base.py:536, in BaseExtractor.load(file_path, base_folder)
    535     return None
--> 536 extractor = BaseExtractor.from_dict(d, base_folder=base_folder)
    537 return extractor

File ~/mambaforge/envs/si_env/lib/python3.9/site-packages/spikeinterface/core/base.py:371, in BaseExtractor.from_dict(d, base_folder)
    370     d = _make_paths_absolute(d, base_folder)
--> 371 extractor = _load_extractor_from_dict(d)
    372 folder_metadata = d.get('folder_metadata', None)

File ~/mambaforge/envs/si_env/lib/python3.9/site-packages/spikeinterface/core/base.py:854, in _load_extractor_from_dict(dic)
    853 # instantiate extrator object
--> 854 extractor = cls(**kwargs)
    856 extractor._annotations.update(dic['annotations'])

File ~/mambaforge/envs/si_env/lib/python3.9/site-packages/spikeinterface/core/zarrrecordingextractor.py:55, in ZarrRecordingExtractor.__init__(self, root_path, storage_options)
     53     root_path_kwarg = root_path_init
---> 55 self._root = zarr.open(root_path_init, mode="r", storage_options=storage_options)
     57 sampling_frequency = self._root.attrs.get("sampling_frequency", None)

File ~/mambaforge/envs/si_env/lib/python3.9/site-packages/zarr/convenience.py:122, in open(store, mode, zarr_version, path, **kwargs)
    121 else:
--> 122     raise PathNotFoundError(path)

PathNotFoundError: nothing found at path ''

During handling of the above exception, another exception occurred:

Exception                                 Traceback (most recent call last)
Cell In[39], line 1
----> 1 we = si.extract_waveforms(recording_saved_zarr, sorting, folder=base_folder / "waveforms", 
      2                           load_if_exists=True, overwrite=False, **job_kwargs)
      3 print(we)

File ~/mambaforge/envs/si_env/lib/python3.9/site-packages/spikeinterface/core/waveform_extractor.py:1374, in extract_waveforms(recording, sorting, folder, mode, precompute_template, ms_before, ms_after, max_spikes_per_unit, overwrite, return_scaled, dtype, sparse, num_spikes_for_sparsity, allow_unfiltered, use_relative_path, seed, load_if_exists, **kwargs)
   1372         shutil.rmtree(folder)
   1373     if load_if_exists and folder.is_dir():
-> 1374         we = WaveformExtractor.load_from_folder(folder)
   1375         return we
   1377 if sparse:

File ~/mambaforge/envs/si_env/lib/python3.9/site-packages/spikeinterface/core/waveform_extractor.py:166, in WaveformExtractor.load_from_folder(cls, folder, with_recording, sorting)
    164         rec_attributes = None
    165     except:
--> 166         raise Exception("The recording could not be loaded. You can use the `with_recording=False` argument")
    168 if sorting is None:
    169     sorting = load_extractor(folder / 'sorting.json',
    170                              base_folder=folder)

Exception: The recording could not be loaded. You can use the `with_recording=False` argument
```
Just in case, I've loaded the recording with 
`recording_saved_zarr = si.ZarrRecordingExtractor(base_folder / "preprocessed.zarr")`
or 
`recording_saved_zarr = si.read_zarr(base_folder / "preprocessed.zarr")`
and got the same result. Reading the file works, e.g. `ZarrRecordingExtractor: 384 channels - 1 segments - 30.0kHz - 909.656s`, but waveform extraction fails.
Any idea what's wrong here? 
Thanks

alejoe91, 2023-04-04:
Thanks @vncntprvst 

Looking into this!

alejoe91, 2023-04-04:
[image]

vncntprvst, 2023-04-06:
The content in base_folder / "waveforms" / "recording.json" is quite long but here's the top part:
```
{
    "class": "spikeinterface.core.binaryfolder.BinaryFolderRecording",
    "module": "spikeinterface",
    "kwargs": {
        "folder_path": "/home/wanglab/data/whisker_asym/sc014/sc014_0324/sc014_0324_001/preprocessed"
    },
    "dumpable": true,
    "version": "0.96.2.dev0",
    "relative_paths": false,
    "annotations": {
        "is_filtered": true,
        "probe_0_planar_contour":
```

vncntprvst, 2023-04-06:
Based on your question, I deleted the base_folder / "waveforms" folder and now it's extracting fine ... 
So, presumably, there was some conflict with a waveform extraction folder previously created? 
In that case, it could help the user to have a more explicit error message.

h-mayorquin, 2023-04-11:
Can you see if you can generate the error deterministically? That is, can you reproduce the issue? 

That would help us to see in what conditions the error should be triggered and tailor both a message and a test for it.

vncntprvst, 2023-04-14:
Good point. Let me try to reproduce it.

alejoe91, 2023-06-12:
@vncntprvst any update on this?

closed 2023-11-06 by alejoe91