SpikeInterface/spikeinterface #1512: no option to pass necessary params for whitening int recording

jguides, 2023-04-10:
A dtype or integer scale argument is required when whitening a recording with integer datatype (see https://github.com/LorenFrankLab/spyglass/issues/445 and https://github.com/LorenFrankLab/spyglass/issues/448). However, there is currently not a way to pass these arguments if you want to whiten just prior to spikesorting with mountainsort4. In the _run_from_folder method of the MountainSort SorterClass, only the recording is sent into the whiten method, with no option to specify the necessary dtype or int_scale parameters (see code snippet below). 

Could this be addressed? Alternatively, could you provide guidance on how to convert the recording to float16 prior to spikesorting?

Thanks in advance for any help with this.

The relevant part of the _run_from_folder method from spikeinterface/sorters/external/mountainsort4.py:
```
        # Whiten
        if p['whiten']:
            if verbose:
                print('whitening')
            recording = whiten(recording=recording)

```

Original error:
```
Running spike sorting on {'nwb_file_name': 'J1620210614_.nwb', 'sort_group_id': 0, 'sort_interval_name': 'raw data valid times no premaze no home no sleep', 'preproc_params_name': 'franklab_tetrode_hippocampus_min_seg', 'team_name': 'JG_DG', 'sorter': 'mountainsort4', 'sorter_params_name': 'franklab_tetrode_hippocampus_30KHz', 'artifact_removed_interval_list_name': 'J1620210614_.nwb_0_raw data valid times no premaze no home no sleep_franklab_tetrode_hippocampus_min_seg_JG_DG_group_0.25_500_8_2_artifact_removed_valid_times'}...
Traceback (most recent call last):
  File "/home/jguidera/anaconda3/envs/spyglass/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py", line 226, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "/home/jguidera/anaconda3/envs/spyglass/lib/python3.9/site-packages/spikeinterface/sorters/external/mountainsort4.py", line 108, in _run_from_folder
    recording = whiten(recording=recording)
  File "/home/jguidera/anaconda3/envs/spyglass/lib/python3.9/site-packages/spikeinterface/core/core_tools.py", line 28, in reader_func
    return source_class(*args, **kwargs)
  File "/home/jguidera/anaconda3/envs/spyglass/lib/python3.9/site-packages/spikeinterface/preprocessing/whiten.py", line 62, in __init__
    assert int_scale is not None, 'For recording with dtype=int you must set dtype=float32 OR set a int_scale'
AssertionError: For recording with dtype=int you must set dtype=float32 OR set a int_scale

---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
Cell In [7], line 77
     73 sorting_key.update({'sorter': sorter,
     74                'sorter_params_name': sorter_params_name,
     75                'artifact_removed_interval_list_name': artifact_removed_interval_list_name})
     76 nd.spikesorting.SpikeSortingSelection.insert1(sorting_key, skip_duplicates=True)
---> 77 nd.spikesorting.SpikeSorting.populate(sorting_key)
     78 curation_key = nd.spikesorting.Curation.insert_curation(sorting_key)
     80 # Waveforms

File ~/anaconda3/envs/spyglass/lib/python3.9/site-packages/datajoint/autopopulate.py:241, in AutoPopulate.populate(self, suppress_errors, return_exception_objects, reserve_jobs, order, limit, max_calls, display_progress, processes, make_kwargs, *restrictions)
    237 if processes == 1:
    238     for key in (
    239         tqdm(keys, desc=self.__class__.__name__) if display_progress else keys
    240     ):
--> 241         error = self._populate1(key, jobs, **populate_kwargs)
    242         if error is not None:
    243             error_list.append(error)

File ~/anaconda3/envs/spyglass/lib/python3.9/site-packages/datajoint/autopopulate.py:292, in AutoPopulate._populate1(self, key, jobs, suppress_errors, return_exception_objects, make_kwargs)
    290 self.__class__._allow_insert = True
    291 try:
--> 292     make(dict(key), **(make_kwargs or {}))
    293 except (KeyboardInterrupt, SystemExit, Exception) as error:
    294     try:

File ~/Src/spyglass/src/spyglass/spikesorting/spikesorting_sorting.py:202, in SpikeSorting.make(self, key)
    196     sorting = si.NumpySorting.from_times_labels(
    197         times_list=detected_spikes["sample_ind"],
    198         labels_list=np.zeros(len(detected_spikes), dtype=np.int),
    199         sampling_frequency=recording.get_sampling_frequency(),
    200     )
    201 else:
--> 202     sorting = sis.run_sorter(
    203         sorter,
    204         recording,
    205         output_folder=sorter_temp_dir.name,
    206         delete_output_folder=True,
    207         **sorter_params,
    208     )
    209 key["time_of_sort"] = int(time.time())
    211 print("Saving sorting results...")

File ~/anaconda3/envs/spyglass/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:143, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    136             container_image = singularity_image
    137     return run_sorter_container(
    138         container_image=container_image,
    139         mode=mode,
    140         **common_kwargs,
    141     )
--> 143 return run_sorter_local(**common_kwargs)

File ~/anaconda3/envs/spyglass/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:163, in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
    160 SorterClass.set_params_to_folder(
    161     recording, output_folder, sorter_params, verbose)
    162 SorterClass.setup_recording(recording, output_folder, verbose=verbose)
--> 163 SorterClass.run_from_folder(output_folder, raise_error, verbose)
    164 if with_output:
    165     sorting = SorterClass.get_result_from_folder(output_folder)

File ~/anaconda3/envs/spyglass/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py:262, in BaseSorter.run_from_folder(cls, output_folder, raise_error, verbose)
    260 if has_error and raise_error:
    261     print(log['error_trace'])
--> 262     raise SpikeSortingError(
    263         f"Spike sorting failed. You can inspect the runtime trace in {output_folder}/spikeinterface_log.json")
    265 return run_time

SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in /tmp/tmpiabcxmxh/spikeinterface_log.json
```

alejoe91, 2023-04-12:
Thanks! Indeed this is probably due to a recent update to the `whiten` function. I suggest we expose the `whiten_kwargs` at the `mountainsort` level and automatically add a `float` dtype in case the recording is `int16`.

You can change the dtype with the `scale` function ;)

```
recording_float = spre.scale(recording_int, dtype="float")
```

That  should fix it for now!

alejoe91, 2023-04-27:
Also fixed by: #1519

closed 2023-04-27 by alejoe91