SpikeInterface/spikeinterface #1514: WobbleMatch TODOs

pauladkisson, 2023-04-11:
A list of desired changes in the Wobble template matching algorithm to be implemented after benchmarking suite is established.  See [PR](https://github.com/SpikeInterface/spikeinterface/pull/1448#discussion_r1161608885) for more detailed rationale.

- refractory_period_frames --> refractory_period_ms
- sparsity --> spikeinterface ChannelSparsity
- margin --> margin_ms
- benchmark find_peaks vs. spikeinterface detect_peaks methods --> replace find_peaks with one of the existing peak detection methods or add find_peaks to the peak detection submodule
- benchmark peak_radius
- Fix IndexError that pops up in certain downsampling + parameter conditions

yger, 2023-04-12:
I would add unify the API with the other matching engines. I don't know if this is the best choice (and thus I'm open to discussion), but we decided with @samuelgarcia  to pass a waveform_extractor as an argument to the engines. I guess wobble could make use of that, it would be better for building pipelines and benchmarks

h-mayorquin, 2023-04-12:
Hi @yger.
I was the one that reviewed code and as I am less familiar with the signature of the other templates this escaped me. I totally agree with you that they should have the same signature. Can you point out to me to the signature of the other methods (is that their `__init__`  to see how that would look like?

The only possible counter-argument is that @pauladkisson did not want to go too far way from the Paninski lab implementation.   What do you think @pauladkisson ?

If you have time it would be great to get some review on the code. I think it would helpful for @pauladkisson to get a review from  someone that actually has expertise on the subject matter as my comments were only about code organization.

alejoe91, 2023-11-06:
Closing this, TM will need a refactoring anyways soon

closed 2023-11-06 by alejoe91