SpikeInterface/spikeinterface #1521: More Comprehensive Testing for Template Matching Algorithms

pauladkisson, 2023-04-12:
As mentioned by [Ramon](https://github.com/SpikeInterface/spikeinterface/pull/1448#pullrequestreview-1375911970), the template matching algs could use a bit more testing such as
- Different recording sampling frequencies
- Different recording dtypes
- Edge cases in the waveform extractor (very large or very small ms_before/ms_after, etc.)

yger, 2023-04-12:
I saw that the PR with the wobble algorithms was merged! Great. However, what was the motivation to make the default signature different from the ones in the other matching engines. Why not accepting a waveform_extractor as input, in order to get the required parameters (templates, nbefore, nafter) in the init? This is a bit annoying, and I don't really get the point, or maybe I missed it? @alejoe91 @pauladkisson @h-mayorquin

alejoe91, 2023-11-06:
Same here. we'll get back to this :)