SpikeInterface/spikeinterface #1523: `MEArecRecordingExtractor` not dumpable anymore

DradeAW, 2023-04-12:
Hi,

With recent commits, I am now unable to extract waveforms because the `MEArecRecordingExtractor` cannot be dumped anymore:

```python
>>> import spikeinterface.extractors as se
>>> 
>>> recording = se.MEArecRecordingExtractor("recording.h5")
>>> recording.dump("/path/to/recording.json")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/base.py", line 459, in dump
    self.dump_to_json(file_path, relative_to=relative_to, folder_metadata=folder_metadata)
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/base.py", line 485, in dump_to_json
    json.dumps(dump_dict, indent=4, cls=SIJsonEncoder),
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/__init__.py", line 238, in dumps
    **kw).encode(obj)
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 201, in encode
    chunks = list(chunks)
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 431, in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 325, in _iterencode_list
    yield from chunks
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 438, in _iterencode
    o = _default(o)
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/core_tools.py", line 109, in default
    return super().default(obj)
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
TypeError: Object of type bytes is not JSON serializable
```

h-mayorquin, 2023-04-12:
@DradeAW 
You need to update your version of probeinterface.

DradeAW, 2023-04-12:
> You need to update your version of probeinterface.

```bash
wyngaard@nspc05:~/dev/spikeinterface/probeinterface$ git pull
Already up-to-date.
```

DradeAW, 2023-04-12:
Wait my bad ...

This conda environment didn't point to the folder but decided to install an old version of probeinterface ^^'
`python setup.py develop` fixed the issue!

Sorry about that

closed 2023-04-12 by DradeAW