SpikeInterface/spikeinterface #1525: Dump error in Lussac

DradeAW, 2023-04-12:
Hi,

In my algorithm (Lussac v2.0), something worked before but doesn't anymore after the updates :/

When trying to export the waveforms on a complex sorting object, this is the error I have:

```
Traceback (most recent call last):
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/bin/lussac", line 33, in <module>
    sys.exit(load_entry_point('lussac', 'console_scripts', 'lussac')())
  File "/export/home1/users/nsr/wyngaard/Documents/lussac2/lussac/main.py", line 58, in main
    pipeline.launch()
  File "/export/home1/users/nsr/wyngaard/Documents/lussac2/lussac/core/pipeline.py", line 47, in launch
    self._run_mono_sorting_module(module, module_key, category, params)
  File "/export/home1/users/nsr/wyngaard/Documents/lussac2/lussac/core/pipeline.py", line 86, in _run_mono_sorting_module
    sub_sorting = module_instance.run(params0)
  File "/export/home1/users/nsr/wyngaard/Documents/lussac2/lussac/modules/remove_bad_units.py", line 42, in run
    self._plot_bad_units(bad_sorting, reasons_for_removal)
  File "/export/home1/users/nsr/wyngaard/Documents/lussac2/lussac/modules/remove_bad_units.py", line 54, in _plot_bad_units
    wvf_extractor = self.extract_waveforms(sorting=bad_sorting, ms_before=1.5, ms_after=2.5, max_spikes_per_unit=500)
  File "/export/home1/users/nsr/wyngaard/Documents/lussac2/lussac/core/module.py", line 176, in extract_waveforms
    return si.extract_waveforms(recording, sorting, folder_path, allow_unfiltered=True, **params)
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/waveform_extractor.py", line 1446, in extract_waveforms
    we = WaveformExtractor.create(recording, sorting, folder, mode=mode, use_relative_path=use_relative_path,
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/waveform_extractor.py", line 279, in create
    sorting.dump(folder / 'sorting.json', relative_to=relative_to)
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/base.py", line 459, in dump
    self.dump_to_json(file_path, relative_to=relative_to, folder_metadata=folder_metadata)
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/base.py", line 485, in dump_to_json
    json.dumps(dump_dict, indent=4, cls=SIJsonEncoder),
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/__init__.py", line 238, in dumps
    **kw).encode(obj)
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 201, in encode
    chunks = list(chunks)
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 431, in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 439, in _iterencode
    yield from _iterencode(o, _current_indent_level)
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 431, in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 376, in _iterencode_dict
    raise TypeError(f'keys must be str, int, float, bool or None, '
TypeError: keys must be str, int, float, bool or None, not int64
```

Sorry I don't have a good 'least reproducible steps' ...

h-mayorquin, 2023-04-12:
Can you write the code that generated the error? I am interested on what you mean by a `complex sorting object` specially.
I need to know the type of your sorting. Is it possible for me to get my hands on it? 

Also, can you print  `dump_dict` in line 485 of base. There is some element there that has a strange dtype (int64) in a place where the encoder is not finding it. I need to see to see a print of the dump_dict in detail or you need to tell me where in that dict (or in your sorting which is where this probably type comes from) there is a key in a dictionary with type int

DradeAW, 2023-04-12:
I can try to send the data to you, but it contains a 257 GB recording so I'm not sure on how to do that ...
Maybe I can try to reproduce the error with a smaller recording.

By complex sorting, I mean that this is a PhySortinxExtractor on top of which multiple actions have been taken (so multiple parents).

I managed to print the `dump_dict` for a not-so complex sorting that crashed:
```python
{'class': 'spikeinterface.core.unitsselectionsorting.UnitsSelectionSorting', 'module': 'spikeinterface', 'kwargs': {'parent_sorting': AlignSortingExtractor: 238 units - 1 segments - 30.0kHz, 'unit_ids': [3, 5, 6, 7, 10, 11, 13, 15, 16, 17, 18, 24, 26, 28, 30, 31, 32, 34, 35, 38, 43, 49, 51, 56, 57, 60, 62, 63, 65, 68, 80, 81, 84, 87, 94, 96, 101, 107, 109, 110, 112, 114, 121, 122, 127, 130, 132, 133, 138, 140, 142, 144, 145, 153, 156, 157, 161, 162, 163, 164, 170, 179, 184, 192, 193, 195, 196, 200, 201, 203, 210, 212, 216, 217, 218, 220, 222, 223, 224, 225, 227, 228, 231, 239, 240], 'renamed_unit_ids': [3, 5, 6, 7, 10, 11, 13, 15, 16, 17, 18, 24, 26, 28, 30, 31, 32, 34, 35, 38, 43, 49, 51, 56, 57, 60, 62, 63, 65, 68, 80, 81, 84, 87, 94, 96, 101, 107, 109, 110, 112, 114, 121, 122, 127, 130, 132, 133, 138, 140, 142, 144, 145, 153, 156, 157, 161, 162, 163, 164, 170, 179, 184, 192, 193, 195, 196, 200, 201, 203, 210, 212, 216, 217, 218, 220, 222, 223, 224, 225, 227, 228, 231, 239, 240]}, 'dumpable': True, 'version': '0.98.0.dev0', 'relative_paths': False, 'annotations': {'name': 'ks2_pj8-3'}, 'properties': {}}
```

h-mayorquin, 2023-04-12:
Probably we don't the recorder. I am thinking that the culprit is the sorting object.

Can you run -after all the postprocessing- the method `dump_to_json` in your sorting object.

Also, can you list the actionst that were taken in the sorter. I am thinking that we would be able to reproduce with the local sorter than can be generated with:

```
from spikeinterface.core.datasets import download_dataset

local_path = download_dataset(remote_path="mearec/mearec_test_10s.h5")
recording, sorting = se.read_mearec(local_path)
```

So, getting that sorting object above, applying your list of actions and then call `dump_to_json` will hopefully reproduce the error locally in our side.

h-mayorquin, 2023-04-12:
Did the `dump_dict` created an error when you did it on the `AlignSortingExtractor`? My feeilng was that it should have generated the same error as the one above.

Can you copy-paste json like the one above? I think

DradeAW, 2023-04-12:
No it did not create an error,

Here is the json file (a bit long):

```json
{
    "class": "spikeinterface.postprocessing.alignsorting.AlignSortingExtractor",
    "module": "spikeinterface",
    "kwargs": {
        "sorting": {
            "class": "spikeinterface.curation.remove_duplicated_spikes.RemoveDuplicatedSpikesSorting",
            "module": "spikeinterface",
            "kwargs": {
                "sorting": {
                    "class": "spikeinterface.core.unitsselectionsorting.UnitsSelectionSorting",
                    "module": "spikeinterface",
                    "kwargs": {
                        "parent_sorting": {
                            "class": "spikeinterface.extractors.phykilosortextractors.PhySortingExtractor",
                            "module": "spikeinterface",
                            "kwargs": {
                                "folder_path": "/mnt/hdd/data/MEArec/45m_1500cells/ks2_pj8-3",
                                "exclude_cluster_groups": null
                            },
                            "dumpable": true,
                            "version": "0.98.0.dev0",
                            "relative_paths": false,
                            "annotations": {},
                            "properties": {}
                        },
                        "unit_ids": [
                            0,
                            1,
                            2,
                            3,
                            4,
                            5,
                            6,
                            7,
                            8,
                            9,
                            10,
                            11,
                            12,
                            13,
                            14,
                            15,
                            16,
                            17,
                            18,
                            19,
                            20,
                            21,
                            22,
                            23,
                            24,
                            25,
                            26,
                            27,
                            28,
                            29,
                            30,
                            31,
                            32,
                            33,
                            34,
                            35,
                            36,
                            37,
                            38,
[... 681 lines cut ...]
            "227": 6,
            "228": 7,
            "229": 2,
            "230": 0,
            "231": -1,
            "232": 0,
            "233": 0,
            "234": 0,
            "235": 0,
            "236": 0,
            "237": 0,
            "238": 0,
            "239": -1,
            "240": 43
        }
    },
    "dumpable": true,
    "version": "0.98.0.dev0",
    "relative_paths": false,
    "annotations": {
        "name": "ks2_pj8-3"
    },
    "properties": {}
}
```

h-mayorquin, 2023-04-13:
@DradeAW It is very strange that this is not generated an error. I tried doing your processing with Mearec and it works for me. Without the files I am unable to reproduce this. I created a branch that should produce a meanigful error here:
https://github.com/SpikeInterface/spikeinterface/pull/1529

Can you run your code against that branch and report the full trace?

This is a last resort so we don't need to exchange files.

DradeAW, 2023-04-13:
Here is the output:

```
Traceback (most recent call last):
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/base.py", line 486, in dump_to_json
    json.dumps(dump_dict, indent=4, cls=SIJsonEncoder),
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/__init__.py", line 238, in dumps
    **kw).encode(obj)
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 201, in encode
    chunks = list(chunks)
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 431, in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 439, in _iterencode
    yield from _iterencode(o, _current_indent_level)
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 431, in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 376, in _iterencode_dict
    raise TypeError(f'keys must be str, int, float, bool or None, '
TypeError: keys must be str, int, float, bool or None, not int64

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/bin/lussac", line 33, in <module>
    sys.exit(load_entry_point('lussac', 'console_scripts', 'lussac')())
  File "/export/home1/users/nsr/wyngaard/Documents/lussac2/lussac/main.py", line 58, in main
    pipeline.launch()
  File "/export/home1/users/nsr/wyngaard/Documents/lussac2/lussac/core/pipeline.py", line 47, in launch
    self._run_mono_sorting_module(module, module_key, category, params)
  File "/export/home1/users/nsr/wyngaard/Documents/lussac2/lussac/core/pipeline.py", line 86, in _run_mono_sorting_module
    sub_sorting = module_instance.run(params0)
  File "/export/home1/users/nsr/wyngaard/Documents/lussac2/lussac/modules/remove_bad_units.py", line 44, in run
    self._plot_bad_units(bad_sorting, reasons_for_removal)
  File "/export/home1/users/nsr/wyngaard/Documents/lussac2/lussac/modules/remove_bad_units.py", line 56, in _plot_bad_units
    wvf_extractor = self.extract_waveforms(sorting=bad_sorting, ms_before=1.5, ms_after=2.5, max_spikes_per_unit=500)
  File "/export/home1/users/nsr/wyngaard/Documents/lussac2/lussac/core/module.py", line 176, in extract_waveforms
    return si.extract_waveforms(recording, sorting, folder_path, allow_unfiltered=True, **params)
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/waveform_extractor.py", line 1446, in extract_waveforms
    we = WaveformExtractor.create(recording, sorting, folder, mode=mode, use_relative_path=use_relative_path,
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/waveform_extractor.py", line 279, in create
    sorting.dump(folder / 'sorting.json', relative_to=relative_to)
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/base.py", line 459, in dump
    self.dump_to_json(file_path, relative_to=relative_to, folder_metadata=folder_metadata)
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/base.py", line 514, in dump_to_json
    raise Exception(f'Error while dumping to json {str(e)}. \n \n \n  Dump dict : {log_value_types_of_dict(dump_dict)}')
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/base.py", line 500, in log_value_types_of_dict
    result[key_to_log] = log_value_types_of_dict(value)
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/base.py", line 502, in log_value_types_of_dict
    result[key_to_log] = [(element, log_value_types_of_dict(element)) for element in value]                
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/base.py", line 502, in <listcomp>
    result[key_to_log] = [(element, log_value_types_of_dict(element)) for element in value]                
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/base.py", line 493, in log_value_types_of_dict
    for key, value in dictionary.items():
AttributeError: 'numpy.int64' object has no attribute 'items'. Did you mean: 'item'?
```

DradeAW, 2023-04-13:
I also added `print(dictionary)` at the beginning of `log_value_types_of_dict()` to try to understand what's going on:

```
{'class': 'spikeinterface.core.unitsselectionsorting.UnitsSelectionSorting', 'module': 'spikeinterface', 'kwargs': {'parent_sorting': AlignSortingExtractor: 238 units - 1 segments - 30.0kHz, 'unit_ids': [3, 5, 6, 7, 10, 11, 13, 15, 16, 17, 18, 24, 26, 28, 30, 31, 32, 34, 35, 38, 43, 49, 51, 56, 57, 60, 62, 63, 65, 68, 80, 81, 84, 87, 94, 96, 101, 107, 109, 110, 112, 114, 121, 122, 127, 130, 132, 133, 138, 140, 142, 144, 145, 153, 156, 157, 161, 162, 163, 164, 170, 179, 184, 192, 193, 195, 196, 200, 201, 203, 210, 212, 216, 217, 218, 220, 222, 223, 224, 225, 227, 228, 231, 239, 240], 'renamed_unit_ids': [3, 5, 6, 7, 10, 11, 13, 15, 16, 17, 18, 24, 26, 28, 30, 31, 32, 34, 35, 38, 43, 49, 51, 56, 57, 60, 62, 63, 65, 68, 80, 81, 84, 87, 94, 96, 101, 107, 109, 110, 112, 114, 121, 122, 127, 130, 132, 133, 138, 140, 142, 144, 145, 153, 156, 157, 161, 162, 163, 164, 170, 179, 184, 192, 193, 195, 196, 200, 201, 203, 210, 212, 216, 217, 218, 220, 222, 223, 224, 225, 227, 228, 231, 239, 240]}, 'dumpable': True, 'version': '0.98.0.dev0', 'relative_paths': False, 'annotations': {'name': 'ks2_pj8-3'}, 'properties': {}}
{'class': 'spikeinterface.core.unitsselectionsorting.UnitsSelectionSorting', 'module': 'spikeinterface', 'kwargs': {'parent_sorting': AlignSortingExtractor: 238 units - 1 segments - 30.0kHz, 'unit_ids': [3, 5, 6, 7, 10, 11, 13, 15, 16, 17, 18, 24, 26, 28, 30, 31, 32, 34, 35, 38, 43, 49, 51, 56, 57, 60, 62, 63, 65, 68, 80, 81, 84, 87, 94, 96, 101, 107, 109, 110, 112, 114, 121, 122, 127, 130, 132, 133, 138, 140, 142, 144, 145, 153, 156, 157, 161, 162, 163, 164, 170, 179, 184, 192, 193, 195, 196, 200, 201, 203, 210, 212, 216, 217, 218, 220, 222, 223, 224, 225, 227, 228, 231, 239, 240], 'renamed_unit_ids': [3, 5, 6, 7, 10, 11, 13, 15, 16, 17, 18, 24, 26, 28, 30, 31, 32, 34, 35, 38, 43, 49, 51, 56, 57, 60, 62, 63, 65, 68, 80, 81, 84, 87, 94, 96, 101, 107, 109, 110, 112, 114, 121, 122, 127, 130, 132, 133, 138, 140, 142, 144, 145, 153, 156, 157, 161, 162, 163, 164, 170, 179, 184, 192, 193, 195, 196, 200, 201, 203, 210, 212, 216, 217, 218, 220, 222, 223, 224, 225, 227, 228, 231, 239, 240]}, 'dumpable': True, 'version': '0.98.0.dev0', 'relative_paths': False, 'annotations': {'name': 'ks2_pj8-3'}, 'properties': {}}
{'parent_sorting': AlignSortingExtractor: 238 units - 1 segments - 30.0kHz, 'unit_ids': [3, 5, 6, 7, 10, 11, 13, 15, 16, 17, 18, 24, 26, 28, 30, 31, 32, 34, 35, 38, 43, 49, 51, 56, 57, 60, 62, 63, 65, 68, 80, 81, 84, 87, 94, 96, 101, 107, 109, 110, 112, 114, 121, 122, 127, 130, 132, 133, 138, 140, 142, 144, 145, 153, 156, 157, 161, 162, 163, 164, 170, 179, 184, 192, 193, 195, 196, 200, 201, 203, 210, 212, 216, 217, 218, 220, 222, 223, 224, 225, 227, 228, 231, 239, 240], 'renamed_unit_ids': [3, 5, 6, 7, 10, 11, 13, 15, 16, 17, 18, 24, 26, 28, 30, 31, 32, 34, 35, 38, 43, 49, 51, 56, 57, 60, 62, 63, 65, 68, 80, 81, 84, 87, 94, 96, 101, 107, 109, 110, 112, 114, 121, 122, 127, 130, 132, 133, 138, 140, 142, 144, 145, 153, 156, 157, 161, 162, 163, 164, 170, 179, 184, 192, 193, 195, 196, 200, 201, 203, 210, 212, 216, 217, 218, 220, 222, 223, 224, 225, 227, 228, 231, 239, 240]}
{'class': 'spikeinterface.postprocessing.alignsorting.AlignSortingExtractor', 'module': 'spikeinterface', 'kwargs': {'sorting': RemoveDuplicatedSpikesSorting: 238 units - 1 segments - 30.0kHz, 'unit_peak_shifts': {0: 0, 1: 22, 2: -1, 3: 2, 4: 0, 5: 44, 6: 0, 7: 0, 8: 0, 9: 3, 10: 0, 11: 44, 12: 1, 13: -2, 14: 0, 15: 0, 16: 1, 17: 0, 18: 48, 19: 0, 20: 8, 21: 0, 22: 0, 23: 0, 24: -2, 25: 0, 26: 43, 27: 0, 28: 0, 29: 0, 30: 0, 31: 43, 32: 2, 33: 0, 34: 0, 35: 0, 36: 0, 37: 4, 38: -3, 39: 0, 40: 0, 41: 0, 42: 0, 43: 44, 44: 1, 45: 0, 46: 0, 47: 0, 48: 8, 49: 0, 50: 1, 51: 10, 52: 0, 53: 0, 54: 0, 55: 4, 56: 1, 57: 54, 58: 0, 59: 0, 60: -1, 62: 0, 63: 45, 64: 1, 65: 45, 66: 0, 67: 0, 68: 43, 69: 0, 70: 0, 71: 0, 72: 0, 73: 0, 74: 0, 75: -1, 76: 0, 77: 0, 79: 0, 80: 47, 81: -2, 82: 0, 83: 0, 84: 44, 85: 0, 86: 0, 87: -2, 88: 0, 89: -8, 90: 0, 91: 0, 92: 0, 93: 0, 94: 2, 95: 0, 96: 44, 97: 0, 98: -1, 99: -1, 100: 0, 101: 43, 102: 0, 103: 0, 104: -3, 105: 0, 106: 0, 107: 0, 108: 0, 109: 43, 110: -1, 111: 0, 112: -1, 113: 0, 114: 45, 115: 0, 116: 1, 117: -1, 118: 0, 119: 0, 120: -1, 121: -1, 122: 1, 123: 50, 124: 0, 125: 10, 126: 0, 127: 44, 128: 0, 129: 13, 130: 0, 131: 0, 132: -2, 133: 43, 134: 7, 135: 0, 136: 0, 137: 0, 138: 44, 139: 0, 140: -1, 142: -2, 143: 0, 144: 44, 145: 0, 146: 0, 147: 48, 148: -1, 149: 0, 150: 0, 151: 0, 152: 1, 153: 52, 154: 3, 155: 0, 156: 1, 157: 0, 158: 0, 159: 0, 160: 0, 161: 44, 162: 45, 163: -4, 164: 48, 165: 0, 166: 1, 167: 0, 168: 0, 169: -1, 170: 43, 171: 0, 172: 0, 173: 1, 174: 7, 175: -1, 176: 0, 177: 1, 178: 11, 179: 2, 180: 1, 181: 0, 182: 0, 183: 0, 184: 48, 185: 0, 186: -1, 187: 0, 188: 0, 189: 0, 190: 1, 191: 17, 192: 10, 193: 3, 194: 4, 195: 4, 196: 0, 197: 0, 198: 0, 199: -1, 200: -3, 201: -3, 202: 15, 203: 1, 204: 1, 205: 0, 206: -1, 207: 1, 208: 43, 209: 0, 210: -3, 211: 0, 212: -2, 213: 0, 214: 1, 215: 14, 216: -4, 217: 0, 218: 1, 219: 1, 220: 7, 221: 1, 222: 4, 223: -1, 224: -1, 225: 0, 226: 1, 227: 6, 228: 7, 229: 2, 230: 0, 231: -1, 232: 0, 233: 0, 234: 0, 235: 0, 236: 0, 237: 0, 238: 0, 239: -1, 240: 43}}, 'dumpable': True, 'version': '0.98.0.dev0', 'relative_paths': False, 'annotations': {}, 'properties': {}}
[... cut ...]
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/base.py", line 504, in <listcomp>
    result[key_to_log] = [(element, log_value_types_of_dict(element)) for element in value]                
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/base.py", line 495, in log_value_types_of_dict
    for key, value in dictionary.items():
AttributeError: 'numpy.int64' object has no attribute 'items'. Did you mean: 'item'?
```

DradeAW, 2023-04-13:
I think I figured out something:

```python
elif isinstance(value, (list, tuple)): # Iterables
    result[key_to_log] = [(element, log_value_types_of_dict(element)) for element in value]
```

`element` here is an `np.int64` and not a `dict`.

h-mayorquin, 2023-04-13:
Hi, @DradeAW 
There was an recursion error in my loggin function (I don't have material to debug so this is one shot). Can you try the new push?

h-mayorquin, 2023-04-13:
@DradeAW 
This part of your dict:
```
{}
{}
{}
{}
{}
{}
{0: 0, 1: 22, 2: -1, 3: 2, 4: 0, 5: 44, 6: 0, 7: 0, 8: 0, 9: 3, 10: 0, 11: 44, 12: 1, 13: -2, 14: 0, 15: 0, 16: 1, 17: 0, 18: 48, 19: 0, 20: 8, 21: 0, 22: 0, 23: 0, 24: -2, 25: 0, 26: 43, 27: 0, 28: 0, 29: 0, 30: 0, 31: 43, 32: 2, 33: 0, 34: 0, 35: 0, 36: 0, 37: 4, 38: -3, 39: 0, 40: 0, 41: 0, 42: 0, 43: 44, 44: 1, 45: 0, 46: 0, 47: 0, 48: 8, 49: 0, 50: 1, 51: 10, 52: 0, 53: 0, 54: 0, 55: 4, 56: 1, 57: 54, 58: 0, 59: 0, 60: -1, 62: 0, 63: 45, 64: 1, 65: 45, 66: 0, 67: 0, 68: 43, 69: 0, 70: 0, 71: 0, 72: 0, 73: 0, 74: 0, 75: -1, 76: 0, 77: 0, 79: 0, 80: 47, 81: -2, 82: 0, 83: 0, 84: 44, 85: 0, 86: 0, 87: -2, 88: 0, 89: -8, 90: 0, 91: 0, 92: 0, 93: 0, 94: 2, 95: 0, 96: 44, 97: 0, 98: -1, 99: -1, 100: 0, 101: 43, 102: 0, 103: 0, 104: -3, 105: 0, 106: 0, 107: 0, 108: 0, 109: 43, 110: -1, 111: 0, 112: -1, 113: 0, 114: 45, 115: 0, 116: 1, 117: -1, 118: 0, 119: 0, 120: -1, 121: -1, 122: 1, 123: 50, 124: 0, 125: 10, 126: 0, 127: 44, 128: 0, 129: 13, 130: 0, 131: 0, 132: -2, 133: 43, 134: 7, 135: 0, 136: 0, 137: 0, 138: 44, 139: 0, 140: -1, 142: -2, 143: 0, 144: 44, 145: 0, 146: 0, 147: 48, 148: -1, 149: 0, 150: 0, 151: 0, 152: 1, 153: 52, 154: 3, 155: 0, 156: 1, 157: 0, 158: 0, 159: 0, 160: 0, 161: 44, 162: 45, 163: -4, 164: 48, 165: 0, 166: 1, 167: 0, 168: 0, 169: -1, 170: 43, 171: 0, 172: 0, 173: 1, 174: 7, 175: -1, 176: 0, 177: 1, 178: 11, 179: 2, 180: 1, 181: 0, 182: 0, 183: 0, 184: 48, 185: 0, 186: -1, 187: 0, 188: 0, 189: 0, 190: 1, 191: 17, 192: 10, 193: 3, 194: 4, 195: 4, 196: 0, 197: 0, 198: 0, 199: -1, 200: -3, 201: -3, 202: 15, 203: 1, 204: 1, 205: 0, 206: -1, 207: 1, 208: 43, 209: 0, 210: -3, 211: 0, 212: -2, 213: 0, 214: 1, 215: 14, 216: -4, 217: 0, 218: 1, 219: 1, 220: 7, 221: 1, 222: 4, 223: -1, 224: -1, 225: 0, 226: 1, 227: 6, 228: 7, 229: 2, 230: 0, 231: -1, 232: 0, 233: 0, 234: 0, 235: 0, 236: 0, 237: 0, 238: 0, 239: -1, 240: 43}
{}
{}
```

I think that's the culprit. But I don't know what is generating it.

DradeAW, 2023-04-13:
If I had to guess, I would say that this is probably the shift from the `AlignmentSorting`.

Here is the output with the latest push:

```
Traceback (most recent call last):
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/base.py", line 486, in dump_to_json
    json.dumps(dump_dict, indent=4, cls=SIJsonEncoder),
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/__init__.py", line 238, in dumps
    **kw).encode(obj)
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 201, in encode
    chunks = list(chunks)
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 431, in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 439, in _iterencode
    yield from _iterencode(o, _current_indent_level)
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 431, in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/lib/python3.10/json/encoder.py", line 376, in _iterencode_dict
    raise TypeError(f'keys must be str, int, float, bool or None, '
TypeError: keys must be str, int, float, bool or None, not int64

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/users/nsr/wyngaard/dev/miniconda3/envs/lussac2/bin/lussac", line 33, in <module>
    sys.exit(load_entry_point('lussac', 'console_scripts', 'lussac')())
  File "/export/home1/users/nsr/wyngaard/Documents/lussac2/lussac/main.py", line 58, in main
    pipeline.launch()
  File "/export/home1/users/nsr/wyngaard/Documents/lussac2/lussac/core/pipeline.py", line 47, in launch
    self._run_mono_sorting_module(module, module_key, category, params)
  File "/export/home1/users/nsr/wyngaard/Documents/lussac2/lussac/core/pipeline.py", line 86, in _run_mono_sorting_module
    sub_sorting = module_instance.run(params0)
  File "/export/home1/users/nsr/wyngaard/Documents/lussac2/lussac/modules/remove_bad_units.py", line 44, in run
    self._plot_bad_units(bad_sorting, reasons_for_removal)
  File "/export/home1/users/nsr/wyngaard/Documents/lussac2/lussac/modules/remove_bad_units.py", line 56, in _plot_bad_units
    wvf_extractor = self.extract_waveforms(sorting=bad_sorting, ms_before=1.5, ms_after=2.5, max_spikes_per_unit=500)
  File "/export/home1/users/nsr/wyngaard/Documents/lussac2/lussac/core/module.py", line 176, in extract_waveforms
    return si.extract_waveforms(recording, sorting, folder_path, allow_unfiltered=True, **params)
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/waveform_extractor.py", line 1446, in extract_waveforms
    we = WaveformExtractor.create(recording, sorting, folder, mode=mode, use_relative_path=use_relative_path,
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/waveform_extractor.py", line 279, in create
    sorting.dump(folder / 'sorting.json', relative_to=relative_to)
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/base.py", line 459, in dump
    self.dump_to_json(file_path, relative_to=relative_to, folder_metadata=folder_metadata)
  File "/export/home1/users/nsr/wyngaard/dev/spikeinterface/spikeinterface/src/spikeinterface/core/base.py", line 531, in dump_to_json
    raise Exception(f'Error while dumping to json {str(e)}. \n \n \n  Dump dict : {log_value_types_of_dict(dump_dict)}')
Exception: Error while dumping to json keys must be str, int, float, bool or None, not int64. 
 
 
  Dump dict : {'class': 'spikeinterface.core.unitsselectionsorting.UnitsSelectionSorting', 'module': 'spikeinterface', 'kwargs': {'parent_sorting': (AlignSortingExtractor: 238 units - 1 segments - 30.0kHz, {'class': 'spikeinterface.postprocessing.alignsorting.AlignSortingExtractor', 'module': 'spikeinterface', 'kwargs': {'sorting': (RemoveDuplicatedSpikesSorting: 238 units - 1 segments - 30.0kHz, {'class': 'spikeinterface.curation.remove_duplicated_spikes.RemoveDuplicatedSpikesSorting', 'module': 'spikeinterface', 'kwargs': {'sorting': (UnitsSelectionSorting: 238 units - 1 segments - 30.0kHz, {'class': 'spikeinterface.core.unitsselectionsorting.UnitsSelectionSorting', 'module': 'spikeinterface', 'kwargs': {'parent_sorting': (PhySortingExtractor: 241 units - 1 segments - 30.0kHz, {'class': 'spikeinterface.extractors.phykilosortextractors.PhySortingExtractor', 'module': 'spikeinterface', 'kwargs': {'folder_path': '/mnt/hdd/data/MEArec/45m_1500cells/ks2_pj8-3', 'exclude_cluster_groups': (None, 'possible_trouble', 'NoneType')}, 'dumpable': True, 'version': '0.98.0.dev0', 'relative_paths': False, 'annotations': {}, 'properties':  [... cut ...] 'int64'), (220, 'possible_trouble', 'int64'), (222, 'possible_trouble', 'int64'), (223, 'possible_trouble', 'int64'), (224, 'possible_trouble', 'int64'), (225, 'possible_trouble', 'int64'), (227, 'possible_trouble', 'int64'), (228, 'possible_trouble', 'int64'), (231, 'possible_trouble', 'int64'), (239, 'possible_trouble', 'int64'), (240, 'possible_trouble', 'int64')]}, 'dumpable': True, 'version': '0.98.0.dev0', 'relative_paths': False, 'annotations': {'name': 'ks2_pj8-3'}, 'properties': {}}
```

h-mayorquin, 2023-04-13:
@DradeAW Thanks, I can reproduce the error now locally. I will come back to you as soon as I solve it.

h-mayorquin, 2023-04-14:
Let's keep this open until we merge #1529

closed 2023-04-14 by h-mayorquin