SpikeInterface/spikeinterface #1526: Issue with Intan data concatenation

Jacopo-Baldi, 2023-04-12:
Hi,

We are trying to use SpikeInterface to load, concatenate and spike sort data recorded with Intan. The following piece of script is what we are using to load and concatenate the .rhd files.

```
all_rhd = list(Path(my_data_path).glob("*.rhd"))

def tr(i):
    rec_extract=si.read_intan(all_rhd[i],stream_id='0')
    return rec_extract.get_traces()

recording_with_probes=NumpyRecording([tr(i) for i in range(len(all_rhd))],sampling_frequency)

recording_with_probes=si.concatenate_recordings([recording_with_probes])
```

And later in the script we want to run the sorting, and we tried the two following syntaxes

```
1:     sorting_MNS = ss.run_mountainsort4(recording=recording_with_probes_preprocessed)

2:     sorting_MNS = ss.run_sorter(sorter_name="mountainsort4", recording=recording_with_probes_preprocessed)
```

But in both cases we end up with this error. Here I put only the final piece of the error report, here is a drive link to two example files: [https://drive.google.com/drive/folders/1xYFwDy2e6x1er_zL6p6bzpw2qqy844zF?usp=sharing]

> /home/common/softs/spikesorting/spikeinterface/venv_0.95.1/lib/python3.8/json/encoder.py in default(self, o)
>     177
>     178         """
> --> 179         raise TypeError(f'Object of type {o.__class__.__name__} '
>     180                         f'is not JSON serializable')
>     181
> 
> TypeError: Object of type ndarray is not JSON serializable
> 

So we think that the issue is that we are using NumpyRecording, but we cannot find in the documentation an alternative to this to load our data and concatenate them.

If anybody has encountered similar issues or has any ideas why we are finding this, please let us know. Any info or suggestion is appreciated.

Jacopo

h-mayorquin, 2023-04-12:
Thanks for sharing the data. I will try to reproduce the error in our side. 
Please do not be afraid of sharing the complete trace. This is what issues are for. Can you include it the complete code AND the trace?

Jacopo-Baldi, 2023-04-13:
Thank you for your reply! Here is the complete trace of the error generated, and I added to the Drive folder the script "tool_analysis_SpikeInterface.ipynb"

>  ---------------------------------------------------------------------------
> TypeError                                 Traceback (most recent call last)
> /tmp/ipykernel_44110/2363140776.py in <cell line: 1>()
> ----> 1 sorting_MNS = ss.run_mountainsort4(recording=recording_with_probes_preprocessed)
> 
> /home/common/softs/spikesorting/spikeinterface/venv_0.95.1/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py in run_mountainsort4(*args, **kwargs)
>     624 
>     625 def run_mountainsort4(*args, **kwargs):
> --> 626     return run_sorter('mountainsort4', *args, **kwargs)
>     627 
>     628 
> 
> /home/common/softs/spikesorting/spikeinterface/venv_0.95.1/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
>     140         )
>     141 
> --> 142     return run_sorter_local(**common_kwargs)
>     143 
>     144 
> 
> /home/common/softs/spikesorting/spikeinterface/venv_0.95.1/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
>     155 
>     156     # only classmethod call not instance (stateless at instance level but state is in folder)
> --> 157     output_folder = SorterClass.initialize_folder(
>     158         recording, output_folder, verbose, remove_existing_folder)
>     159     SorterClass.set_params_to_folder(
> 
> /home/common/softs/spikesorting/spikeinterface/venv_0.95.1/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py in initialize_folder(cls, recording, output_folder, verbose, remove_existing_folder)
>     134         rec_file = output_folder / 'spikeinterface_recording.json'
>     135         if recording.is_dumpable:
> --> 136             recording.dump_to_json(rec_file)
>     137         else:
>     138             d = {'warning': 'The recording is not dumpable'}
> 
> /home/common/softs/spikesorting/spikeinterface/venv_0.95.1/lib/python3.8/site-packages/spikeinterface/core/base.py in dump_to_json(self, file_path, relative_to, folder_metadata)
>     481         file_path = self._get_file_path(file_path, ['.json'])
>     482         file_path.write_text(
> --> 483             json.dumps(check_json(dump_dict), indent=4),
>     484             encoding='utf8'
>     485         )
> 
> /home/common/softs/spikesorting/spikeinterface/venv_0.95.1/lib/python3.8/json/__init__.py in dumps(obj, skipkeys, ensure_ascii, check_circular, allow_nan, cls, indent, separators, default, sort_keys, **kw)
>     232     if cls is None:
>     233         cls = JSONEncoder
> --> 234     return cls(
>     235         skipkeys=skipkeys, ensure_ascii=ensure_ascii,
>     236         check_circular=check_circular, allow_nan=allow_nan, indent=indent,
> 
> /home/common/softs/spikesorting/spikeinterface/venv_0.95.1/lib/python3.8/json/encoder.py in encode(self, o)
>     199         chunks = self.iterencode(o, _one_shot=True)
>     200         if not isinstance(chunks, (list, tuple)):
> --> 201             chunks = list(chunks)
>     202         return ''.join(chunks)
>     203 
> 
> /home/common/softs/spikesorting/spikeinterface/venv_0.95.1/lib/python3.8/json/encoder.py in _iterencode(o, _current_indent_level)
>     429             yield from _iterencode_list(o, _current_indent_level)
>     430         elif isinstance(o, dict):
> --> 431             yield from _iterencode_dict(o, _current_indent_level)
>     432         else:
>     433             if markers is not None:
> 
> /home/common/softs/spikesorting/spikeinterface/venv_0.95.1/lib/python3.8/json/encoder.py in _iterencode_dict(dct, _current_indent_level)
>     403                 else:
>     404                     chunks = _iterencode(value, _current_indent_level)
> --> 405                 yield from chunks
>     406         if newline_indent is not None:
>     407             _current_indent_level -= 1
> 
[... 53 lines cut ...]
>     406         if newline_indent is not None:
>     407             _current_indent_level -= 1
> 
> /home/common/softs/spikesorting/spikeinterface/venv_0.95.1/lib/python3.8/json/encoder.py in _iterencode_list(lst, _current_indent_level)
>     323                 else:
>     324                     chunks = _iterencode(value, _current_indent_level)
> --> 325                 yield from chunks
>     326         if newline_indent is not None:
>     327             _current_indent_level -= 1
> 
> /home/common/softs/spikesorting/spikeinterface/venv_0.95.1/lib/python3.8/json/encoder.py in _iterencode(o, _current_indent_level)
>     436                     raise ValueError("Circular reference detected")
>     437                 markers[markerid] = o
> --> 438             o = _default(o)
>     439             yield from _iterencode(o, _current_indent_level)
>     440             if markers is not None:
> 
> /home/common/softs/spikesorting/spikeinterface/venv_0.95.1/lib/python3.8/json/encoder.py in default(self, o)
>     177 
>     178         """
> --> 179         raise TypeError(f'Object of type {o.__class__.__name__} '
>     180                         f'is not JSON serializable')
>     181 
> 
> TypeError: Object of type ndarray is not JSON serializable

h-mayorquin, 2023-04-13:
OK, can you run this and tell me if this is produces the same error in your system:

```
from pathlib import Path  
from spikeinterface.extractors import read_intan, NumpyRecording
from spikeinterface.core.segmentutils import concatenate_recordings

my_data_path = Path("/home/heberto/Downloads/my_data_path/")  # Modify this

all_rhd = list(Path(my_data_path).glob("*.rhd"))

def tr(i):
    rec_extract= read_intan(all_rhd[i],stream_id='0')
    return rec_extract.get_traces()

sampling_frequency = read_intan(all_rhd[0], stream_id="0").get_sampling_frequency()
recording_with_probes=NumpyRecording([tr(i) for i in range(len(all_rhd))],sampling_frequency)

recording_with_probes= concatenate_recordings([recording_with_probes])

import spikeinterface.preprocessing as spre

recording_with_probes_f = spre.bandpass_filter(recording_with_probes, freq_min=300, freq_max=6000)
print(recording_with_probes_f)
recording_with_probes_cmr = spre.common_reference(recording_with_probes_f, reference='global', operator='median')
recording_with_probes_preprocessed = recording_with_probes_cmr

import spikeinterface.sorters as ss
sorting_MNS = ss.run_mountainsort4(recording=recording_with_probes_preprocessed)
```

Does this cover the essence of what are you doing? Am I missing any step? Also, how are you installing and running spikeinterface?

alejoe91, 2023-04-13:
Thanks @h-mayorquin 

@Jacopo-Baldi you don't need to make numpy recordings! You can directly concatenate the intan recordings with:

```
all_rhd = list(Path(my_data_path).glob("*.rhd"))

recording_list = [si.read_intan(rhd_file, stream_id='0') for rhd_file in all_rhd]

recording_with_probes=si.concatenate_recordings(recording_list)
```

This will probably fix your sumping issue since, as you suggest, the `NumpyRecording` is what makes it not dumpable. A second advantage of this approach is that it's only going to load traces *lazily*, while constructing the NumpyRecordings will load all traces to your RAM (and this is not needed and not recommended)!

Jacopo-Baldi, 2023-04-13:
Thanks @alejoe91 and @h-mayorquin for the suggestions, I put together your two pieces (bypassing the NumpyRecording solved it) and now it works. I added the set_probegroup step because the sorting was giving an error of missing properties, and now it seems to be running without errors.
Thank you for your help!

alejoe91, 2023-04-13:
Glad it works! Cheers :)

closed 2023-04-13 by alejoe91

h-mayorquin, 2023-04-13:
It is still strange that Numpy is generating the error. It should be dumpable now. Let' see if this pops out somewhere else.