SpikeInterface/spikeinterface #1536: argument plot_templates=True in plot_unit_waveforms_density_map() causes an error

rat-h, 2023-04-13:
```
>>> import spikeinterface.full as si
>>> from spikeinterface import widgets as spw
>>> si.__version__
'0.97.1'
>>> we  = si.WaveformExtractor.load_from_folder("TestRun-Tr2wPP/waveforms")
>>> spw.plot_unit_waveforms_density_map(we,unit_ids=[0]); show()
<spikeinterface.widgets.matplotlib.unit_waveforms_density_map.UnitWaveformDensityMapPlotter object at 0x7fdf469cf730>
>>> spw.plot_unit_waveforms_density_map(we,unit_ids=[0],plot_templates=True); show()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "~/spikes/lib/python3.10/site-packages/spikeinterface/widgets/base.py", line 116, in widget_func
    W.do_plot(W.backend, **W.backend_kwargs)
  File "~/spikes/lib/python3.10/site-packages/spikeinterface/widgets/base.py", line 52, in do_plot
    self.check_backend_kwargs(plotter, backend, **backend_kwargs)
  File "~/spikes/lib/python3.10/site-packages/spikeinterface/widgets/base.py", line 46, in check_backend_kwargs
    raise Exception(f"{k} is not a valid plot argument or backend keyword argument. "
Exception: plot_templates is not a valid plot argument or backend keyword argument. Possible backend keyword arguments for matplotlib are: ['figure', 'ax', 'axes', 'ncols', 'figsize', 'figtitle']
```

alejoe91, 2023-04-14:
@rat-h indeed `plot_templates` is not an argument of the function: https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.widgets.plot_unit_waveforms_density_map

Maybe you want to use `sw.plot_unit_waveforms` instead?
See https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.widgets.plot_unit_waveforms

rat-h, 2023-04-14:
I marked by >
```
>>> from spikeinterface import widgets as spw
>>> help(spw.plot_unit_waveforms_density_map)
Help on function plot_unit_waveforms_density_map in module spikeinterface.widgets.base:

plot_unit_waveforms_density_map(waveform_extractor, channel_ids=None, unit_ids=None, sparsity=None, same_axis=False, use_max_channel=False, peak_sign='neg', unit_colors=None, backend=None, **backend_kwargs)
    Plots unit waveforms using heat map density.
    
    Parameters
    ----------
    waveform_extractor: WaveformExtractor
    channel_ids: list
        The channel ids to display
    unit_ids: list
        List of unit ids.
>  plot_templates: bool
        If True, templates are plotted over the waveforms
    sparsity : ChannelSparsity or None
        Optional ChannelSparsity to apply.
        If WaveformExtractor is already sparse, the argument is ignored
    use_max_channel: bool default False
        Use only the max channel
    peak_sign: str "neg"
        Used to detect max channel only when use_max_channel=True 
    unit_colors: None or dict
        A dict key is unit_id and value is any color format handled by matplotlib.
        If None, then the get_unit_colors() is internally used.
    same_axis: bool
        If True then all density are plot on the same axis and then channels is the union
        all channel per units.
    set_title: bool
        Create a plot title with the unit number if True.
    plot_channels: bool
        Plot channel locations below traces, only used if channel_locs is True
    
    
    backend: str
        ['matplotlib']
    **backend_kwargs: kwargs
    
        matplotlib:
    
        * figure: Matplotlib figure. When None, it is created. Default None
        * ax: Single matplotlib axis. When None, it is created. Default None
        * axes: Multiple matplotlib axes. When None, they is created. Default None
        * ncols: Number of columns to create in subplots.  Default 5
        * figsize: Size of matplotlib figure. Default None
        * figtitle: The figure title. Default None

```
the same in the documentation you referred.

zm711, 2023-04-14:
@alejoe91 

It's in the doc-string, which might be contributing to the confusion. I can work on some doc updates for the widgets next unless you want to just fix that right away.

[image]

alejoe91, 2023-04-14:
Oh sorry I didn't see it! Yes we should have a pass in the docstring and make sure they're up to date...

rat-h, 2023-04-14:
@alejoe91 or maybe you can make this argument work?
:)

alejoe91, 2023-04-14:
Honestly, I think that the `plot_templates` was never implemented for this function, but it's there for a bad copy/paste :P

I'll assign this to @samuelgarcia, since he made the `plot_unit_waveforms_density_map` in the first place. It's a good idea to have the possibility to add the template on top of the densities

alejoe91, 2023-11-06:
Fixed by @zm711

closed 2023-11-06 by alejoe91