SpikeInterface/spikeinterface #1538: No channel locations in OpenEphys metadata

guidomeijer, 2023-04-14:
First off, thank you for developing this amazing tool! Everything works fine for SpikeGLX data but I'm a bit confused about something about the OpenEphys GUI data I have. Whereas SpikeGLX saves the channel locations in the metadata, OpenEphys does not seem to do that. When I run
```
rec = se.read_openephys(oe_path, stream_id='1')
bad_channel_ids, all_channels = spre.detect_bad_channels(rec)
```
I get the error
```
Traceback (most recent call last):

  Cell In[3], line 6
    bad_channel_ids, all_channels = spre.detect_bad_channels(rec)

  File ~\mambaforge\envs\spikeinterface\Lib\site-packages\spikeinterface\preprocessing\detect_bad_channels.py:158 in detect_bad_channels
    channel_locations = recording.get_channel_locations()

  File ~\mambaforge\envs\spikeinterface\Lib\site-packages\spikeinterface\core\baserecordingsnippets.py:305 in get_channel_locations
    raise Exception('There are no channel locations')

Exception: There are no channel locations
```
I have a channel map file that you would typically use as the chanMap parameter when running Kilosort in MATLAB. This file contains the information about the channel locations. However I don't see how I would add the channel map information into the spikeinterface pipeline. Also `run_sorter` does not accept chanMap as a parameter.
```
rec = run_sorter('kilosort3', rec, output_folder=output_path, verbose=True, chanMap=chan_path, docker_image=True)
```
Throws an `AttributeError: Bad parameters: [\'chanMap\']` error. 
I'm on spikeinterface version 0.97.1 on a Windows machine. 
Any help would be appreciated!

guidomeijer, 2023-04-14:
Thanks @alejoe91, the data was collected with OpenEphys GUI version 0.5.4 so that's indeed the explanation. Your solution works great. Here's the code that worked for me:
```
cmap = scipy.io.loadmat(chan_path)
channel_locs = np.hstack((cmap['xcoords'], cmap['ycoords']))
rec.set_channel_locations(channel_locs)
```
One additional small question: I'd like to run the sample shift preprocessing on these data (which again works fine for SpikeGLX) but it gives this error
```
Traceback (most recent call last):

  Cell In[49], line 2
    rec = spre.phase_shift(rec)

  File ~\mambaforge\envs\spikeinterface\Lib\site-packages\spikeinterface\core\core_tools.py:26 in reader_func
    return source_class(*args, **kwargs)

  File ~\mambaforge\envs\spikeinterface\Lib\site-packages\spikeinterface\preprocessing\phase_shift.py:42 in __init__
    assert "inter_sample_shift" in recording.get_property_keys(), "'inter_sample_shift' is not a property!"

AssertionError: 'inter_sample_shift' is not a property!
```
Do you know how to get the inter_sample_shift for these OpenEphys data? If not, not a big deal!

alejoe91, 2023-04-14:
Yep, you can get the sample shifts as follows. 
Note that we recently discovered that for AP streams there was a slight mistake in our implementation (see #1511), which makes phase shift sub-optimal (but still better than not doing it!)

Before that PR gets merged, this is what you should do:
```
from spikeinterface.extractors.neuropixels_utils import get_neuropixels_sample_shifts

total_channels = 384
num_channels_per_adc = 12 # for NP1, 16 for NP2

# compute inter_sample_shifts
inter_sample_shifts = get_neuropixels_sample_shifts(total_channels, num_channels_per_adc)
rec.set_property("inter_sample_shift", inter_sample_shift)
```

After #1511 is merged, you can also correctly use the `num_cycles` argument:
Before that PR gets merged, this is what you should do:
```
from spikeinterface.extractors.neuropixels_utils import get_neuropixels_sample_shifts

total_channels = 384
num_channels_per_adc = 12 # for NP1, 16 for NP2
num_cycles = 13 # for AP, 12 for LF (for NP2 it's always 16)

# compute inter_sample_shifts
inter_sample_shifts = get_neuropixels_sample_shifts(total_channels, num_channels_per_adc, num_cycles)
rec.set_property("inter_sample_shift", inter_sample_shift)
```

Then `inter_sample_shift` is added as a property, it's automatically used by the `phase_shift` function.

guidomeijer, 2023-04-14:
Great, thanks! 

Sorry one last thing: when running `interpolate_bad_channels` it says `ValueError: A probe must be attached to use bad channel interpolation. Use set_probe(...)`. I tried adding one like so but it still gives the same error
```
probe = Probe(ndim=2, si_units='um')
probe.set_contacts(positions=channel_locs, shapes='circle', shape_params={'radius': 5})
probe.set_device_channel_indices(np.arange(channel_locs.shape[0]))
rec.set_probe(probe)
```

alejoe91, 2023-04-14:
Almost there!!

`rec.set_probe(probe)` --> `rec.set_probe(probe, in_place=True)` (or `rec_with_probe = rec.set_probe(probe)`)

guidomeijer, 2023-04-14:
Thanks! The prepocessing steps now pass without error but when I run the spike sorting like so
```
rec = run_sorter('kilosort3', rec, output_folder=os.path.join(root, 'Kilosort3'), verbose=True, docker_image=True)
```
I still get `ValueError: A probe must be attached to use bad channel interpolation. Use set_probe(...)`
Any idea why?

alejoe91, 2023-04-14:
Mmm can it be that you are using `rec` as return variable?

Should be:
```
sort = run_sorter(...)
```

guidomeijer, 2023-04-14:
Still the same error unfortunately..

alejoe91, 2023-04-14:
Oh, is it possible that you set the probe after preprocessing?

Can you try to set the probe as a first step and then apply the pre-processing steps?

guidomeijer, 2023-04-14:
It's the first thing I do, here is the code I run:
```
# Load in recording
rec = se.read_openephys(oe_path, stream_id=root[-1])

# Add channel locations
probe = Probe(ndim=2, si_units='um')
probe.set_contacts(positions=channel_locs, shapes='circle', shape_params={'radius': 5})
probe.set_device_channel_indices(np.arange(channel_locs.shape[0]))
rec.set_probe(probe, in_place=True)
                
# Set inter sample shift            
inter_sample_shifts = get_neuropixels_sample_shifts(384, 12)
            
# Correct for inter sample shifts
rec = spre.highpass_filter(rec)
rec = spre.phase_shift(rec, inter_sample_shift=inter_sample_shifts)

# Detect and interpolate bad channels
bad_channel_ids, all_channels = spre.detect_bad_channels(rec)
rec = spre.interpolate_bad_channels(rec, bad_channel_ids)
rec = spre.highpass_spatial_filter(rec)

# Run Kilosort3
sort = run_sorter('kilosort3', rec, output_folder=os.path.join(root, 'Kilosort3'), verbose=True, docker_image=True)
```

alejoe91, 2023-04-14:
Can you try to skip the `spre.interpolate_bad_channels(rec, bad_channel_ids)`?
Or try to subsitutute the iterpolate + highpass with `spre.common_reference()`

The reason of the failure could be the following:

The `spre.detect_bad_channels(rec)` (by default it uses the "IBL" approach) and it returns different channel labels (`all_channels`): "good"\"noise"\"dead"\"out" (see https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.preprocessing.detect_bad_channels)

If you have channels outside of the brain (so they are at the boundary), they should not be interpolated, but directly removed. Otherwise interpolation will be an effective extrapolation and it could mess up the highpass spatial filter:

```
bad_channel_ids, channel_labels = spre.detect_bad_channels(rec)
out_channels = rec.channel_ids[channel_labels=="out"]
noise_and_dead_channels = rec.channel_ids[channel_labels in ("noise", "dead")]

# remove out channels
rec = rec.remove_channels(out_channels)
# interpolate the rest
rec = spre.interpolate_bad_channels(rec, noise_and_dead_channels)
rec = spre.highpass_spatial_filter(rec)

# remove interpolated channels after highpass spatial
rec = rec.remove_channels(noise_and_dead_channels)
```

I think that removing the interpolated channels is also a good practice, since they do not provide any "new" information and will only be a spatially filtered version of surrounding channels (but of course it will help for spatial filtering, because spatial continuity is assumed).

Note, that you can also visualize the recording in a notebook with the `ipywidgets` backend, which is very convenient.
Docs here: https://spikeinterface.readthedocs.io/en/latest/modules/widgets.html#id5

guidomeijer, 2023-04-19:
It's working now, thanks for all your help!

alejoe91, 2023-04-19:
@guidomeijer  can I ask you if the problem was indeed due to the `out` channels? Or if you had to switch to common ref?

guidomeijer, 2023-04-19:
I had to switch to common ref, I tried the `out` channels thing you suggested but that still gave issues.

alejoe91, 2023-04-19:
I'll reopen then ;) Would you ind sharing the dataset with us?

alejoe91, 2023-11-06:
Closing for inactivity

closed 2023-11-06 by alejoe91