SpikeInterface/spikeinterface #1542: max  unit information in min function calls

rat-h, 2023-04-14:
## background
my colleagues record from very young animals (before and just after birth). The neuron dynamics are very different at these ages. Spikes are wider/slower with much longer after-hyperpolarization. Well, this makes spike sorting quite hard. So we have tested quite a lot of algorithms (huge thanks to `spikeinterface`) with default parameters, and all of them do not yield so many single units or even multiunits. So now we want to _play with the parameters_ of sorters that produce the best results.

## What we need
We want to get as much information as possible about each unit so that a human can evaluate whether it is a single unit, multiunit, or garbage. I'm aware of many widgets for results visualization and of `plot_sorting_summary` (which works only, unfortunately, in `sortingview`), as well as the overall concept of using `sortingview` for data curation. However, just visualization is different from what we need.

So how to extract the information listed below by calling as few functions as possible?
- on which electrode is the unit maximum, and what are the other electrodes should be shown for traces (like in `spikeinterface_gui`)?
- how to extract all unit appearances on traces?
- how to get unit mean and std on each electrode

Of course, I already discovered very useful functions like `compute_correlograms` , `compute_isi_histograms`, and `compute_template_metrics`

alejoe91, 2023-04-27:
Hi @rat-h 

To get the extremum channel and the sparsity of each unit, checkout these functions:
- [get_template_extremum_channel](https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.core.get_template_extremum_channel)
- [compute_sparsity](https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.core.compute_sparsity)

To extract all waveforms for all spikes, you can run the `extract_waveforms()` function and using the `max_spikes_per_unit=None` argument.

You can get template means and std on each electrode with:
```
template_means = we.get_all_templates(mode='average')
template_stdss = we.get_all_templates(mode='std')
```

rat-h, 2023-04-27:
@alejoe91 thank you for the answer. Very helpful!

The last bit, which is still an open question, is how to get "snapshots" of each spike for a given unit. It should actually be 2D or 3D numpy arrays with 200-300 ms samples for each relevant channel (sparsity can help to cat all others) for each spike. So the shape of the array should be (num of samples, num of channels, num of spikes). Can I get something like that in a simple way?

alejoe91, 2023-04-27:
Oh, ok so after you have extracted waveforms for all spikes, you can use the `all_wfs_unit = we.get_waveforms(unit_id)` function. For example, if unit 10 has 3000 spikes, `all_wfs_unit` will have a shape of: `(3000, num_samples, num_channels)`.

You can control the `num_samples` by setting the `ms_before`, `ms_after` args when calling the [`extract_waveforms()` function](https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.core.extract_waveforms)

rat-h, 2023-04-27:
perfect! Thank you!

closed 2023-04-27 by rat-h