SpikeInterface/spikeinterface #1546: Wrong probe map from open ephys and unexpected position from peak position

Kayv-cmb, 2023-04-17:
Hi all,

Here is my issue (with some plot).
I started using a 4 shanks neuropixel probe for my recording and when I try to load the open ephys (version 0.64)  recording into spikeinterface (version 0.96.1), my channel location are different from the one I have in my settings.xml
```
from probeinterface import Probe, ProbeGroup
from probeinterface import generate_linear_probe, generate_multi_shank

directory = '/camp/home/combadk/working/raw_data/raw_NP/NP_230315_A/2023-03-15_18-07-02'
raw_rec = si.read_openephys(directory, stream_id ='0',block_index=1)
raw_rec

fig, ax = plt.subplots(figsize=(15, 10))
si.plot_probe_map(raw_rec, ax=ax, with_channel_ids=True)
ax.set_ylim(-800, 750)
```

It give this result
[image]

When I would expect this probe map, that I created from the settings.xml from open ephys.
[image]

I changed the probe with those line and did some preprocessing.
```
df_probe = raw_rec.get_probe().to_dataframe()
chan_map = io.loadmat( '/camp/home/combadk/working/combadk/Kilosort/configFiles/chanMap.mat')
chan_location = pd.DataFrame(chan_map['xcoords'][0,:],chan_map['ycoords'][0,:])
df_probe['x'] = chan_map['xcoords'][0,:]
df_probe['y'] = chan_map['ycoords'][0,:]
df_probe

probe = Probe.from_dataframe(df_probe)
probe.set_device_channel_indices(df_probe.index)
rec = raw_rec.set_probe(probe)
rec1 = si.highpass_filter(rec, freq_min=400.)
rec4 = si.common_reference(rec1, operator="median", reference="global")
preprocess_recording = rec4
job_kwargs = dict(n_jobs=40, chunk_duration='1s', progress_bar=True)
peaks = detect_peaks(preprocess_recording,  method='locally_exclusive', noise_levels=noise_levels_int16,
                     detect_threshold=5, local_radius_um=50., **job_kwargs)

peak_locations = localize_peaks(preprocess_recording, peaks, method='center_of_mass', **job_kwargs)
# check for drifts
fs = rec.sampling_frequency
fig, ax = plt.subplots(figsize=(10, 8))
ax.scatter(peaks['sample_ind'] / fs, peak_locations['y'], color='k', marker='.',  alpha=0.002)
# we can also use the peak location estimates to have an insight of cluster separation before sorting
fig, ax = plt.subplots(figsize=(15, 10))
si.plot_probe_map(preprocess_recording, ax=ax, with_channel_ids=True)
ax.set_ylim(-800, 800)

ax.scatter(peak_locations['x'], peak_locations['y'], color='purple', alpha=0.002)
```

However the result from me checking on the drift looks like my first probe map which I expect to be wrong.
[image]

Any help would be appreciated! Thank you in advance!

Kayv-cmb, 2023-04-17:
Hi @alejoe91,

Of course, I opened it at :
https://github.com/SpikeInterface/probeinterface/issues/185#issue-1671691730

alejoe91, 2023-04-19:
Thanks, closing this then!

closed 2023-04-19 by alejoe91