SpikeInterface/spikeinterface #1547: Possible bug in MultiSortingComparison._populate_spiketrains

cristofer-holobetz, 2023-04-18:
Hi! I have been attempting to generate an `AgreementSorting` object based on putative units derived from KiloSort2 and Mountainsort4.

For some units, `agreement_spiketrains` derived from `MultiSortingComparison.get_agreement_sorting` of KS2 and MS4 putative units do not correctly reflect the intersection of the input spiketrains.

I think this is because of the re-use of variable name `unit_id` on lines 97 and 98 in the method `_populate_spiketrains`

From reading the code, it seems there are three relevant 'reference frames' for unit ids. Each sorting has its own 'reference frame', and there is a third, global reference frame that combines the units from each sorting.

In the inner loop of `_populate_spiketrains`, `unit_id` references this 'global' reference frame. This global `unit_id` is later used to save the agreement_spiketrain in `spike_trains_segment[unit_id]`.

On line 97, `unit_id` is reassigned to a value from one of the original sorting objects' reference frames. This ends up overwriting some agreement_spiketrains with totally unrelated spiketrains that happen to share the same `unit_id` but from the other sorting.

As an example from my sortings, KS2's unit 183 is a good match with MS4's unit 112. When `unit_id == 183`, `spike_trains_segment[183] `is appropriately saved as the intersection of spikes between `KS2[183]` and `MS4[112]`.

However, when `unit_id == 820` in the global reference frame, line 97: `unit_id == list(sorter_unit_ids.values())[0]` sets `unit_id` equal to 183 because 820 in the global reference frame maps to 183 in the MS4 reference frame.

Because the variable `unit_id` is reassigned here, the subsequent saving of the spiketrain, which should have been to `spike_trains_segment[820]`, ends up overwriting `spike_trains_segment[183]`.

I think this can easily be fixed by renaming the `unit_id` variable on lines 97 and 98 to something else, i.e. `this_sorter_unit_id`.

As an aside, the language in the tutorial notebooks and readthedocs was a little confusing to me. An important feature of the comparisons module is that it can let you extract spiketrains containing only the spikes agreed upon by multiple sorters. I assumed that this would be the default behavior, but it's necessary to change the default parameter of `spiketrain_mode` from 'union' to 'intersection' to achieve this. It might be helpful to make this clear in the documentation or to change the default from 'union' to 'intersection'.

Thanks so much!

h-mayorquin, 2023-04-19:
Hi, @cristofer-holobetz , thanks for opening an issue.

The central issue is that:
> For some units, agreement_spiketrains derived from MultiSortingComparison.get_agreement_sorting of KS2 and MS4 putative units do not correctly reflect the intersection of the input spiketrains

Can you elaborate on that. Showing us how they look and how they look and if possible the code that you are running will help us greatly with the issue.

alejoe91, 2023-04-19:
Thanks @cristofer-holobetz 

Yes by default the agreement spike trains are the `union`, we should highlight it more in the docs. This is a choice: imagine one of the compared sorters finds more "real" spikes of the same unit, then the union would keep those.

Since you already identified the source of the problem, would you have time to make a PR?

h-mayorquin, 2023-04-19:
Also, as a tip, you can reference specific lines of the repo in the following way. You open in github the specific file and can as in this gif (pressing shift while selecting lines):

[image]

This will generate the following linke that you can paste here to make your point stronger:
https://github.com/SpikeInterface/spikeinterface/blob/b850677b14e0ff8ebcd2d6075b364081114cbd67/src/spikeinterface/comparison/multicomparisons.py#L92-L95

cristofer-holobetz, 2023-04-19:
I'll submit a PR. For reference, here are plots that illustrate the problem and solution. Blue are spikes from KS2, orange are spikes from MS4, and green are supposed to be the intersection of these spikes. Plot titles are the `unit_ids` w.r.t. the different `SortingExtractors`

Before the fix:
[image]

After the fix:
[image]

cristofer-holobetz, 2023-04-20:
I submitted PR 1560 to resolve this at https://github.com/SpikeInterface/spikeinterface/pull/1560.

closed 2023-04-21 by cristofer-holobetz