SpikeInterface/spikeinterface #1559: numba compile error

MichaelCoulter, 2023-04-19:
hi. im getting this error with peak detection. im using spikeinterface 0.98.0. thanks. this happens while running the spyglass command: SpikeSorting.populate()

Running spike sorting on {'nwb_file_name': 'CH101_20210624_.nwb', 'sort_group_id': 11, 'sort_interval_name': 'raw data valid times', 'preproc_params_name': 'franklab_tetrode_hippocampus', 'team_name': 'mcoulter section', 'sorter': 'clusterless_thresholder', 'sorter_params_name': 'clusterless_fixed_2', 'artifact_removed_interval_list_name': 'CH101_20210624_.nwb_raw data valid times_11_franklab_tetrode_hippocampus_ampl_100_2ms_artifact_removed_valid_times'}...
detect peaks: 0%
0/2683 [00:00<?, ?it/s]
---------------------------------------------------------------------------
CompilerError                             Traceback (most recent call last)
Cell In [9], line 3
      1 spikesorting_selection_keys = recording_keys[2:3]
      2 #
----> 3 SpikeSorting.populate(spikesorting_selection_keys)
      4 SpikeSortingSelection & spikesorting_selection_keys

File ~/anaconda3/envs/spyglass2/lib/python3.9/site-packages/datajoint/autopopulate.py:230, in AutoPopulate.populate(self, suppress_errors, return_exception_objects, reserve_jobs, order, limit, max_calls, display_progress, processes, make_kwargs, *restrictions)
    226 if processes == 1:
    227     for key in (
    228         tqdm(keys, desc=self.__class__.__name__) if display_progress else keys
    229     ):
--> 230         error = self._populate1(key, jobs, **populate_kwargs)
    231         if error is not None:
    232             error_list.append(error)

File ~/anaconda3/envs/spyglass2/lib/python3.9/site-packages/datajoint/autopopulate.py:281, in AutoPopulate._populate1(self, key, jobs, suppress_errors, return_exception_objects, make_kwargs)
    279 self.__class__._allow_insert = True
    280 try:
--> 281     make(dict(key), **(make_kwargs or {}))
    282 except (KeyboardInterrupt, SystemExit, Exception) as error:
    283     try:

File ~/spyglass/src/spyglass/spikesorting/spikesorting_sorting.py:203, in SpikeSorting.make(self, key)
    200     sorter_params.pop("whiten", None)
    202     # Detect peaks for clusterless decoding
--> 203     detected_spikes = detect_peaks(recording, **sorter_params)
    204     sorting = si.NumpySorting.from_times_labels(
    205         times_list=detected_spikes["sample_ind"],
    206         labels_list=np.zeros(len(detected_spikes), dtype=np.int),
    207         sampling_frequency=recording.get_sampling_frequency(),
    208     )
    209 else:

File ~/spikeinterface/src/spikeinterface/sortingcomponents/peak_detection.py:103, in detect_peaks(recording, method, pipeline_nodes, gather_mode, folder, names, **kwargs)
     99 init_args = (recording, method, method_args, extra_margin, pipeline_nodes)
    100 processor = ChunkRecordingExecutor(recording, func, init_func, init_args,
    101                                    gather_func=gather_func, job_name='detect peaks',
    102                                    mp_context=mp_context, **job_kwargs)
--> 103 processor.run()
    105 outs = gather_func.finalize_buffers(squeeze_output=squeeze_output)
    106 return outs

File ~/spikeinterface/src/spikeinterface/core/job_tools.py:344, in ChunkRecordingExecutor.run(self)
    342 worker_ctx = self.init_func(*self.init_args)
    343 for segment_index, frame_start, frame_stop in all_chunks:
--> 344     res = self.func(segment_index, frame_start, frame_stop, worker_ctx)
    345     if self.handle_returns:
    346         returns.append(res)

File ~/spikeinterface/src/spikeinterface/sortingcomponents/peak_detection.py:154, in _detect_peaks_chunk(segment_index, start_frame, end_frame, worker_ctx)
    151     trace_detection = traces
    153 # TODO: handle waveform returns
--> 154 peak_sample_ind, peak_chan_ind = method_class.detect_peaks(trace_detection, *method_args)
    156 if extra_margin > 0:
    157     peak_sample_ind += extra_margin

File ~/spikeinterface/src/spikeinterface/sortingcomponents/peak_detection.py:357, in DetectPeakLocallyExclusive.detect_peaks(cls, traces, peak_sign, abs_threholds, exclude_sweep_size, neighbours_mask)
    354     peak_mask_pos = peak_mask.copy()
    356 peak_mask = traces_center < -abs_threholds[None, :]
--> 357 peak_mask = _numba_detect_peak_neg(traces, traces_center, peak_mask, exclude_sweep_size,
    358                                 abs_threholds, peak_sign, neighbours_mask)
    360 if peak_sign == 'both':
    361     peak_mask = peak_mask | peak_mask_pos
[... 100 lines cut ...]

File ~/anaconda3/envs/spyglass2/lib/python3.9/site-packages/numba/core/compiler_machinery.py:303, in PassManager._runPass(self, index, pss, internal_state)
    300 # Check that if the pass is an instance of a FunctionPass that it hasn't
    301 # emitted ir.Dels.
    302 if isinstance(pss, FunctionPass):
--> 303     enforce_no_dels(internal_state.func_ir)
    305 if self._ENFORCING:
    306     # TODO: Add in self consistency enforcement for
    307     # `func_ir._definitions` etc
    308     if _pass_registry.get(pss.__class__).mutates_CFG:

File ~/anaconda3/envs/spyglass2/lib/python3.9/site-packages/numba/core/ir_utils.py:2199, in enforce_no_dels(func_ir)
   2197 if dels:
   2198     msg = "Illegal IR, del found at: %s" % dels[0]
-> 2199     raise CompilerError(msg, loc=dels[0].loc)

CompilerError: Failed in object mode pipeline (step: remove phis nodes)
Illegal IR, del found at: del $66for_iter.3

File "../../spikeinterface/src/spikeinterface/sortingcomponents/peak_detection.py", line 450:
    def _numba_detect_peak_neg(traces, traces_center, peak_mask, exclude_sweep_size,
        <source elided>
                    continue
                for neighbour in range(num_chans):
                ^

alejoe91, 2023-04-20:
@MichaelCoulter what version of numba are you using?

MichaelCoulter, 2023-04-20:
numba: 0.55.2

alejoe91, 2023-04-20:
Can you try to upgrade to 0.56.3?

MichaelCoulter, 2023-04-20:
it still crashes but i get a new error now:
Running spike sorting on {'nwb_file_name': 'CH101_20210624_.nwb', 'sort_group_id': 11, 'sort_interval_name': 'raw data valid times', 'preproc_params_name': 'franklab_tetrode_hippocampus', 'team_name': 'mcoulter section', 'sorter': 'clusterless_thresholder', 'sorter_params_name': 'clusterless_fixed_2', 'artifact_removed_interval_list_name': 'CH101_20210624_.nwb_raw data valid times_11_franklab_tetrode_hippocampus_ampl_100_2ms_artifact_removed_valid_times'}...
detect peaks: 0%
0/2683 [00:00<?, ?it/s]
/home/mcoulter/spikeinterface/src/spikeinterface/sortingcomponents/peak_detection.py:442: NumbaWarning: 
Compilation is falling back to object mode WITH looplifting enabled because Function _numba_detect_peak_neg failed at nopython mode lowering due to: LLVM IR parsing error
<string>:1455:19: error: invalid cast opcode for cast from 'i16' to 'float'
  %".949" = fpext i16 %".890" to float
                  ^

  @numba.jit(parallel=False)
20-Apr-23 09:29:48 finding looplift candidates
/home/mcoulter/spikeinterface/src/spikeinterface/sortingcomponents/peak_detection.py:442: NumbaWarning: 
Compilation is falling back to object mode WITHOUT looplifting enabled because Function "_numba_detect_peak_neg" failed type inference due to: Cannot determine Numba type of <class 'numba.core.dispatcher.LiftedLoop'>

File "../../spikeinterface/src/spikeinterface/sortingcomponents/peak_detection.py", line 446:
    def _numba_detect_peak_neg(traces, traces_center, peak_mask, exclude_sweep_size,
        <source elided>
        num_chans = traces_center.shape[1]
        for chan_ind in range(num_chans):
        ^

  @numba.jit(parallel=False)
/home/mcoulter/anaconda3/envs/spyglass3/lib/python3.9/site-packages/numba/core/object_mode_passes.py:151: NumbaWarning: Function "_numba_detect_peak_neg" was compiled in object mode without forceobj=True, but has lifted loops.

File "../../spikeinterface/src/spikeinterface/sortingcomponents/peak_detection.py", line 445:
    def _numba_detect_peak_neg(traces, traces_center, peak_mask, exclude_sweep_size,
        <source elided>
                               abs_threholds, peak_sign, neighbours_mask):
        num_chans = traces_center.shape[1]
        ^

  warnings.warn(errors.NumbaWarning(warn_msg,
/home/mcoulter/anaconda3/envs/spyglass3/lib/python3.9/site-packages/numba/core/object_mode_passes.py:161: NumbaDeprecationWarning: 
Fall-back from the nopython compilation path to the object mode compilation path has been detected, this is deprecated behaviour.

For more information visit https://numba.readthedocs.io/en/stable/reference/deprecation.html#deprecation-of-object-mode-fall-back-behaviour-when-using-jit

File "../../spikeinterface/src/spikeinterface/sortingcomponents/peak_detection.py", line 445:
    def _numba_detect_peak_neg(traces, traces_center, peak_mask, exclude_sweep_size,
        <source elided>
                               abs_threholds, peak_sign, neighbours_mask):
        num_chans = traces_center.shape[1]
        ^

  warnings.warn(errors.NumbaDeprecationWarning(msg,
/home/mcoulter/spikeinterface/src/spikeinterface/sortingcomponents/peak_detection.py:442: NumbaWarning: 
Compilation is falling back to object mode WITHOUT looplifting enabled because Function _numba_detect_peak_neg failed at nopython mode lowering due to: LLVM IR parsing error
<string>:1416:19: error: invalid cast opcode for cast from 'i16' to 'float'
  %".930" = fpext i16 %".871" to float
                  ^

  @numba.jit(parallel=False)
---------------------------------------------------------------------------
CompilerError                             Traceback (most recent call last)
Cell In [3], line 54
     52 position_info_param_name = "default_decoding"
     53 # want to skip tet 100 and 101,
---> 54 populate_spike_threshold(recording_keys[2:3], 
     55                          mark_param_name=mark_param_name, 
     56                          position_info_param_name=position_info_param_name )

Cell In [3], line 14, in populate_spike_threshold(spikesorting_selection_keys, mark_param_name, position_info_param_name)
      1 def populate_spike_threshold(
      2         spikesorting_selection_keys: list,
      3         mark_param_name='default',
   (...)
      8     # either do spike sorting or marks - not both
      9     ## Populate spike sorting
     10     SpikeSortingSelection().insert(
[... 165 lines cut ...]

File ~/anaconda3/envs/spyglass3/lib/python3.9/site-packages/numba/core/compiler_machinery.py:318, in PassManager._runPass(self, index, pss, internal_state)
    315 # Check that if the pass is an instance of a FunctionPass that it hasn't
    316 # emitted ir.Dels.
    317 if isinstance(pss, FunctionPass):
--> 318     enforce_no_dels(internal_state.func_ir)
    320 if self._ENFORCING:
    321     # TODO: Add in self consistency enforcement for
    322     # `func_ir._definitions` etc
    323     if _pass_registry.get(pss.__class__).mutates_CFG:

File ~/anaconda3/envs/spyglass3/lib/python3.9/site-packages/numba/core/ir_utils.py:2194, in enforce_no_dels(func_ir)
   2192 if dels:
   2193     msg = "Illegal IR, del found at: %s" % dels[0]
-> 2194     raise CompilerError(msg, loc=dels[0].loc)

CompilerError: Failed in object mode pipeline (step: remove phis nodes)
Illegal IR, del found at: del $66for_iter.3

File "../../spikeinterface/src/spikeinterface/sortingcomponents/peak_detection.py", line 450:
    def _numba_detect_peak_neg(traces, traces_center, peak_mask, exclude_sweep_size,
        <source elided>
                    continue
                for neighbour in range(num_chans):
                ^

MichaelCoulter, 2023-04-20:
it looks like the input we are sending to spikeinterface ('recording') is float, but the function in spikeinterface (_numba_detect_peak_neg) is expecting an int16. can this function be modified to take either data type or does it need to be int16?

alejoe91, 2023-04-21:
I see! The `numba` function was written by @samuelgarcia, so I'll let him comment here.

If it's not possible, though, we should at least switch to the numpy implementation if the recording is float!

MichaelCoulter, 2023-04-21:
we realized the recording for us should be int and it was changed to float by mistake. we changed it back and now it works. im not sure if this function would need to accept float for other users though.

MichaelCoulter, 2023-04-27:
ah okay. yes that would make sense with what i saw. the recording datatype was int16 and then was changed to float32 and then it crashed. we have changed back to int16, so it seems fine for now.

closed 2023-04-27 by MichaelCoulter