SpikeInterface/spikeinterface #1568: BaseExtractor.dump_to_json error when trying to run sorter

programerkotik, 2023-04-21:
Hi!

I am trying to run spike sorter from spikeinterface on my data that are from Neuralynx. The raw data were in .ncs format but I am converting them to NumPy arrays first and then to the NumpyRecording object. 
Then I process with an object as follows: 
1. I set a probe by `recording.set_probes(probe, in_place=True)`. The probe I create using the `generate_linear_probe` function from probeinterface. 
2. Then I do bandpass filtering using `bandpass_filter` from spikeinterface.preprocessing. 
3. And I set common reference using `common_reference`. 

So, then I send this recording to the `run_sorter` function:

`ss.run_sorter(sorter_name="mountainsort4", recording=recording_cmr)`

I get an error from run_sorter failing in BaseExtractor.dump_to_json. I get the same type of error when trying to run other sorters as well. The full error is

```
`AssertionError                            Traceback (most recent call last)
[.../processing/spike-sorting/spike_sorting.ipynb](https://file+.vscode-resource.vscode-cdn.net/Users/septentrionalem/Desktop/study/thesis_2023_travelling_waves/processing/spike-sorting/spike_sorting.ipynb) Cell 11 in 1
----> [1](vscode-notebook-cell:.../processing/spike-sorting/spike_sorting.ipynb#X13sZmlsZQ%3D%3D?line=0) sorting_TDC = ss.run_sorter(sorter_name="mountainsort4", recording=recording_cmr)
      [2](vscode-notebook-cell:.../processing/spike-sorting/spike_sorting.ipynb#X13sZmlsZQ%3D%3D?line=1) print(sorting_TDC)

File [/opt/miniconda3/envs/neuro/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:143](https://file+.vscode-resource.vscode-cdn.net/opt/miniconda3/envs/neuro/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:143), in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
    136             container_image = singularity_image
    137     return run_sorter_container(
    138         container_image=container_image,
    139         mode=mode,
    140         **common_kwargs,
    141     )
--> 143 return run_sorter_local(**common_kwargs)

File [/opt/miniconda3/envs/neuro/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:158](https://file+.vscode-resource.vscode-cdn.net/opt/miniconda3/envs/neuro/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py:158), in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
    155 SorterClass = sorter_dict[sorter_name]
    157 # only classmethod call not instance (stateless at instance level but state is in folder)
--> 158 output_folder = SorterClass.initialize_folder(
    159     recording, output_folder, verbose, remove_existing_folder)
    160 SorterClass.set_params_to_folder(
    161     recording, output_folder, sorter_params, verbose)
    162 SorterClass.setup_recording(recording, output_folder, verbose=verbose)

File [/opt/miniconda3/envs/neuro/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py:134](https://file+.vscode-resource.vscode-cdn.net/opt/miniconda3/envs/neuro/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py:134), in BaseSorter.initialize_folder(cls, recording, output_folder, verbose, remove_existing_folder)
    132 rec_file = output_folder [/](https://file+.vscode-resource.vscode-cdn.net/) 'spikeinterface_recording.json'
    133 if recording.is_dumpable:
--> 134     recording.dump_to_json(rec_file)
    135 else:
    136     d = {'warning': 'The recording is not dumpable'}
File [/opt/miniconda3/envs/neuro/lib/python3.9/site-packages/spikeinterface/core/base.py:476](https://file+.vscode-resource.vscode-cdn.net/opt/miniconda3/envs/neuro/lib/python3.9/site-packages/spikeinterface/core/base.py:476), in BaseExtractor.dump_to_json(self, file_path, relative_to, folder_metadata)
    464 def dump_to_json(self, file_path=None, relative_to=None, folder_metadata=None):
    465     """
    466     Dump recording extractor to json file.
    467     The extractor can be re-loaded with load_extractor_from_json(json_file)
   (...)
    474         If not None, file_paths are serialized relative to this path
    475     """
--> 476     assert self.check_if_dumpable()
    477     dump_dict = self.to_dict(include_annotations=True,
    478                              include_properties=False,
    479                              relative_to=relative_to,
    480                              folder_metadata=folder_metadata)
    481     file_path = self._get_file_path(file_path, ['.json'])

AssertionError: 

```
I have installed spikeinterface through pip. And I am using it on Mac with an M1 chip. 
Here is also the list of packages I have in a conda environment:

```
# packages in environment at /opt/miniconda3/envs/neuro:
#
# Name                    Version                   Build  Channel
anyio                     3.6.2                    pypi_0    pypi
[... 128 lines cut ...]
tk                        8.6.12               hb8d0fd4_0  
toolz                     0.12.0                   pypi_0    pypi
tornado                   6.2              py39h1a28f6b_0  
tqdm                      4.64.1                   pypi_0    pypi
traitlets                 5.5.0                    pypi_0    pypi
tzdata                    2022f                h04d1e81_0  
urllib3                   1.26.15                  pypi_0    pypi
viziphant                 0.2.0                    pypi_0    pypi
wcwidth                   0.2.5              pyhd3eb1b0_0  
webencodings              0.5.1                    pypi_0    pypi
websocket-client          1.4.1                    pypi_0    pypi
wheel                     0.37.1             pyhd3eb1b0_0  
xarray                    2023.3.0                 pypi_0    pypi
xz                        5.2.6                h1a28f6b_0  
yaml                      0.2.5                h1a28f6b_0  
zarr                      2.14.2                   pypi_0    pypi
zeromq                    4.3.4                hc377ac9_0  
zipp                      3.10.0                   pypi_0    pypi
zlib                      1.2.13               h5a0b063_0  
```

Would you know what is the cause of the error? 

Thanks a lot,
Anna

alejoe91, 2023-04-21:
Hi,

The problem is the `NumpyRecording`, which is not associated to a persistent folder.

You can fix it by savinf to a binary file (after setting the probe) with:
```
recording = recording.save(folder="some-folder")
```

programerkotik, 2023-04-22:
Hi,
Thanks a lot. The suggestion worked!

closed 2023-04-22 by programerkotik