SpikeInterface/spikeinterface #1575: Running mountainsort with multiple cores

Jacopo-Baldi, 2023-04-24:
Hi everyone, I am working with SpikeInterface to sort data and encountered this issue while trying to run the sorter in parallel. I think that I need to set the parameter n_jobs or num_workers to tell the function to run over multiple cores, but I tried multiple options, among which:
```
job_kwargs = dict(n_jobs=60, progress_bar=True)

spikeinterface.core.set_global_job_kwargs(**job_kwargs)
``` 
and
`sorting_MNS = ss.run_mountainsort4(recording=myRec,num_workers = 60)`
and
`sorter_params = dict(num_workers=60)
sorting_MNS = ss.run_mountainsort4(recording=myRec,**sorter_params)`

None of these seem to have any effect, and the sorting runs with just 1 core, but it does not give an error.

I also tried the next two but it gives the same error in both cases.

`sorting_MNS = ss.run_mountainsort4(recording=myRec,**job_kwargs)
`
`sorting_MNS = ss.run_sorter(sorter_name='mountainsort4',recording=myRec,**job_kwargs)`

> ---------------------------------------------------------------------------

> AttributeError                            Traceback (most recent call last)
> /tmp/ipykernel_64544/2182925487.py in <cell line: 4>()
>       2 
>       3 myRec=recs[0].channel_slice([c])
> ----> 4 sorting_MNS = ss.run_sorter(sorter_name='mountainsort4',recording=myRec,**job_kwargs)
> 
> ~/.local/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
>     141         )
>     142 
> --> 143     return run_sorter_local(**common_kwargs)
>     144 
>     145 
> 
> ~/.local/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
>     158     output_folder = SorterClass.initialize_folder(
>     159         recording, output_folder, verbose, remove_existing_folder)
> --> 160     SorterClass.set_params_to_folder(
>     161         recording, output_folder, sorter_params, verbose)
>     162     SorterClass.setup_recording(recording, output_folder, verbose=verbose)
> 
> ~/.local/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py in set_params_to_folder(cls, recording, output_folder, new_params, verbose)
>     164                 bad_params.append(p)
>     165         if len(bad_params) > 0:
> --> 166             raise AttributeError('Bad parameters: ' + str(bad_params))
>     167 
>     168         params.update(new_params)
> 
> AttributeError: Bad parameters: ['n_jobs', 'chunk_size', 'progress_bar']

I tried different values of the number of cores, including -1 which should mean use all the available ones, but with no effect. I also tried using another sorter (tridesclous) and the problem seems to be the same.
Is there something wrong with the way I am using the parameter for multiple core usage? I am sure that at a certain point in the past I had managed to do it, but since then I lost the script I had used, so the environment should be able to do it, and I tried using ipyparallel and it works so I know that the machine supports parallel computing.
Any help would be greatly appreciated.

Jacopo-Baldi, 2023-04-25:
> Hi Jacopo. The job_kwargs logic is mainly use internally by spiekinterface for rec.save(..., **job_kwargs) or detect_peak(..., **job_kwargs) all external sorters have there own way to make the parallel computing. I am not sure how this can be handle for montainsort. @magland any idea ?

Hi Sam,
Thank you for your reply! I thought that the parameter num_workers in the method of mountainsort (the second thing I tried) was specifically supposed to exploit the parallel computing, is it not?

Jacopo-Baldi, 2023-04-27:
@samuelgarcia Do you think that it would be crazy to try to use ipyparallel to run mountainsort4 in parallel one core for each channel (which can be sorted independently between one another) changing the function run_mountainsort4 so that it does not save the data always in the same folder (which would overwrite the results when parallelized)?

alejoe91, 2023-04-27:
@Jacopo-Baldi we already have a mechanism to parallelize over groups! Do you want to run each channel in parallel?

In that case you can do the following (I assume you have 60 cores in your machine):
```
import spikeinterface.sorters as ss

# set channel groups
recording.set_channel_groups(np.arange(recording.get_num_channels()))

sorting_by_group = ss.run_sorter_by_property("mountainsort4", recording, grouping_property="group", working_folder="ms4", engine="joblib", engine_kwargs={"n_jobs": 60})
```
The `sorting_by_group` is a single sorting object, but each unit will have a `group` property with the channel group on which it was found! Let me know if it works!

alejoe91, 2023-04-27:
@Jacopo-Baldi I'm not @magland :P

Can you try to add the following: `engine_kwargs={"n_jobs": 60, "backend": "multiprocessing"}`

Jacopo-Baldi, 2023-04-27:
@alejoe91 Ops sorry for my oversight :D
I tried the modifications you said, now there is this error instead.

> ---------------------------------------------------------------------------
> RemoteTraceback                           Traceback (most recent call last)
> RemoteTraceback: 
> """
> Traceback (most recent call last):
>   File "/home/common/jbaldi/Scripts/spikeinterface/lib/python3.8/multiprocessing/pool.py", line 125, in worker
>     result = (True, func(*args, **kwds))
>   File "/home/common/jbaldi/Scripts/spikeinterface/lib/python3.8/site-packages/joblib/_parallel_backends.py", line 595, in __call__
>     return self.func(*args, **kwargs)
>   File "/home/common/jbaldi/Scripts/spikeinterface/lib/python3.8/site-packages/joblib/parallel.py", line 262, in __call__
>     return [func(*args, **kwargs)
>   File "/home/common/jbaldi/Scripts/spikeinterface/lib/python3.8/site-packages/joblib/parallel.py", line 262, in <listcomp>
>     return [func(*args, **kwargs)
>   File "/home/jbaldi/.local/lib/python3.8/site-packages/spikeinterface/sorters/launcher.py", line 38, in _run_one
>     run_sorter(sorter_name, recording, output_folder=output_folder,
>   File "/home/jbaldi/.local/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py", line 143, in run_sorter
>     return run_sorter_local(**common_kwargs)
>   File "/home/jbaldi/.local/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py", line 165, in run_sorter_local
>     sorting = SorterClass.get_result_from_folder(output_folder)
>   File "/home/jbaldi/.local/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py", line 280, in get_result_from_folder
>     raise SpikeSortingError(
> spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in spikeinterface_log.json
> """
> 
> The above exception was the direct cause of the following exception:
> 
> SpikeSortingError                         Traceback (most recent call last)
> /tmp/ipykernel_23353/3784849406.py in <cell line: 1>()
> ----> 1 sorting_by_group = ss.run_sorter_by_property("mountainsort4", myRec, grouping_property="group", working_folder="ms4", engine="joblib", engine_kwargs={"n_jobs": 60,"backend":"multiprocessing"})
> 
> ~/.local/lib/python3.8/site-packages/spikeinterface/sorters/launcher.py in run_sorter_by_property(sorter_name, recording, grouping_property, working_folder, mode_if_folder_exists, engine, engine_kwargs, verbose, docker_image, singularity_image, **sorter_params)
>     117                                                                f"a recording property!"
>     118     recording_dict = recording.split_by(grouping_property)
> --> 119     sorting_output = run_sorters([sorter_name], recording_dict, working_folder,
>     120                                  mode_if_folder_exists=mode_if_folder_exists,
>     121                                  engine=engine,
> 
> ~/.local/lib/python3.8/site-packages/spikeinterface/sorters/launcher.py in run_sorters(sorter_list, recording_dict_or_list, working_folder, sorter_params, mode_if_folder_exists, engine, engine_kwargs, verbose, with_output, docker_images, singularity_images)
>     273         n_jobs = engine_kwargs.get('n_jobs', -1)
>     274         backend = engine_kwargs.get('backend', 'loky')
> --> 275         Parallel(n_jobs=n_jobs, backend=backend)(
>     276             delayed(_run_one)(task_args) for task_args in task_args_list)
>     277 
> 
> /home/common/jbaldi/Scripts/spikeinterface/lib/python3.8/site-packages/joblib/parallel.py in __call__(self, iterable)
>    1054 
>    1055             with self._backend.retrieval_context():
> -> 1056                 self.retrieve()
>    1057             # Make sure that we get a last message telling us we are done
>    1058             elapsed_time = time.time() - self._start_time
> 
> /home/common/jbaldi/Scripts/spikeinterface/lib/python3.8/site-packages/joblib/parallel.py in retrieve(self)
>     933             try:
>     934                 if getattr(self._backend, 'supports_timeout', False):
> --> 935                     self._output.extend(job.get(timeout=self.timeout))
>     936                 else:
>     937                     self._output.extend(job.get())
> 
> /home/common/jbaldi/Scripts/spikeinterface/lib/python3.8/multiprocessing/pool.py in get(self, timeout)
>     769             return self._value
>     770         else:
> --> 771             raise self._value
>     772 
>     773     def _set(self, i, obj):
> 
> SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in spikeinterface_log.json
> 

While this is the spikeinterface_log.json

> {
    "sorter_name": "mountainsort4",
    "sorter_version": "unknown",
    "datetime": "2023-04-27T16:34:45.363503",
    "runtime_trace": [],
    "error": true,
    "error_trace": "Traceback (most recent call last):\n  File \"/home/jbaldi/.local/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py\", line 226, in run_from_folder\n    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)\n  File \"/home/jbaldi/.local/lib/python3.8/site-packages/spikeinterface/sorters/external/mountainsort4.py\", line 133, in _run_from_folder\n    old_api_sorting = mountainsort4.mountainsort4(**ms4_params)\n  File \"/home/common/jbaldi/Scripts/spikeinterface/lib/python3.8/site-packages/mountainsort4/mountainsort4.py\", line 38, in mountainsort4\n    MS4.sort()\n  File \"/home/common/jbaldi/Scripts/spikeinterface [... cut ...] 114, in h5py._proxy.dset_rw\nOSError: [Errno 28] Can't write data (file write failed: time = Thu Apr 27 16:35:44 2023\n, filename = '/tmp/tmpv8voyzwv/timeseries.hdf5', file descriptor = 73, errno = 28, error message = 'No space left on device', buf = 0x5557983d94b0, total write size = 4212160, bytes this sub-write = 4212160, bytes actually written = 18446744073709551615, offset = 0)\n",
    "run_time": null
}

Which is quite weird because I have plenty of space on the device (multiple terabytes)

Jacopo-Baldi, 2023-05-02:
If with sorting by property it does not work, do you think it would be too difficult to modify the run_sorter scripts to parallelize them myself? Like changing the folder structure that they create so that it does not overwrite the results while multiple run_sorter run on different channels with ipyparallel

Jacopo-Baldi, 2023-05-09:
I have not found any workaroud yet, @alejoe91 do you have any ideas? Is it something in the spikeinterface library that should be changed or an issue in the way we are using it?

Jacopo-Baldi, 2023-05-12:
I am still stuck on the same issue, is there anything I can do to fix it?

alejoe91, 2023-05-12:
Hi @Jacopo-Baldi 

Not sure..seems like something is wrong with multiprocessing. 
let me include @magland here, who is the author of mountainsort. Maybe he can help!

magland, 2023-05-12:
Hi @Jacopo-Baldi I'm not sure what you can do. My only suggestions (I know it's not very helpful) are to (a) Look through the source code of [mountainsort4](https://github.com/magland/mountainsort4), or (b) have a look at [mountainsort5](https://github.com/flatironinstitute/mountainsort5). Note that with ms5, there is an [open PR](https://github.com/SpikeInterface/spikeinterface/pull/1624) to incorporate the latest version into spikeinterface.

alejoe91, 2023-11-06:
@Jacopo-Baldi still facing this issue?

zm711, 2024-06-21:
I think the movement to MS5, couple to the lack of activity warrants closing this. @Jacopo-Baldi if this is still an issue please feel free to reopen or open a new issue.

closed 2024-06-21 by zm711